Screenly/Anthias #1566: Screenly not starting after manual install and reboot

kirkmorrow, 2022-01-12:
I followed the instructions on https://support.screenly.io/hc/en-us/articles/360042164173-Getting-started-with-Screenly-OSE to manually install on a fresh Raspberry Pi OS install on an RPi3 Model B.  No error messages.  After reboot I expected Screenly to start per the instructions but instead it booted to the default Raspberry Pi OS gui screen.

Steps to reproduce this issue, eg:

* Use Raspberry Pi Imager to install Raspberry Pi OS on fresh SD card
* Boot Raspberry Pi 3, Model B with SD card
* Follow the configuration setups on the Raspberry Pi configuration tool that comes up automatically
* Enable GUI, not CLI
* Enable auto-login
* Set password as required
* bash <(curl -sL https://www.screenly.io/install-ose.sh)
* Reboot when prompted
* Wait for the reboot to complete and see that Screenly did not start

* Raspberry Pi Hardware Version: 3, Model B
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: Whatever the above command gets

kirkmorrow, 2022-01-12:
Never mind, I followed these instructions instead which appear to work:

https://support.screenly.io/hc/en-us/articles/360025164713-How-to-flash-out-Screenly-to-an-SD-card

closed 2022-01-12 by kirkmorrow

oekomat, 2022-03-11:
> Never mind, I followed these instructions instead which appear to work:
> 
> https://support.screenly.io/hc/en-us/articles/360025164713-How-to-flash-out-Screenly-to-an-SD-card

After booting with LAN I get  a barcode with login. But I don`t have a acocount yet. In Version from 2020 I don*t need a account, but now?

oekomat, 2022-03-11:
> I followed the instructions on https://support.screenly.io/hc/en-us/articles/360042164173-Getting-started-with-Screenly-OSE to manually install on a fresh Raspberry Pi OS install on an RPi3 Model B. [...]

I have the same problem, after booting my screen is grey. But in the browser from my PC in the same network, i can*t get the webinterface per IP

vpetersson, 2022-03-12:
> After booting with LAN I get a barcode with login. But I don`t have a acocount yet. In Version from 2020 I don*t need a account, but now?

You're using the commercial version of Screenly, rather than Screenly OSE. For the commercial version, you need to use the dashboard (login.screenlyapp.com) to manage the device(s).