Screenly/Anthias #1568: Installation script is outdated. Some packages have no installation candidate.

NassimBentarka, 2022-02-07:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

The main installation script does not install properly on Raspbian Lite. My environment is Raspberry Pi Zero W. It looks like python-pip was renamed to python3-pip and other packages were renamed as well.
The result is that Screenly doesn't install on a fresh Raspbian Lite on Raspberry Pi Zero W.

#### Reproduction Steps

On a fresh Raspbian Lite on Raspberry Pi Zero W, run: 

`bash <(curl -sL https://www.screenly.io/install-ose.sh)`

#### Output

`       _____                           __         ____  _____ ______
      / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
      \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
     ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
    /____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                     /____/
Screenly OSE requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
________________________________________

Which version/branch of Screenly OSE would you like to install:

 Press (1) for the Production branch, which is the latest stable.
 Press (2) for the Development/Master branch, which has the latest features and fixes, but things may break.

You selected: Production

Do you want Screenly OSE to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to perform a full system upgrade as well? (y/N)
+ REPOSITORY=https://github.com/screenly/screenly-ose.git
+ sudo mkdir -p /etc/ansible
+ sudo tee /etc/ansible/hosts
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt update -y
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease           
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease        
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
+ sudo apt-get purge -y python-pyasn1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'python-pyasn1' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ sudo apt-get install -y --no-install-recommends git-core libffi-dev libssl-dev python-dev python-pip python-setuptools python-wheel whois
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'git' instead of 'git-core'
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
E: Unable to locate package python-wheel
[... 195 lines cut ...]
+ '[' production = production ']'
+ '[' false = false ']'
+ set +x
/dev/fd/63: line 240: mkpasswd: command not found
The default raspberry pi password was not detected, continuing with installation...
++ git rev-parse --abbrev-ref HEAD
/dev/fd/63: line 259: git: command not found
++ git rev-parse --short HEAD
/dev/fd/63: line 259: git: command not found
++ lsb_release -a
No LSB modules are available.
+ echo -e 'Screenly version: @\nDistributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye'
+ '[' false = false ']'
+ set +x
Installation completed.
You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)
`
#### Environment

* Raspberry Pi Hardware Version: Zero W
* Raspberry Pi Network Setup: USB Ethernet Adapter
* Screenly OSE Version: Current production version as of 2/6/22

extremeskillz, 2022-02-08:
Same boat there. Trying to install on Bullseye on a fresh RPI3b+ Build. 

I see there a builds for buster but we need Bullseye too. Thanks!

NassimBentarka, 2022-02-09:
> Same boat there. Trying to install on Bullseye on a fresh RPI3b+ Build.
> 
> I see there a builds for buster but we need Bullseye too. Thanks!

Thanks for mentioning that Bullseye is not yet supported. Just saved me tons of frustration, I just downgraded to Buster!

extremeskillz, 2022-02-12:
Yea, so as a public service announcement to all, anyone currently planning to install Screenly by the script please make sure your base PI install is Buster not Bullseye as of this date. Hopefully they have an update and updated script that supports Bullseye coming out soon as I would like to move to Bullseye to keep my PI systems current.

When the update is release that supports Bullseye this issue can be closed.

shaqaruden, 2022-02-14:
Python 2 is still required for Screenly.

semaf, 2022-03-20:
The installation script is not anyway compatible to Raspberry Pi 4 and Bullseye. Python2 still used, Docker and all requirements not installed correctly. Even a manually installation is not working very well.

vpetersson, 2022-09-23:
Closing this out as it has now been resolved.

closed 2022-09-23 by vpetersson