Screenly/Anthias #1571: Assets are not loading, screen keeps black

HOEMichelR, 2022-02-22:
#### Overview of the Issue
Screenly fresh installed on Raspberry Pi 4 B - Assets are not showing up
Screenly is starting with the Splash screen, I can load Screenly web interface, can add assets, but the screen still keeps black

#### Reproduction Steps
* Installed the latest Raspbian Buster Image 
* Installed developer branch from Screenly: $ bash <(curl -sL https://www.screenly.io/install-ose.sh)
* Considered all hints from #1389 especially @ealmonte32 

#### Environment
* Raspberry Pi Hardware Version: 4 B
* Raspberry Pi Network Setup: LAN

[image]

I hope, someone can help me?
Thanks in advance

vpetersson, 2022-02-22:
Is the time set correctly?

HOEMichelR, 2022-02-22:
Yes:
[image]

HOEMichelR, 2022-02-23:
Here is an actual log file created with journalctl:
[LogFile.txt](https://github.com/Screenly/screenly-ose/files/8122388/LogFile.txt)

Does Screenly have an ownl logfile itself?

HOEMichelR, 2022-02-23:
Update: I've tried the same installation routine with another Raspi 4B, but Rev. 1.1 -> it's working there, but with an older buster image (2021-01-11-raspios-buster-armhf-lite)

My Raspi with the problems is a 4B is Rev. 1.2 - on this one, the old buster image does not load:
[image]

So I took the latest one, but with this image I have above-mentioned problems.

ealmonte32, 2022-02-24:
@HOEMichelR 
Ok, I see..

Well, I tested with the last buster version and end result worked.
During initial installation, the following packages were not installed and are required, which you can manually install even before running the install script:
`python-backports.ssl-match-hostname`
`python-backports.shutil-get-terminal-size`

I also noticed `docker-compose` was missing so installed that too.

So flash the SD card with the version below and install these packages and then re-run of the installation which should be fast until it gets to the part that needs to build the containers, but it's a Pi4 so shouldn't be too bad. 

So, again, download the latest buster version directly: [2021-05-07 raspios buster 64-bit lite](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip)
Then follow the instructions above. If your Revision 1.2 Pi4B only functions with newer bootloader, we can even copy and paste the start*.elf files from the newer bullseye onto a buster SD card, but lets not do that type of workaround just yet, I also need to figure out why Bullseye is not installing properly and see if it's just installing the specific legacy packages that it needs and setting python2 as default interpreter.

ealmonte32, 2022-02-24:
Forgot to mention, if you do get it to load assets and everything but still have a black screen, you need to show the specific containers logs, not the one you sent me.
To get the container logs type:
`docker container logs screenly_srly-ose-viewer_1`

and send a screenshot of the results or text.
since your web server is running and allowing you to add assets, etc, that means the server container is working so no need to get those logs unless the reason why it is black is because you are putting some pictures that might throw an error.

Does the screen show the default assets? Turn them on and test them to make sure that we isolate the problem.

[image]

when you get the logs also give the logs for the server just in case, even though I think is the viewer which will show us the reason for black screen:
`docker container logs screenly_srly-ose-server_1`

HOEMichelR, 2022-02-24:
Thank you very much @ealmonte32 for your great support, but I couldn't establish a running version of Screenly with your instructions. This time, Screenly was not installed entirely, because there was no web server at all?
Pi starts and after a about 10 minutes, screen goes black and the connection is lost (still with SSH no connection)

I followed your instructions:
1) Took the image 2021-05-07-raspios-buster-arm64-lite and the Raspi startet without any problems
2) Did $ sudo apt update and $ sudo apt upgrade
3) Installed following packages
    $ sudo apt-get install python-backports.ssl-match-hostname
    $ sudo apt-get install python-backports.shutil-get-terminal-size
    $ sudo apt-get install docker-compose
4) Installed Screenly (developer branch)
5) Did a restart, and Pi stopped at login
5) $ sudo plymouth-set-default-theme -R screenly -> comes with following error:
[image]

Pictures during install:
[image]
[image]

Trying to get Screenly log-files without success:
[image]

journalctl:
[LogFiles.txt](https://github.com/Screenly/screenly-ose/files/8133156/LogFiles.txt)

By the way: Within the log files I see following: Machine model: **Raspberry Pi 4 Model B Rev 1.5**
[image]
[image]

ealmonte32, 2022-02-24:
@HOEMichelR 

Hmm.. I think we just need to get the right steps in place because the only reason I mentioned some of those install commands was in case you get the errors I got and it did not hurt to install them, but I also wonder why you are trying to run the plymouth theme command? you dont need to do any of that, those were some troubleshooting commands when we were trying to figure out Pi4 issues with plymouth theme not loading, but that was resolved with simply the script reinstalling the plymouth package at end of complete installation.

When the pi started up properly right after the installation of the lite image, you normally should run `sudo raspi-config` to configure your pi for things like locale and memory, etc. Then you run the bash command with install script.

That fatal error on the ansible task "docker-ce" is a similar error I saw on one of my first tries with recent installs, although not that message you got about it being unavailable, but you need this package, so try `sudo apt install docker-ce -y` manually and then if it installs run the bash install script again, this should work and the docker commands would then also work, and the containers would be built.

Now that we know we can get the buster lite working on that Pi4, we know it's just a matter of properly installing all the required packages before the Screenly is operating normally.
Let me know when you finish these steps.

HOEMichelR, 2022-02-25:
Thanks @ealmonte32 for your patience, but I'm less than a linux beginner ;-)

1) Took the image "2021-05-07 raspios buster 64-bit lite" and installed it
2) Installed "docker-compose" and "docker-ce" as you mentioned above
3) Installed Screenly via script $ bash <(curl -sL https://www.screenly.io/install-ose.sh)

Same result, no Screenly server/viewer are working. I tried it with/without docker, but with the same result.

How can I provide the installation output/logs to you? I can't scroll up or copy all the text.
But I've made a video: https://1drv.ms/v/s!AqlgVE4L_3TngXfS6KLeWbLZuuYQ?e=bKU7Yv

HOEMichelR, 2022-02-25:
Some additional information:
Web-Server was installed after all, because I can open Screenly via IP-Address remotely, but on the Raspi, Screenly is not started, only the linux shell is visible.

When I click to System Info:
[image]

I only get an error:
[image]

HOEMichelR, 2022-02-25:
Here are the log files from Screenly (installed with raspios-buster-armhf-lite - the one I originally used)
[Server.txt](https://github.com/Screenly/screenly-ose/files/8141826/Server.txt)
[Viewer.txt](https://github.com/Screenly/screenly-ose/files/8141828/Viewer.txt)

ealmonte32, 2022-02-25:
After watching your video I see two issues, the docker-ce says unavailable when trying to install from ansible, which is a problem in itself, but you kind of solve that by manually installing it prior to ansible running.

[image]

But, I think the main reason your viewer keeps failing is because I see your errors say python3, Screenly was coded for python2, and even though some python3 commands/code work the same as python2, you will not be able to run Screenly with python3 being the default, not sure how your system defaulted to this. Other users that have reported this mention that they installed Raspbian OS bullseye, but I know that you installed buster so that wouldn't be the reason...

Do me a favor and type this in:
`sudo update-alternatives --config python`

See if you can make it default to python2.7, and then run this file:
`./screenly/bin/upgrade_containers.sh`
 
One of the errors you can see is the docker container trying to be created for screenly viewer but it fails with error.

[image]

HOEMichelR, 2022-02-25:
I get the message, that there are no alternatives for pyhton

But, when I type **python --version**, I get the result: Python 2.7.16

HOEMichelR, 2022-02-25:
I have 3 folders in /usr/lib:

- python2.7
- python3
- python3.7

HOEMichelR, 2022-02-25:
When I take my second image (32 bit raspios-buster-armhf-lite) there is the same python version installed and also these 3 folders are existing; recreating the Screenly containers works there:
[image]
But still not Viewer with assets is visible

ealmonte32, 2022-02-25:
Ok, so the one where the containers were properly created doesnt give you python3 errors or docker-ce ones, ok good, now we can check why the viewer is still black.

First thing is, make sure you enable default assets on the Settings area of the Screenly web GUI. Then, in the command line type `docker container logs screenly_srly-ose-viewer_1`
You can press tab for auto-completion when typing commands.

Also, you can run `docker container ls -a` to show me the list of containers running or their status, because if they are not running then you wont see any logs.

HOEMichelR, 2022-02-25:
Viewer logs: [Viewer.txt](https://github.com/Screenly/screenly-ose/files/8143607/Viewer.txt)
There seems to be type mismatches 

List of containers: Viewer seems to start again and again:
[image]

ealmonte32, 2022-02-25:
The error about model number not found on the viewer logs would just return a Unknown Pi revision, that should not cause the error to prevent the viewer from starting, the actual reason why the script is not fully executing is the TypeError string integer one, and this usually means the line is being interpreted by a python version that doesnt take the code that way which is what leads me to believe python3 is trying to execute the entry.sh file rather than python2.7... but that requires some thorough troubleshooting.. I wonder if you remove python3 and all python3 packages, and reboot, to see if it defaults to that and all scripts run as that and that for now is your workaround..

HOEMichelR, 2022-02-25:
How do I remove python3 cleanly?

ealmonte32, 2022-02-25:
If you want to try and see if just adding your board revision manually would allow the entry.sh to run, you can easily do so by editing the raspberry helper file and adding your board version..

I write this just in case you want to learn to do this and are willing to try.

```
How do I SSH into a running container

    Use docker ps to get the name of the existing container.
    Use the command docker exec -it <container name> /bin/bash to get a bash shell in the container.
    Generically, use docker exec -it <container name> <command> to execute whatever command you specify in the container.
```

by reading the above, you should get an idea about what it is doing, basically you cant edit the `lib/raspberry_pi_helper.py` file outside of the container because it wouldnt make a change to the container that was already created/built and running, so what you want to do is get into that container and edit the file, save the file, close it, exit the container, and restart it.

so.. first, we go into the container:
`docker exec -it screenly_srly-ose-viewer_1 bash`

this gets you to a shell inside the container, its ok if its running.
then, while in the container you should get a command line shell#, in there we will use text editor nano to make the changes to the file.
but first we need to install nano cause its easier for me to explain vs using Vi.
so install nano by running: `apt install nano -y`

then when it finishes, run:
`nano ./lib/raspberry_pi_helper.py`

then, scroll down all the way to the lines where you see the board versions and revisions..
now, since you dont really care for it to be the right info of the pi and this is just a test, you can go to any of the currently written models and change the board version from one of the ones that say `Model 4B` and change the board number on the left from the one `b03114` to `b03115`, so you're basically telling the program that if it reads the board version b03115 which is the board you have, to just say it is a Model 4B, again we are just doing this as a test so that the helper file doesnt get an unknown model version, which again I dont think is the main error, but doesnt hurt to test.
Once you make that change, press `Ctrl+X` and then press `Y` then enter, this should save the file in the current container.
when you pressed enter you should be back at the command prompt/shell with #.
Now you can exit by typing `exit` and pressing enter.

Then, back at the regular pi user shell, restart the container:
`docker container restart screenly_srly-ose-viewer_1` press enter.

what are the results?

ealmonte32, 2022-02-25:
> How do I remove python3 cleanly?

run: `sudo apt remove python3 -y`

ealmonte32, 2022-02-25:
One problem might be encountered by removing python3 though, there is a possibility some dist-packages might be linked and removed.. so don't do that first, do the editing of the board revision thing first.. that way if that error comes back in the viewer container logs about TypeError then we know it isn't because of the new board not being recognized.

HOEMichelR, 2022-02-25:
docker exec -it screenly_srly-ose-viewer_1 bash throws an error: Container is restarting, wait until the container is running

HOEMichelR, 2022-02-25:
removing python3 also did not solve the problem :-(

ealmonte32, 2022-02-25:
> docker exec -it screenly_srly-ose-viewer_1 bash throws an error: Container is restarting, wait until the container is running

So then viewer container is not even built, so the changes can be made in the actual ./lib/raspberry_pi_helper.py file while you are logged on as pi in the regular shell..
that way, when you build containers, it will be created with this revision already in them..

ok so the same instructions about editing the ./lib/raspberry_pi_helper.py file, but this time you need to edit it from within pi:

at the pi shell, edit:
`nano ./screenly/lib/raspberry_pi_helper.py`

and then when done and saved, run the `./screenly/bin/upgrade_containers.sh` script

ealmonte32, 2022-02-25:
I mean the container is built but it cant run/start since the error prevents it from starting properly, so make the changes that will then recreate the container with proper board revision in the file.

HOEMichelR, 2022-02-25:
Did as you told me, but the container **screenly_srly-ose-viewer_1** is still restarting

HOEMichelR, 2022-02-27:
I understand the error message as follows: it expects an integer instead of a string
[image]
But why should this error occur with Pi in revision 1.5 and not in revision 1.4? The python code remains the same.
I also added the new revision `b03115` with no effect
[image]

@ealmonte32 I also tried to update the code file in viewer.py from 
`media_player = VLCMediaPlayer() if get_raspberry_model(get_raspberry_code()) == 'Model 4B' else OMXMediaPlayer()`
to
`media_player = VLCMediaPlayer()`

But that has no effect. I'm not a programmer, but I think the code file has to be compiled freshly.

Now I'd need help from professionals?

HOEMichelR, 2022-02-27:
@nadinespiesse I can agree with you
I did a setup on the Pi 4 and put the micro SD card into an older Pi 2 and Screenly was working properly.
Vice versa, I did a setup on Pi 2 and put the SD-card into Pi 4 rev. 1.5 and Screenly does not work - respectively it throws the above error

ealmonte32, 2022-02-28:
With regards to the code, you're correct, the code doesn't change so some package is seeing the revision 1.5 of the board and is not happy.. remember all of the underlying packages each have their supported hardware, etc.

Update your RPi firmware: `sudo rpi-update`
Just to be on the latest kernel.

Yes you need to rebuild the containers if changing code for permanent fix, but for testing you can change the code within the container like I showed you using `docker exec -it container-name bash` and installing nano text editor for editing the files (`sudo apt install nano -y`) and saving the file and restarting the container (`docker container restart container-name`)

For example I make a change in the raspberry pi helper python file of the server container, and I restart it and the changes take effect:

[image]
[image]

But, the above won't work for containers that are not started, so since yours keeps restarting, you need to edit the files before they get copied into the src folder as the app portion of container when built.

So the process is that each container copies source files from the screenly repo that gets downloaded to the pi user's home folder, so if you edit any of the files in the `./screenly/` directory after the fact, you need to rebuild the containers by force, the `upgrade_containers.sh` file does not force a copy over of source files and rebuilds the container, for that you need to add the --build to the script at the end of the `pull -d` line, so make it `pull -d --build`

So, lets say that the change you made of making Screenly only use OMXPlayer and not VLC for RPI4, then that means you have discovered an underlying package issue with VLC for example, again just being very basic in terms of the issue.

But, I think that we need to add your board revision to the repo file, and rebuild the containers, that way all of the sources will copy over to the containers with the correct model version for your board.

So, make the changes like you did for `b03115` but do them in the pi users home:
`nano ./screenly/lib/raspberry_pi_helper.py`

then run the edited upgrade_containers.sh script with the added --build on it:
[image]

I am troubleshooting this as I go with you and I am getting an error we should not be getting, which is that celery is version 5.2.2 but that is only for python3, we should be in versions 4.x, so I will create a PR for this but until that is approved and changed and tested, you can edit the requirements.txt file and make the changes yourself manually or else the upgrade_containers.sh script will give you the error I got about celery 5.2.2 and urllib3 being 1.26.x
Make celery 4.4.7 and urllib3 1.25.11 in `./screenly/requirements/requirements.txt`

[image]

Save it and run the edited upgrade_containers.sh script again.
Tell me what errors you see as it rebuilds the containers.

HOEMichelR, 2022-02-28:
@ealmonte32: you are a **hero** - thank you sooo much! I appreciate your continuous help :-)

After adding the new model `b03115` and rebuilding the containers according to your instructions, Screenly works like a charm.
Also the **System Info** shows valid data now.

For my understanding: currently, this is the temporary solution, until you do the changes on the original source, test and approve it?

ealmonte32, 2022-02-28:
@HOEMichelR 
Yes this is temporary but @vpetersson  will approve the PR when he has some time and then all users with rev 1.5 will not run into the issue.

ealmonte32, 2022-03-01:
@HOEMichelR 
The updated board revision has been merged to the master branch so you can close this issue.
I want to try and clean up all these issues that are stale and/or solved.

ealmonte32, 2022-03-01:
You got the same raspberry pi model error again? 
This is weird since the revision is already in the merged branch and if you did a fresh install today after the file was merged and confirmed that you saw it in the directory, it doesn't make sense that you would get the model error... hmm..

So this model error you just sent a screenshot of, is that from running `docker logs screenly_srly-ose-viewer_1` ?

Also, to confirm, when you say fresh install, you literally flashed the SD card again with the raspbian lite buster image? 

With regards to the celery and urllib3, I double checked and urllib3 is compatible with python2.7 so that can stay, I will do a dependency check again because I dont understand why it works fine under a different Pi without making any requirements changes, unless you had flashed a SD card with a pre-built screenly image instead of using the bash install script..

HOEMichelR, 2022-03-01:
I took following image `2021-05-07-raspios-buster-armhf-lite.img`, like I did x times before, flashed the SD-card,
did necessary settings with `sudo raspi-config`
started `sudo apt update` and `sudo apt upgrade`
and then used the bash script (developer branch) and did not take a pre-built image

Current Log files from `docker logs screenly_srly-ose-viewer_1 and `docker container ls -a`, no screenshots:
`[ViewerLogs.txt](https://github.com/Screenly/screenly-ose/files/8163870/ViewerLogs.txt)
[Container.txt](https://github.com/Screenly/screenly-ose/files/8163872/Container.txt)

HOEMichelR, 2022-03-01:
After doing your previous adviced changes, Screenly works like a charm again.

- added **--build** to `upgrade_containers.sh`  
- changed celery to 4.4.7 and urllib3 to 1.25.11 in `./screenly/requirements/requirements.txt`

Then started `upgrade_containers.sh` and there it is!

ealmonte32, 2022-03-01:
Thanks for the update.

I am doing further testing to see why the current master branch worked with celery 5.5.2 and urllib3 1.26.5 when I previously did an install on Pi3B. It doesn't make sense because the buster lite base OS is the same, so the packages that get installed on the containers should have been the same also and if incompatible I should have gotten the same error as you.. will keep you posted.

ealmonte32, 2022-03-02:
@HOEMichelR 

Ok so based on the troubleshooting I have done so far, while having celery being 5.2.2 on the requirements.txt actually installs 4.4.7 if you use Pi3 (from all my tests).

I checked every container and `pip freeze` displays celery as 4.4.7 even after the requirements.txt is at 5.2.2, so I would need to figure out what part (if any) the hardware plays into these packages being installed differently while the installation code and requirements file being the exact same..

I am leaning towards something to do with the way the wheels were configured to install for celery, because the [piwheels](https://piwheels.org/simple/celery/) celery repo shows that 4.4.7 has the py2.py3 tags while 5.2.2 has only py3 tag..

I think based on what the [changelogs](https://pyup.io/changelogs/celery/) for celery show, if pip is installing the package and it is told celery==5.2.2 then based on the removed backwards compatibility and installation support deprecated for python2, it should not downgrade itself to the supported version unless it is programmed this way in the wheel..

<br>

[image]

semaf, 2022-03-18:
There is a new Raspberry Pi 4 8GB Rev 1.5 (d03115) and it is missing in `./screenly/lib/raspberry_pi_helper.py`.

Hardware: BCM2711
Revision: d03115
Model: Raspberry Pi 4 Model B Rev 1.5

I have the same issue that after installation nothing is shown in screen. Followed here the steps until here https://github.com/Screenly/screenly-ose/issues/1571#issuecomment-1053950112 which gave me following messages:

```
pi@raspberrypi:~ $ sudo -E docker-compose \
> -f /home/pi/screenly/docker-compose.yml \
> -f /home/pi/screenly/docker-compose.override.yml \
> up -d --build
WARNING: The DOCKER_TAG variable is not set. Defaulting to a blank string.
WARNING: The MY_IP variable is not set. Defaulting to a blank string.
WARNING: The DEVICE_TYPE variable is not set. Defaulting to a blank string.
WARNING: The VIEWER_MEMORY_LIMIT_KB variable is not set. Defaulting to a blank string.
Building srly-ose-server
ERROR: Couldn't connect to Docker daemon at http+docker://localhost - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.
```

Changed celery 4.4.7 and urllib3 1.25.11 too and started `upgrade_containers.sh`

ealmonte32, 2022-03-20:
> There is a new Raspberry Pi 4 8GB Rev 1.5 (d03115) and it is missed in `./screenly/lib/raspberry_pi_helper.py`.
> 
> ```
> Hardware: BCM2711
> Revision: d03115
> Model: Raspberry Pi 4 Model B Rev 1.5
> ```
> 
> Using:
> 
> ```
> Raspberry Pi OS Lite
> Release date: January 28th 2022
> System: 32-bit
> Kernel version: 5.10
> Debian version: 11 (bullseye)
> ```
> 
> Changed celery 4.4.7 and urllib3 1.25.11 too and started `upgrade_containers.sh`
> 
> and getting the error
> 
> ```
> Traceback (most recent call last):
>   File "viewer.py", line 50, in <module>
>     media_player = VLCMediaPlayer() if get_raspberry_model(get_raspberry_code()) == 'Model 4B' else OMXMediaPlayer()
>   File "/usr/src/app/lib/diagnostics.py", line 128, in get_raspberry_model
>     return raspberry_pi_helper.lookup_raspberry_pi_revision(raspberry_pi_revision)['model']
> TypeError: string indices must be integers, not str
> ./bin/start_viewer.sh: line 47: kill: (37) - No such process
> Traceback (most recent call last):
>   File "viewer.py", line 50, in <module>
>     media_player = VLCMediaPlayer() if get_raspberry_model(get_raspberry_code()) == 'Model 4B' else OMXMediaPlayer()
>   File "/usr/src/app/lib/diagnostics.py", line 128, in get_raspberry_model
>     return raspberry_pi_helper.lookup_raspberry_pi_revision(raspberry_pi_revision)['model']
> TypeError: string indices must be integers, not str
> ./bin/start_viewer.sh: line 47: kill: (37) - No such process
> Traceback (most recent call last):
>   File "viewer.py", line 50, in <module>
>     media_player = VLCMediaPlayer() if get_raspberry_model(get_raspberry_code()) == 'Model 4B' else OMXMediaPlayer()
>   File "/usr/src/app/lib/diagnostics.py", line 128, in get_raspberry_model
>     return raspberry_pi_helper.lookup_raspberry_pi_revision(raspberry_pi_revision)['model']
> TypeError: string indices must be integers, not str
> ./bin/start_viewer.sh: line 47: kill: (37) - No such process
> ```

@semaf 
Bullseye is not supported, you must use raspbian buster lite.
Also, if you do make the changes for the new board, please see the comments [here](https://github.com/Screenly/screenly-ose/issues/1571#issuecomment-1055781436) on how to edit the `upgrade_containers.sh` file in order for the containers to be rebuilt with the new addition of the board or else it wont work.
When you read the comments, make your way backwards so you can see where you need to add the `--build` parameter..

Scott-Rose, 2022-04-01:
Are these the steps you took?

- New installation of Buster Lite
- sudo apt update
- sudo apt install git
- sudo apt-get install docker-compose
- bash <(curl -sL https://www.screenly.io/install-ose.sh)
- Development branch
- No system upgrade
- Y to manage network
- Add --build to the end of screenly/bin/upgrade_containers.sh
	so that it reads "up -d --build"
	Ctrl-O, Enter, Ctrl-X
- Edit the file screenly/requirements/requirements.txt
	Change celery to 4.4.7
	Change urllib3 to 1.25.11
	Ctrl-O, Enter, Ctrl-X
- sudo ./screenly/bin/upgrade_containers.sh

I followed these steps on my Pi4B and Screenly launches but assets just load as a blank white screen.

Is there any chance you are willing to provide an disk image of your installation?

Scott-Rose, 2022-04-01:
A quick update,

Screenly seems to accept and display content fine, but only assets that are the same resolution of the display.

That would be fine, but it is also very pixelated, especially along colour gradients. Even the splash screen is pixelated in this way.

This seems to be the same issue as https://forums.screenly.io/t/image-pixelation/570

Any ideas?

semaf, 2022-04-01:
> Are these the steps you took?
>
>
>
> - New installation of Buster Lite
>
> - sudo apt update
>
> - sudo apt install git
>
> - sudo apt-get install docker-compose
>
> - bash <(curl -sL [...]

Which Pi revision do you have?
After you start screenly script, click Y, 2, Y, Y.

Edit files and restart installation with bash <(curl -sL https://www.screenly.io/install-ose.sh)

Once you finish type command docker and docker-compose to check if any errors available.

HOEMichelR, 2022-04-01:
These were my installation steps:

-) Took following buster image (previous one): https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip
-) update and upgrade OS `sudo apt update` `sudo apt upgrade`
-) install Screenly `bash <(curl -sL https://www.screenly.io/install-ose.sh)`
    Y, 2, N, N
    Some errors occured, so I did this step twice, afterwards all containers were created successfully
-) add missing model, if yours is not already in this file `$ nano ./screenly/lib/raspberry_pi_helper.py`
-) add --build at the end after up -d `sudo nano ./screenly/bin/upgrade_containers.sh`
-) change versions `sudo nano ./screenly/requirements/requirements.txt`
    celery  -> 4.4.7
    urllib3 -> 1.25.11
-) rebuild containers `./screenly/bin/upgrade_containers.sh`

That's it - afterward, Screenly is working like a charm

semaf, 2022-04-01:
My steps exactly was

-) Took following buster image (previous one): https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip
-) update and upgrade OS sudo apt update sudo apt upgrade
-) install Screenly bash <(curl -sL https://www.screenly.io/install-ose.sh)
Y, 2, N, N
Some errors occured but ignored
-) add missing model, if yours is not already in this file $ nano ./screenly/lib/raspberry_pi_helper.py
-) add --build at the end after up -d sudo nano ./screenly/bin/upgrade_containers.sh
-) change versions sudo nano ./screenly/requirements/requirements.txt
celery -> 4.4.7
urllib3 -> 1.25.11
-) Install screenly again with Y-2-Y-Y bash <(curl -sL https://www.screenly.io/install-ose.sh)
-) type docker and docker-compose in command line to see if any errors appear. If not restart and use it

Scott-Rose, 2022-04-04:
Thanks heaps @HOEMichelR and @semaf for those detailed instructions, and of course @ealmonte32 for your troubleshooting. 

My revision of the Pi (b03115) was already in the raspberry_pi_helper.py file. I got my Screenly up and running but still had issues with colour banding on assets with gradients. I installed fbi to view some images independently of Screenly and found the issue was not only within Screenly.

I was tipped off by [this reddit post](https://www.reddit.com/r/raspberry_pi/comments/l193vh/can_anyone_confirm_to_me_that_the_rpi4b_can_only/), and after commenting out `dtoverlay=vc4-fkms-v3d` in the RPi's `config.txt` file my colour banding issues disappeared.

purebordem, 2022-04-06:
Fwiw I have been having the same issues and was able to remedy it via similar steps to @HOEMichelR and @semaf. My steps were...

`sudo apt update`
`sudo apt upgrade -y`
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
- y, 2, n, n
`sudo nano ./screenly/bin/upgrade_containers.sh`
add `--build` at the end of file after up `-d` on the same line
`sudo nano ./screenly/requirements/requirements.txt`
- Change `celery` -> 4.4.7
- Change `urllib3` -> 1.25.11
`sudo apt install docker-compose -y`
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
- y, 2, n, n
`reboot`
`sudo apt install docker-ce -y`
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
- y, 2, n, n
`reboot`

semaf, 2022-05-01:
> Fwiw I have been having the same issues and was able to remedy it via similar steps to @HOEMichelR and @semaf. My steps were...
>
> `sudo apt update` `sudo apt upgrade -y` `bash <(curl -sL [...]

You need to upgrade the containers after "--build" has been added and the versions of requirements changed.

ealmonte32, 2023-02-08:
@HOEMichelR 
Please close this since we know running `./screenly/bin/upgrade_containers.sh` fixes this for the rare times that somehow the containers dont get created during bash install.

HOEMichelR, 2023-02-10:
@ealmonte32 thank you - I completely forgot; 
will do it immediately

closed 2023-02-10 by HOEMichelR