Screenly/Anthias #1578: Balena Build failed: "Error: The command '/bin/sh -c pip install --no-cache-dir -r /tmp/requirements.txt' returned a non-zero code: 1 [Info]"

matrover, 2022-03-28:
#### Overview of the Issue

Tryed to deploy with balena by clicking on the button in the readme, tryed different fleets (RaspPI4, RaspPI3 32bit and 64bit) 
Build log: 

```
[Info]                Starting build for screenly-ose2, user XXXX
[Info]                Dashboard link: https://dashboard.balena-cloud.com/apps/1919298/devices
[Info]                Building on arm03
[Info]                No suitable previous release for caching, skipping cache pull
[Info]                Starting to build srly-ose-nginx
[Info]                Starting to build redis
[Info]                Starting to build srly-ose-server
[Info]                Starting to build srly-ose-websocket
[Info]                Starting to build srly-ose-viewer
[Info]                Starting to build srly-ose-celery
[Success]             Successfully built service srly-ose-websocket
[Success]             Successfully built service redis
[Success]             Successfully built service srly-ose-nginx
[Success]             Successfully built service srly-ose-viewer
[Info]                Uploading images
[Success]             Successfully uploaded images
[Error]               Some services failed to build:
[Error]                 Service: srly-ose-server
[Error]                   Error: The command '/bin/sh -c pip install --no-cache-dir -r /tmp/requirements.txt' returned a non-zero code: 1
[Error]                 Service: srly-ose-celery
[Error]                   Error: The command '/bin/sh -c pip install --no-cache-dir -r /tmp/requirements.txt' returned a non-zero code: 1
[Info]                Built on arm03
[Error]               Not deploying release.
```

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Click on deploy with balena 
* Select fleet 
* wait for build completion (failure)

#### Environment

* Raspberry Pi Hardware Version: 3 and 4 (32bit and 64bit)
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: latest (2cc041e)

matrover, 2022-03-31:
Found the problem, it cant find the celery version 5.2.2. 

The latest it can find is 4.4.7

```
Collecting celery==5.2.2 (from -r /tmp/requirements.txt (line 9))
  Could not find a version that satisfies the requirement celery==5.2.2 (from -r /tmp/requirements.txt (line 9)) (from versions: 0.1.2, 0.1.4, 0.1.6, 0.1.7, 0.1.8, 0.1.10, 0.1.11, 0.1.12, 0.1.13, 0.1.14, 0.1.15, 0.2.0, 0.3.0, 0.3.7, 0.3.20, 0.4.0, 0.4.1, 0.6.0, 0.8.0, 0.8.1, 0.8.2, 0.8.3, 0.8.4, 1.0.0, 1.0.1, 1.0.2, 1.0.3, 1.0.4, 1.0.5, 1.0.6, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.1.4, 2.2.0, 2.2.1, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 2.2.6, 2.2.7, 2.2.8, 2.2.9, 2.2.10, 2.3.0, 2.3.1, 2.3.2, 2.3.3, 2.3.4, 2.3.5, 2.4.0, 2.4.1, 2.4.2, 2.4.3, 2.4.4, 2.4.5, 2.4.6, 2.4.7, 2.5.0, 2.5.1, 2.5.2, 2.5.3, 2.5.5, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.0.4, 3.0.5, 3.0.6, 3.0.7, 3.0.8, 3.0.9, 3.0.10, 3.0.11, 3.0.12, 3.0.13, 3.0.14, 3.0.15, 3.0.16, 3.0.17, 3.0.18, 3.0.19, 3.0.20, 3.0.21, 3.0.22, 3.0.23, 3.0.24, 3.0.25, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.1.4, 3.1.5, 3.1.6, 3.1.7, 3.1.8, 3.1.9, 3.1.10, 3.1.11, 3.1.12, 3.1.13, 3.1.14, 3.1.15, 3.1.16, 3.1.17, 3.1.18, 3.1.19, 3.1.20, 3.1.21, 3.1.22, 3.1.23, 3.1.24, 3.1.25, 3.1.26.post1, 3.1.26.post2, 4.0.0rc3, 4.0.0rc4, 4.0.0rc5, 4.0.0rc6, 4.0.0rc7, 4.0.0, 4.0.1, 4.0.2, 4.1.0, 4.1.1, 4.2.0rc1, 4.2.0rc2, 4.2.0rc3, 4.2.0rc4, 4.2.0, 4.2.1, 4.2.2, 4.3.0rc1, 4.3.0rc2, 4.3.0rc3, 4.3.0, 4.3.1, 4.4.0rc1, 4.4.0rc2, 4.4.0rc3, 4.4.0rc4, 4.4.0rc5, 4.4.0, 4.4.1, 4.4.2, 4.4.3, 4.4.4, 4.4.5, 4.4.6, 4.4.7)

No matching distribution found for celery==5.2.2 (from -r /tmp/requirements.txt (line 9)) 
```

rmmcdr, 2022-05-12:
Hi matrover!
I've the same problem. And what did you do to solve or avoid it?

Black8all, 2022-06-09:
Does anyone have a solution for this problem? I'm trying to deploy Screenly with Balena and I get the same error.

matrover, 2022-06-10:
@rmmcdr & @Black8all i ended up removing the minimal version of celery in the requirements files and it worked!

moorsey, 2022-06-14:
Many thanks @matrover for the fix on this one, worked for me

rmmcdr, 2022-06-22:
Thanks a lot@matrover !! It works perfectly 👍