Screenly/Anthias #1580: Screenly bootscreen shows, then blackscreen Raspberry Pi 4B

sl8tr, 2022-04-06:
Hey!

I’m really stuck with my Raspberry 4 installation.

So I’ve installed the developers build with no problems, but after a restart the screenly bootscreen is shown and then it switches to a black screen. Only ALT + F1 is working and gets me into the terminal.

I’ve deleted the /home/pi/screenly folder and reran the installation. Still not working.
Started the Installation several times. 

I used the Pi imager and preset for Pi OS Lite (Legacy) with Debian Buster and also downloaded and installed an older one.

I hope you can help me out. I have to get 2 of them done for a company. :/

carlesjavierre, 2022-04-25:
Same issue here, 4 Pi's in a row. Will try investigate further. I have installed one a month ago and it worked correctly. (Yes, with the rbp4)

carlesjavierre, 2022-04-25:
I have tried the new installs I made with the "old" pi that I had working with its own install before: THEY WORK. However, the newer hardware I bought doesn't. 

The board that is working has the following:

HW version: BCM2711
Revision: b03114
Model: RBP 4B Rev 1.4

The board not working:
HW version: BCM2711
Revision: b03115
Model: RBP 4B Rev 1.5

However, it seems that the 1.5 revision is correctly added to the list on [here](https://github.com/Screenly/screenly-ose/blob/master/lib/raspberry_pi_helper.py#L144)

I hope this helps debugging.

paFFr, 2022-05-05:
We've encountered the same issue, Raspberry Pi 4B Rev 1.5, using the Default Buster Lite image and doing a manual installation there won't be any errors during the installation, but on reboot it will show just the Splash Screen and switches to a black screen afterwards. As carlesjavierre wrote the WebUI will be reachable but going to system info will show an internal error.

I also tried to comment out line 49-51 on viewer.py but this sadly didn't change anything in my case.

I have no idea how to get this to work anymore.

carlesjavierre, 2022-05-05:
@paFFr it requires that you edit the docker-compose so it replaces the file inside the container (that'ts why I said i did some fiddly stuff). I'll edit the comment later on so its full, as it might help others. I just didn't want to post a nasty patch, but if you can get it to work its fine for me!

carlesjavierre, 2022-05-05:
@paFFr I have updated with full instructions. Please try it out and let me know :smile:

paFFr, 2022-05-05:
@carlesjavierre Thank you for taking the effort and updating your instructions, sadly it didn't seem to change anything on my end, as I still get only a black screen after the initial splash screen :(

closed 2022-09-29 by nicomiguelino