Screenly/Anthias #1587: Can't build on balena

mmclane, 2022-06-30:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Can't build for balena any more.. 

When running ./bin/deploy_to_balena.sh with the latest balena cli you get an error:

 4:13PM: ~/repos/public/screenly-ose [master|✔] k8s:[main] > ./bin/deploy_to_balena.sh
Here are your Balena apps:
Note: the command "balena apps" was replaced in CLI version v13.0.0.
Please use "balena fleets" instead.

I upgraded to the latest version because I was getting other errors.

If I modify line 23 of ./bin/deploy_to_balena.sh and change it to balena fleets, I get further but still an error:

```
Here are your Balena apps:
ID      NAME         SLUG                    DEVICE TYPE  ONLINE DEVICES DEVICE COUNT
1946063 screenly     docnetwork/screenly     raspberry-pi 1              1
1880537 screenly-ose docnetwork/screenly-ose raspberrypi3 1              1

Enter the app name of the app you want to deploy to (needs to be SLUG): docnetwork/screenly
app is not a recognized balena command.

Did you mean: ? 
  api-key generate
  key

Run balena help -v for a list of available commands,
 or balena help <command> for detailed help on a specific command.

```

#### Reproduction Steps

Download balena cli v13.6.1 (latest)
run ./bin/deploy_to_balena.sh

#### Environment

* Raspberry Pi Hardware Version:
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
* Screenly OSE Version:

mmclane, 2022-07-13:
This is on a raspberry pi zero W.  

I have also tried to install screenly on the device without balena and have failed to get that to work too.

mmclane, 2022-07-13:
Oh, am I am using a monitor plugged in via HDMI currently.

vpetersson, 2022-09-13:
This should have been resolved in #1599

vpetersson, 2022-09-13:
It's unclear why, but the build phase of balena still doesn't work properly.

mmclane, 2022-09-19:
I am definitely still getting the same error.

vpetersson, 2022-09-21:
We are working on revamping how we support Balena. Stay tuned for updates, but in the meantime, we are now automatically deploying to Balena [as part of CI](https://github.com/Screenly/screenly-ose/blob/master/.github/workflows/docker-build.yaml#L106). You should be able to just mimic what we're doing there in the meantime.

nicomiguelino, 2024-06-21:
@mmclane, we have already revamped the [docs for deploying Anthias on your own Balena fleets/devices](https://github.com/Screenly/Anthias/blob/0fd729b019c61260059556c751dd328ec78fee77/docs/balena-fleet-deployment.md).

Please reopen this issue if you're still facing similar issues.

closed 2024-06-21 by nicomiguelino