Screenly/Anthias #1588: Internal Server Error when booting without network connection

MisterAnderson91, 2022-07-24:
#### Overview of the Issue

When booting latest dev build installed on latest buster-lite legacy image with no active network connection, a white screen appears which then after a pause says "Internal Server Error".

#### Reproduction Steps

Steps to reproduce this issue:

1. Flash card with latest official buster-lite image (labelled "legacy")
2. Attached network cable (with internet access)
3. Boot image and raspi-config and set all locale settings plus hostname.
4. sudo apt update && sudo apt upgrade
5. sudo apt install pip-python python-backports.ssl-match-hostname python-backports.shutil-get-terminal-size docker docker-compose
6. sudo pip install ansible==2.10.7
7. bash <(curl -sL https://screenly.io/install-ose.sh) - Y, 2, Y, N - Check for errors during script (none found, just warnings about deprecation of python 2)
8. Reboot, screenly will be working fine showing connection address for web interface, eventually fade to dark screen with Screenly logo
9. Disconnect cable and reboot, Screenly splash screen appears but when wifi-AP connection instructions should appear, get white screen instead with eventual error message which then cycles to black and reappears regularly.
10. Plug cable back in (or connect to wifi network with nmtui or similar) and wait, or reboot, and connection screen/black logo screen will reappear.

#### Environment

* Raspberry Pi Hardware Version: Pi 4B v1.4
* Raspberry Pi Network Setup: Tried with both ethernet and built-in wifi, same result
* Screenly OSE Version: Latest dev build as of this post

I believe this has been posted before but I could not find a post which completely matched these symptoms. It seems that whatever system controls the wifi-access-point-mode (a docker container?) is broken. It doesn't matter if network-manager was installed as an option or not.

I tried testing connecting to a wifi network that has no internet access (android phone hotspot with all data connections turned off), and this also gives the same "Internal Server Error" screen but the blank white screen right before the error message seems to be much shorter (maybe 5 seconds vs 20). If data connection is re-enabled on the phone (and therefore to the pi through hotspot) it will cycle back to the black screenly logo screen.

I don't know what reason there is for this to need a WAN connection but I presume it's a bug.

If anyone can provide a premade image that works on the Pi 4 series, that would be appreciated, even if it's an old build.

dsvinovacaogp, 2022-07-25:
I was able to use the version using docker by following these steps.

[Assets are not loading, screen keeps black](https://github.com/Screenly/screenly-ose/issues/1571#issuecomment-1090801374)

fbsanches, 2022-07-25:
I was able to use the version using docker by following these steps.

[Assets are not loading, screen keeps black](https://github.com/Screenly/screenly-ose/issues/1571#issuecomment-1090801374)

MisterAnderson91, 2022-07-27:
I tried the steps linked, however it introduces a new problem: instead of offering the wifi hotspot page or the internal server error page, it now always shows a connection page for managing content at the IP address in use at some point during the setup. It does not change even when rebooted with no connection, or an alternate connection with a confirmed-different IP address range.

There was an error during the part in those steps where it was running the upgrade-containers step. It complained about urllib3 being 1.26.5 even with the relevant change to requirements.txt

fbsanches, 2022-07-27:
@MisterAnderson91 Do you have any other raspberry to perform network tests? 

When I did the mentioned steps to run with docker, it worked like a glove for me. It ran normally and the containers went up normally.
I had a problem with a raspberry another time that its video chip was having problems and it was found to be defective. It didn't play videos at any time, but it was used for other activities and applications, except playing videos. In this case, if you can test with another rasp it would be ideal. 

Did you put --build in the upgrade_containers.sh file and change the celery version? I ran this method on 4gb and 8gb PI4s.

MisterAnderson91, 2022-07-27:
I follow the steps that were listed as linked. I'm not sure what playing videos has to do with the hotspot/management info page not work.

I seem to be getting extremely inconsistent results just running the install script at all, with random parts failing sometimes and then working others. I am consistently reimaging to run new tests, using the same buster-lite image (latest), with different USB sticks and SD cards to make sure they are not faulty. Sometimes ansible fails to install, sometimes python-backports must be installed manually, sometimes everything goes smoothly like this most recent test.

I am just now doing another fresh run, however instead of doing a full run of the install script, I ran it to the point that ansible was installed and screenly git was cloned, then cancelled it. Then I made the changes to requirements.txt and upgrade_containers.sh and restarted the script.

I manage to catch the exact error about urllib3:

"requests 2.24.0 has requirement urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1, but you'll have usrlib3 1.26.5 which is incompatible."

I don't know why I have 1.26.5 when I changed the requirements.txt to 1.25.11 but it's the only error I get during the install now.

It takes quite a long time to rebuild these containers and it's late so I will provide an update on if this method worked tomorrow.

MisterAnderson91, 2022-07-27:
> When I did the mentioned steps to run with docker, it worked like a glove for me. It ran normally and the containers went up normally.

Any chance you could grab an image of a working device and share it? It's very frustrating that the pre-made image hasn't been updated for 2 years and the install script was relied upon, only for it to quickly stop working. If the pre-made image worked on a pi 4, none of this would be an issue.

vpetersson, 2022-07-27:
We have finally made room in the budget to hire a developer to work fully on Screenly OSE. A job description will be posted shortly.

fbsanches, 2022-07-27:
It's frustrating. Seeing such a big project lose strength, but what great news @vpetersson . What version of your raspberry @MisterAnderson91 ?

MisterAnderson91, 2022-07-27:
Very exciting news! I hope everything gets back on track for the project.

I've been testing on both a pi 4B v1.4 and a pi 3B v1.2, however the pi 3B won't be available ongoing so I can't use the old premade image.

If I have any time I will continue to troubleshoot this, especially that error about it seemingly ignoring the urllib3 version requirement, but I may have to step away from this for a bit. I will close this issue now though since rebuilding the containers technically fixes it.

MisterAnderson91, 2022-07-27:
Solution that worked for me:

`sudo apt update`
`sudo apt upgrade`
`bash <(curl -sL https://www.screenly.io/install-ose.sh`
(Y, 2, Y, N)
Wait for ansible to install and screenly directory to be created in ~/ then CTRL-C to stop script.
`nano screenly/bin/upgrade_containers.sh`
Add `--build` to last line after `-d`
`nano screenly/requirements/requirements.txt`
Change celery to 4.4.8
Change urllib3 to 1.25.11
`bash <(curl -sL https://www.screenly.io/install-ose.sh
reboot`

Final system seems to work with or without internet access, but shows incorrect management IP address on screen and does not offer a hotspot when no connection detected.

closed 2022-07-27 by MisterAnderson91