Screenly/Anthias #1589: Failed reboot

fbsanches, 2022-07-28:
Reboot sometimes fails. Sometimes it reboots and sometimes it doesn't. I see in the celery logs that the command is received but not rebooted. Add sudo to the reboot class command line to test but it still doesn't work. Does anyone have any ideas to get around this and do a direct reboot from the host.

fbsanches, 2022-07-28:
I don't know if this helps, but the difficulty is sending the command from inside the container to the host?

Something like:
[[Docker - Rebooting raspberry pi host from within docker container](https://stackoverflow.com/questions/55349923/docker-rebooting-raspberry-pi-host-from-within-docker-container)](https://stackoverflow.com/questions/55349923/docker-rebooting-raspberry-pi-host-from-within-docker-container)

ealmonte32, 2022-11-23:
Hi @fbsanches 

Rebooting the host Pi from the web GUI is currently not supported since the containers should not be sending commands outside of themselves for security purposes, only where really necessary.
One of the main points of the containers is to isolate them from the host and contain them so that if anything crashes you can simply restart the container itself and not the whole Pi.
To restart the whole Pi device, simply SSH into it and run `sudo reboot`.

nicomiguelino, 2024-06-08:
@fbsanches, @ealmonte32
As of now, reboot and shutdown via the web UI is supported.
The reboot/shutdown Celery tasks doesn't get received on situations where the number of active tasks reaches the concurrency limit (which defaults to the number of CPUs of the device).

One possible fix is to set time limit for tasks that takes a while to finish, so that it wouldn't fill up the task queue. The `--concurrency` value can also be increased. (See [here](https://docs.celeryq.dev/en/stable/reference/cli.html#cmdoption-celery-worker-c) for details.)

nicomiguelino, 2024-06-08:
I'm closing this in favor of #1920.

closed 2024-06-08 by nicomiguelino