Screenly/Anthias #1590: MP4 not playing & crashing on RaspberryPi 4

Tissy, 2022-07-29:
#### Overview of the Issue

JPEGS play fine, however when playing an MP4 video, Screenly crashes and reboots.  Did not happen on RPI3.

#### Reproduction Steps

Add MP4 assest in normal way.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 2GB
* Raspberry Pi Network Setup: Connected via Ethernet and internal WiFi
* Screenly OSE Version:  Current from GitHub
'Load Average: 0.55
Free Space: 24G
Memory : Total: 1797 \| Used: 631 \| Free: 647 \| Shared: 63 \| Buff: 40 \| Available: 1078
Raspberry Pi Model: Model 4B (Sony UK)
Screenly OSE Version: @
`

Crash log attached below.

[digitaldisplay_bold-breeze-28.07.22_18_02_34_(+0100).txt](https://github.com/Screenly/screenly-ose/files/9218981/digitaldisplay_bold-breeze-28.07.22_18_02_34_.%2B0100.txt)

Many thanks,

vpetersson, 2022-09-23:
@Tissy I can confirm this. I've reproduced it, but not been able to resolve it. It's related to the `vlc` playback inside the viewer container.

vpetersson, 2022-09-23:
Some more logs:

```
./bin/start_viewer.sh: line 36: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
Home directory not accessible: Permission denied
[01762858] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Home directory not accessible: Permission denied
[01772900] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Loading browser...
Generating asset-list...
Current url is http://srly-ose-nginx:80/splash-page
Current url is http://srly-ose-nginx:80/static/img/loading.png
Showing asset https://stage-weather.srly.io (webpage)
Current url is https://stage-weather.srly.io
Sleeping for 10
Showing asset bbb_sunflower_2160p_30fps_normal.mp4 (video)
Current url is null
[017eb9e0] mmal_codec decoder: VCSM init succeeded: Legacy
```

miyako-ep, 2022-10-19:
I'm not sure of the details, but I can get mp4 to show up with the following settings:

- /boot/config.txt
```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d
# dtoverlay=vc4-kms-v3d
```

my environment:
- Raspberry Pi 4 model B
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```

Are the following related?
- https://forums.raspberrypi.com/viewtopic.php?t=338765

vpetersson, 2022-10-19:
Great lead. Yes, could indeed be related. Are you able to get this work with video on Screenly OSE too, or just using regular `vlc`?

nicomiguelino, 2022-10-19:
Hello @miyako-ep, I've applied changes on my Pi 4 (with Screenly OSE installed). I've restarted the Pi as well. However, there's still no video (just sounds).

Here's what I got from the logs...

```
screenly-srly-ose-viewer-1  | Generating asset-list...
screenly-srly-ose-viewer-1  | Showing asset First YouTube Video (video)
screenly-srly-ose-viewer-1  | [f4f1c638] mmal_codec decoder: VCSM init succeeded: CMA
screenly-srly-ose-viewer-1  | Current url is null
screenly-srly-ose-viewer-1  | [f4f1c638] main decoder error: buffer deadlock prevented
screenly-srly-ose-viewer-1  | error: XDG_RUNTIME_DIR not set in the environment.
screenly-srly-ose-viewer-1  | error: XDG_RUNTIME_DIR not set in the environment.
screenly-srly-ose-viewer-1  | [ef710c38] gles2 generic error: parent window not available
screenly-srly-ose-viewer-1  | [ef71e058] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
screenly-srly-ose-viewer-1  | [ef710c38] xcb generic error: window not available
screenly-srly-ose-viewer-1  | [ef71e058] mmal_xsplitter vout display error: Failed to open Xsplitter:xcb_x11 module
```

#### Environment

``` PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

miyako-ep, 2022-10-20:
I use Screenly OSE. With the above settings, I am able to play videos on Screenly OSE. Does this answer your question?
I just installed the OS (OS Lite), ran the installation commands (bash <(curl ...) , and set up the above.

@nicomiguelino 
Completely guessing, but if you can hear the sound, something in the video file doesn't sound good.
I thought the video is playing?

> screenly-srly-ose-viewer-1  | [f4f1c638] main decoder error: buffer deadlock prevented

I'm not getting this error in my environment, so maybe this is a clue.

miyako-ep, 2022-10-20:
By the way, I did a lot of research for the above response, so here are my notes on it. It had nothing to do with the video playback issue, though.  I'm sorry if it's none of my business.

> ./bin/start_viewer.sh: line 36: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory

It seems that the cgroup version has changed? Would `/sys/fs/cgroup/memory.swap.current` work?

> Home directory not accessible: Permission denied

The error disappeared when I did the following in `screenly-srly-ose-viewer-1` container:
```
# chown viewer:1000 /data
# chown -R viewer:1000 /data/.config/
```
Looks like vlc(python-vlc?) creates `/data/.config/pulse/*-runtime` file.

> [01762858] vlcpulse audio output error: PulseAudio server connection failure: Connection refused

I'm not sure about this one. But I could silence it with the following workaround (Will this lose sound?):
```
diff --git a/lib/media_player.py b/lib/media_player.py
index 91754bb..17f2c61 100644
--- a/lib/media_player.py
+++ b/lib/media_player.py
@@ -28,7 +28,8 @@ class MediaPlayer:
 class VLCMediaPlayer(MediaPlayer):
     def __init__(self):
         MediaPlayer.__init__(self)
-        self.instance = vlc.Instance()
+        opt = "--aout=adummy"
+        self.instance = vlc.Instance(opt)
         self.player = self.instance.media_player_new()

         self.player.audio_output_set('alsa')
```

PR #1640 (2022-10-31): Fix MP4 video playback issue.
Might fix #1590

#### Checklist

- [x] Modify relevant Ansible files so that /boot/config.txt can be updated as well.
- [ ] Test changes on Balena.

#### References

- https://github.com/RPi-Distro/vlc/issues/3
closed 2022-10-31 by nicomiguelino