Screenly/Anthias #1591: Unable to --build upgrade_containers.sh

MisterAnderson91, 2022-08-01:
#### Overview of the Issue

srly-ose-server failing to build.

When adding --build to upgrade-containters.sh as listed [here](https://github.com/Screenly/screenly-ose/issues/1588#issuecomment-1197350225), srly-ose-server now fails to build.

```
Error compiling Cython file:
    ------------------------------------------------------------
    ...
    cdef load_traceback
    cdef Waiter
    cdef wait
    cdef iwait
    cdef reraise
    cpdef GEVENT_CONFIG
          ^
    ------------------------------------------------------------
    
    src/gevent/_gevent_cgreenlet.pxd:182:6: Variables cannot be declared with 'cpdef'. Use 'cdef' instead.
    Compiling src/gevent/greenlet.py because it changed.
    [1/1] Cythonizing src/gevent/greenlet.py
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-VvolsI/gevent/setup.py", line 294, in <module>
        EXT_MODULES.append(cythonize1(mod))
      File "_setuputils.py", line 261, in cythonize1
        common_utility_include_dir=COMMON_UTILITY_INCLUDE_DIR,
      File "/tmp/pip-build-env-NGMcUL/lib/python2.7/site-packages/Cython/Build/Dependencies.py", line 1129, in cythonize
        cythonize_one(*args)
      File "/tmp/pip-build-env-NGMcUL/lib/python2.7/site-packages/Cython/Build/Dependencies.py", line 1296, in cythonize_one
        raise CompileError(None, pyx_file)
    Cython.Compiler.Errors.CompileError: src/gevent/greenlet.py
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-VvolsI/gevent/
ERROR: Service 'srly-ose-server' failed to build: The command '/bin/sh -c pip install --no-cache-dir -r /tmp/requirements.txt' returned a non-zero code: 1
```

#### Reproduction Steps

Steps to reproduce this issue:

`sudo apt update`
`sudo apt upgrade`
`bash <(curl -sL https://www.screenly.io/install-ose.sh`
(Y, 2, Y, N)
Wait for ansible to install and screenly directory to be created in ~/ then CTRL-C to stop script.
`nano screenly/bin/upgrade_containers.sh`
Add `--build` to last line after `-d`
`nano screenly/requirements/requirements.txt`
Change celery to 4.4.8
Change urllib3 to 1.25.11
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
(Y, 2, Y, N)

#### Environment

- Raspberry Pi Hardware Version: Pi 4B v1.4
- Raspberry Pi Network Setup: ethernet
- Screenly OSE Version: Latest dev build as of this post

HOEMichelR, 2022-08-11:
Same problem here :-(
I've successfully installed Screenly OSE earlier this year (february and march), but now I'm running into the same problem.

[image]

HOEMichelR, 2022-08-22:
I managed to install Screenly OSE on a Pi 4B v1.5 with following settings:
[https://github.com/Screenly/screenly-ose/issues/1593](url)

Thanks to @emilyyap

vpetersson, 2022-09-13:
Resolved in #1599.

closed 2022-09-13 by vpetersson