Screenly/Anthias #1592: Different results on different LCD TVs

Tissy, 2022-08-04:
Different results on different TVs.  I have a Samsung 65" LCD TV and the images play fine at full screen.  However when the Rasp 4 is plugged into a smaller Samsung 42" TV, the images are shrunk down and there is a blank white screen displayed longer than the transitioning images.  Any idea what could be causing this please.

I have adjusted `RESIN_HOST_CONFIG_gpu_mem` from 16 to 192.  Are there any other system variable that need to be changed (unable to check if above has resolved due to remote location of screen).

#### Reproduction Steps

Add JPEG assest in normal way.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 2GB
* Raspberry Pi Network Setup: Connected via Ethernet and internal WiFi
* Screenly OSE Version:  Current from GitHub

`Load Average: 0.55`
`Free Space: 24G`
`Memory : Total: 1797 \| Used: 631 \| Free: 647 \| Shared: 63 \| Buff: 40 \| Available: 1078`
`Raspberry Pi Model: Model 4B (Sony UK)`
`Screenly OSE Version: @`

Example of issue.

[image]

Many thanks,

ealmonte32, 2022-11-09:
So on the smaller TV everything else plays full screen except images? need more details to know if its a resolution issue at the host or something within the viewer..

vpetersson, 2026-05-08:
This dates from the legacy DispmanX / overscan setup; current Anthias uses the Qt webview on DRM/KMS and `RESIN_HOST_CONFIG_gpu_mem` is no longer relevant. Closing as stale — please reopen with current details if it still happens.