Screenly/Anthias #1593: Suggestion to amend requirements

emilyyap, 2022-08-11:
Hello, 
I was attempting to install Screenly OSE today on RPi3B+ and experienced several failed builds. I managed to finally get the unicorn (yay!). It might be worth looking into the latest versions and updating them. The more persistent error was gevent and it might be related to Cython. The changes that finally made it work were:

In requirements.txt:
celery==4.4.7
gevent==21.12.0
urllib3==1.25.0

In requirements.viewer.txt:
urllib3==1.25.0

In requirements-websocket.txt:
gevent==21.12.0

vpetersson, 2022-08-11:
@emilyyap feel free to do a pull request that addresses these. We're [hiring](https://screenly.workable.com/jobs/2622289) a developer to work on OSE, so until we have found a good fit, things are moving slowly.

vpetersson, 2022-09-12:
I've pushed a few fixes for installation today. The containers are yet to be tackled, but it should be possible to install the experimental branch both on Pi3 and Pi4 now.

closed 2022-09-13 by vpetersson

MisterAnderson91, 2022-09-15:
Is the older Buster-lite image still needed, or is this now working on Bullseye?

vpetersson, 2022-09-15:
No need for Buster anymore. Everything should work on Bullseye. The Docker images are still using Buster because Bullseye dropped support for Python2, but that doesn't affect the host OS.

PR #1599 (2022-09-13): Fix container builds
## Changes

* Closes #1593. (Kudos to @emilyyap for patch.)
* Rolls back a number of Python package updates as they were not available in Python 2.

I successfully built (and pushed) new containers on a Pi 4. For whatever reason, cross-compiling still doesn't work.