Screenly/Anthias #1605: Update issue - Could not find a version that satisfies the requirement ansible-core==2.12

anayrat, 2022-09-16:
#### Overview of the Issue

Hello, I tried to upgrade my screenly on my Rpi.
First, I updated the system then, I tried to upgrade screenly  by either running `$ bash <(curl sL https://www.screenly.io/installose.sh) ` or `/home/pi/screenly/bin/run_upgrade.sh`
Unfortunately, I am stuck with this error:

```
Collecting ansible-core==2.12
  Could not find a version that satisfies the requirement ansible-core==2.12 (from versions: 0.0.1a1, 2.11.0b1, 2.11.0b2, 2.11.0b3, 2.11.0b4, 2.11.0rc1, 2.11.0rc2, 2.11.0, 2.11.1rc1, 2.11.1, 2.11.2rc1, 2.11.2, 2.11.3rc1, 2.11.3, 2.11.4rc1, 2.11.4, 2.11.5rc1, 2.11.5, 2.11.6rc1, 2.11.6, 2.11.7rc1, 2.11.7, 2.11.8rc1, 2.11.8, 2.11.9rc1, 2.11.9, 2.11.10rc1, 2.11.10, 2.11.11rc1, 2.11.11, 2.11.12rc1, 2.11.12)
No matching distribution found for ansible-core==2.12
```

Upgrade fail later:

```
TASK [screenly : Install pip dependencies] **************************************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": ["/usr/bin/pip2", "install", "--no-cache-dir", "--upgrade", "-r", "/home/pi/screenly/requirements/requirements.host.txt"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nCollecting ansible-core==2.12 (from -r /home/pi/screenly/requirements/requirements.host.txt (line 1))\n\n:stderr:   Could not find a version that satisfies the requirement ansible-core==2.12 (from -r /home/pi/screenly/requirements/requirements.host.txt (line 1)) (from versions: 0.0.1a1, 2.11.0b1, 2.11.0b2, 2.11.0b3, 2.11.0b4, 2.11.0rc1, 2.11.0rc2, 2.11.0, 2.11.1rc1, 2.11.1, 2.11.2rc1, 2.11.2, 2.11.3rc1, 2.11.3, 2.11.4rc1, 2.11.4, 2.11.5rc1, 2.11.5, 2.11.6rc1, 2.11.6, 2.11.7rc1, 2.11.7, 2.11.8rc1, 2.11.8, 2.11.9rc1, 2.11.9, 2.11.10rc1, 2.11.10, 2.11.11rc1, 2.11.11, 2.11.12rc1, 2.11.12)\nNo matching distribution found for ansible-core==2.12 (from -r /home/pi/screenly/requirements/requirements.host.txt (line 1))\n"}

PLAY RECAP **********************************************************************************************************************************************************************************************************************************
localhost                  : ok=34   changed=10   unreachable=0    failed=1    skipped=6    rescued=0    ignored=0   

+ /home/pi/screenly/bin/upgrade_containers.sh
[+] Running 0/6
 ⠿ srly-ose-websocket Error                                                                                                                                                                                                            
 ⠿ srly-ose-nginx Error                                                                                                                                                                                     
 ⠿ srly-ose-server Error                                                                                                                                                                                    
 ⠿ redis Error                                                                                                                                                                                                    
 ⠿ srly-ose-viewer Error                                                                                                                                                                                    
 ⠿ srly-ose-celery Error                                                                                                                                                                                     
WARNING: Some service image(s) must be built from source by running:
    docker compose build 
```

I don't understand why docker has been installed? Now, I can't get access to web interface :(

#### Environment

* Raspberry Pi Hardware Version: 
```
Raspberry Pi 2 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
```
* Raspberry Pi Network Setup: ethernet
* Screenly OSE Version: production?

vpetersson, 2022-09-16:
We haven't had a chance to test Pi 2 support since resurrecting the project. If you want to help test this, I would recommend switching to a Bullseye and do a fresh installation.

> I don't understand why docker has been installed?

The current version has been fully dockerized for some time, which is why this is a requirement.

anayrat, 2022-09-16:
Thanks.
I upgraded to bullseye.
The installation fail during execution of upgrade_containers.sh :
```
[+] Running 0/6
 ⠿ srly-ose-websocket Error                                                                                                                                                                                                             
 ⠿ srly-ose-nginx Error                                                                                                                                                                                                                 
 ⠿ srly-ose-celery Error                                                                                                                                                                                                                
 ⠿ srly-ose-server Error                                                                                                                                                                                                                
 ⠿ redis Error                                                                                                                                                                                                                          
 ⠿ srly-ose-viewer Error                                                                                                                                                                                                                
WARNING: Some service image(s) must be built from source by running:
    docker compose build srly-ose-websocket redis srly-ose-viewer srly-ose-server srly-ose-celery srly-ose-nginx
Error response from daemon: manifest for screenly/srly-ose-websocket:latest-pi2 not found: manifest unknown: manifest unknown
```

vpetersson, 2022-09-16:
Ah yeah that's probably pi2 related. I got a potential fix in the works.

anayrat, 2022-09-17:
Unfortunately, the build failed after several hours :sob: 

So screenly no longer support rpi2?

vpetersson, 2022-09-17:
I have a work-in-progress change that should in theory bring it back. I think I should have a pi2 still somewhere, but the focus has been on pi3 and pi4, since that's the vast majority of users.

vpetersson, 2022-09-17:
I've pushed my branch now, and it should in theory add support for p1 (zero etc too) and pi2.

Once [this job](https://github.com/Screenly/screenly-ose/actions/runs/3072763303/jobs/4964457508) has finished, the images should be available on Docker Hub to pull (assuming the job finishes successfully).

anayrat, 2022-09-18:
Thanks!
It seems to work, but it is really slow.
Looking at logs, it seems several containers are started continuously : 
```
sept. 18 22:12:46 raspberrypi kernel: br-c5da82905855: port 6(veth4cee650) entered disabled state
sept. 18 22:12:46 raspberrypi kernel: device veth4cee650 left promiscuous mode
sept. 18 22:12:46 raspberrypi kernel: br-c5da82905855: port 6(veth4cee650) entered disabled state
sept. 18 22:12:46 raspberrypi NetworkManager[317]: <info>  [1663531966.8237] device (veth4cee650): released from master device br-c5da82905855
sept. 18 22:12:47 raspberrypi systemd[1]: run-docker-netns-80af363091ad.mount: Succeeded.
sept. 18 22:12:47 raspberrypi systemd[537]: run-docker-netns-80af363091ad.mount: Succeeded.
sept. 18 22:12:47 raspberrypi systemd[537]: var-lib-docker-overlay2-b8b22e90fb3eaa0874fbaba3d566b833023756b92bf56e8d793aa4beabc945f9-merged.mount: Succeeded.
sept. 18 22:12:47 raspberrypi systemd[1]: var-lib-docker-overlay2-b8b22e90fb3eaa0874fbaba3d566b833023756b92bf56e8d793aa4beabc945f9-merged.mount: Succeeded.
sept. 18 22:12:57 raspberrypi sudo[3580]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/journalctl -f
sept. 18 22:12:57 raspberrypi sudo[3580]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
sept. 18 22:13:00 raspberrypi kernel: br-c5da82905855: port 6(vethbc78ba0) entered blocking state
sept. 18 22:13:00 raspberrypi kernel: br-c5da82905855: port 6(vethbc78ba0) entered disabled state
sept. 18 22:13:00 raspberrypi kernel: device vethbc78ba0 entered promiscuous mode
sept. 18 22:13:00 raspberrypi NetworkManager[317]: <info>  [1663531980.4228] manager: (veth6948cc6): new Veth device (/org/freedesktop/NetworkManager/Devices/51)
sept. 18 22:13:00 raspberrypi NetworkManager[317]: <info>  [1663531980.4563] manager: (vethbc78ba0): new Veth device (/org/freedesktop/NetworkManager/Devices/52)
sept. 18 22:13:02 raspberrypi containerd[370]: time="2022-09-18T22:13:02.287415859+02:00" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
sept. 18 22:13:02 raspberrypi containerd[370]: time="2022-09-18T22:13:02.291203832+02:00" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
sept. 18 22:13:02 raspberrypi containerd[370]: time="2022-09-18T22:13:02.291799718+02:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
sept. 18 22:13:02 raspberrypi containerd[370]: time="2022-09-18T22:13:02.294736388+02:00" level=info msg="starting signal loop" namespace=moby path=/run/containerd/io.containerd.runtime.v2.task/moby/56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26 pid=3614 runtime=io.containerd.runc.v2
sept. 18 22:13:02 raspberrypi systemd[1]: Started libcontainer container 56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26.
sept. 18 22:13:04 raspberrypi kernel: eth0: renamed from veth6948cc6
sept. 18 22:13:04 raspberrypi kernel: IPv6: ADDRCONF(NETDEV_CHANGE): vethbc78ba0: link becomes ready
sept. 18 22:13:04 raspberrypi kernel: br-c5da82905855: port 6(vethbc78ba0) entered blocking state
sept. 18 22:13:04 raspberrypi kernel: br-c5da82905855: port 6(vethbc78ba0) entered forwarding state
sept. 18 22:13:04 raspberrypi NetworkManager[317]: <info>  [1663531984.9891] device (vethbc78ba0): carrier: link connected
sept. 18 22:13:06 raspberrypi systemd[1]: docker-56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26.scope: Succeeded.
sept. 18 22:13:06 raspberrypi systemd[1]: docker-56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26.scope: Consumed 1.146s CPU time.
sept. 18 22:13:06 raspberrypi dockerd[473]: time="2022-09-18T22:13:06.613297694+02:00" level=info msg="ignoring event" container=56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
sept. 18 22:13:06 raspberrypi containerd[370]: time="2022-09-18T22:13:06.615980822+02:00" level=info msg="shim disconnected" id=56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26
sept. 18 22:13:06 raspberrypi containerd[370]: time="2022-09-18T22:13:06.616655770+02:00" level=warning msg="cleaning up after shim disconnected" id=56585b3d96c62a261c3a2a86a2dc4cb54eb835721bcdfa1910ef2442a851ff26 namespace=moby
sept. 18 22:13:06 raspberrypi containerd[370]: time="2022-09-18T22:13:06.616927229+02:00" level=info msg="cleaning up dead shim"
sept. 18 22:13:06 raspberrypi containerd[370]: time="2022-09-18T22:13:06.743847312+02:00" level=warning msg="cleanup warnings time=\"2022-09-18T22:13:06+02:00\" level=info msg=\"starting signal loop\" namespace=moby pid=3704 runtime=io.containerd.runc.v2\n"
sept. 18 22:13:06 raspberrypi avahi-daemon[309]: Joining mDNS multicast group on interface vethbc78ba0.IPv6 with address fe80::4c06:e1ff:fe42:f441.
sept. 18 22:13:06 raspberrypi avahi-daemon[309]: New relevant interface vethbc78ba0.IPv6 for mDNS.
sept. 18 22:13:06 raspberrypi avahi-daemon[309]: Registering new address record for fe80::4c06:e1ff:fe42:f441 on vethbc78ba0.*.
sept. 18 22:13:09 raspberrypi kernel: br-c5da82905855: port 6(vethbc78ba0) entered disabled state
sept. 18 22:13:09 raspberrypi kernel: veth6948cc6: renamed from eth0
[... cut ...]
sept. 18 22:13:36 raspberrypi kernel: br-c5da82905855: port 6(veth4402569) entered disabled state
sept. 18 22:13:36 raspberrypi kernel: device veth4402569 left promiscuous mode
sept. 18 22:13:36 raspberrypi kernel: br-c5da82905855: port 6(veth4402569) entered disabled state
sept. 18 22:13:36 raspberrypi avahi-daemon[309]: Withdrawing address record for fe80::dcf3:e4ff:fea9:2cd on veth4402569.
sept. 18 22:13:36 raspberrypi NetworkManager[317]: <info>  [1663532016.4809] device (veth4402569): released from master device br-c5da82905855
```

anayrat, 2022-09-19:
It seems, screenly wrote much more information on disk. I am afraid this will destroy rpi sdcard :(

anayrat, 2022-09-19:
`screenly-srly-ose-websocket-1` container throw many errors:
```Traceback (most recent call last):
  File "websocket_server_layer.py", line 5, in <module>
    import zmq.green as zmq
ImportError: No module named zmq.green
Traceback (most recent call last):
  File "websocket_server_layer.py", line 5, in <module>
    import zmq.green as zmq
ImportError: No module named zmq.green
```

anayrat, 2022-09-19:
I stopped the container, I don't know what its role is?

vpetersson, 2022-09-20:
Fo

> `screenly-srly-ose-websocket-1` container throw many errors:
> 
> ```
>   File "websocket_server_layer.py", line 5, in <module>
>     import zmq.green as zmq
> ImportError: No module named zmq.green
> Traceback (most recent call last):
>   File "websocket_server_layer.py", line 5, in <module>
>     import zmq.green as zmq
> ImportError: No module named zmq.green
> ```

I found the issue. There a dependency got lost somewhere in the clean up process.

> I stopped the container, I don't know what its role is?

If memory serves me right, everything should work fine without it. it is for making the web interface more snappy.

anayrat, 2022-09-20:
Thanks, does it worth adding a whole container for this?

vpetersson, 2022-09-20:
We're really optimizing for Pi3 and onward, where it doesn't impact the system much. It might be worth removing on older boards.

vpetersson, 2022-09-23:
Closing this out as it now appears to work.

closed 2022-09-23 by vpetersson

anayrat, 2022-10-04:
Since your fix about zmq dependency, load reduced drastically. Thanks!