Screenly/Anthias #1613: How to install OSE on Raspberry PI 4?

tynie, 2022-09-28:
Hey guys,

i´m really in trouble getting Screenly OSE installed on my Raspberry PI 4.

What kind of image do i have to install? Or is it necessary to install Debian Buster and make a manual installation trough console?

Can anyone give me a hint?

Thank you!

vpetersson, 2022-09-28:
It should install just fine on a fresh installation of Bullseye (minimal). I've run through that a few times lately without issues. I don't think I've tested Buster as of lately, so just reflash with Bullseye and you should hopefully be good to go.

closed 2022-09-28 by vpetersson

tynie, 2022-09-28:
okay so "bash <(curl -sL https://www.screenly.io/install-ose.sh)" should just run on bullseye? I´ll try thanks.

vpetersson, 2022-09-28:
Yep.

tynie, 2022-09-28:
okay tried Bullseye minimal, no Desktop and getting this when running bash <(curl -sL https://www.screenly.io/install-ose.sh) as root

```
/dev/fd/63: line 1: syntax error near unexpected token `<'
/dev/fd/63: line 1: `<!doctype html> <html> <head> <script>!function(e,t,a,n,r){e[n]=e[n]||[],e[n].push({"gtm.start":(new Date).getTime(),event:"gtm.js"});var g=t.getElementsByTagName(a)[0],m=t.createElement(a),s="dataLayer"!=n?"&l="+n:"";m.async=!0,m.src="https://www.googletagmanager.com/gtm.js?id="+r+s,g.parentNode.insertBefore(m,g)}(window,document,"script","dataLayer","GTM-W77ZZG6");</script> <meta charset="utf-8"> <meta name="viewport" content="width=device-width, initial-scale=1, shrink-to-fit=no"> <title>Screenly - Something went wrong</title> <meta name="description" content=""><meta name="author" content="Screenly"> <meta property="og:site_name" content="Screenly"> <meta property="og:title" content="Screenly - Something went wrong"> <meta property="og:image" content="/assets/screenly-twitter-0a1ba046940de5ff8f4ab889ea5dcd18128549136e042f83a30e2a4d5b254a29.jpg"> <meta property="og:url" content="https://www.screenly.io/404.html"> <meta property="og:type" content="website"> <meta property="og:description" content=""> <meta property="og:article:author" content="Screenly"> <meta name="twitter:card" content="summary"> <meta name="twitter:title" content="Screenly - Something went wrong"> <meta name="twitter:site" content="@ScreenlyApp"> <meta name="twitter:description" content=""> <meta name="twitter:image" content="/assets/screenly-twitter-0a1ba046940de5ff8f4ab889ea5dcd18128549136e042f83a30e2a4d5b254a29.jpg"> <link rel="preload" href="/assets/base/_fonts-6b8850d8eb561c8b727d0c8ca1e7b08d20c94f07ccea470caf8f684e2405d057.css" as="style"> <link rel="stylesheet" type="text/css" href="/assets/base/_fonts-6b8850d8eb561c8b727d0c8ca1e7b08d20c94f07ccea470caf8f684e2405d057.css"> <link rel="preload" href="/assets/vendors/font-awesome/font-awesome-3d4ef4f0396e116904adf875588c1b97fd7aa448c415cd7992ef3413104c87fe.css" as="style"> <link rel="stylesheet" type="text/css" href="/assets/vendors/font-awesome/font-awesome-3d4ef4f0396e116904adf875588c1b97fd7aa448c415cd7992ef3413104c87fe.css"> <link rel="stylesheet" type="text/css" href="/assets/main-b272565c4115b9796527c53663b8b418c3651ccd168e4ac2511d1e008bec4b0b.css"> <link rel="preconnect" href="https://fonts.googleapis.com/css?family=Montserrat:200,300,400,500,600" rel="stylesheet"> <link rel="alternate" type="application/rss+xml" title="Screenly RSS" href="https://www.screenly.io/rss.xml"> <link rel="stylesheet" type="text/css" href="/assets/syntax-46d47743c00b6ca35abdc27b1e25b20e6c594370bd31639649b40f40d7ad238c.css"> <script src="https://browser.sentry-cdn.com/5.5.0/bundle.min.js" crossorigin="anonymous"></script> <script>Sentry.init({dsn:"https://81ccfd7cefbf4b3d8c91e11a0c2b7dff@sentry.screenlyapp.com/13"});</script> </head> <body class="page-not-found"> <div id="wrapper"> <header id="header"> <nav> <div class="navbar navbar-expand-lg navbar-dark"> <a class="navbar-brand nav-home" href="https://www.screenly.io"> <img src="/assets/logo-purple-full-a46925cff6207b1e6eaeacb700192dd9351ce476478e282adb50a174cac44e37.svg" alt="Screenly logo"> </a> <div class="collapse navbar-collapse order-2 order-lg-1" id="navbarSupportedContent"> <ul class="navbar-nav mr-auto"> <li class="nav-item "> <a class="nav-link nav-features" href="https://www.screenly.io/features/">Features</a> </li> <li class="nav-item "> <a class="nav-link nav-pricing" href="https://www.screenly.io/pricing/">Pricing </a> </li> <li class="nav-item "> <a class="nav-link nav-players" href="https://www.screenly.io/digital-signage-players/">Players</a> </li> <li class="nav-item "> <a class="nav-link nav-use-cases" href="https://www.screenly.io/use-cases/">Case Studies</a> </li> <li class="nav-item "> <a class="nav-link nav-blog" href="https://www.screenly.io/blog/">Resources</a> </li> <li class="nav-item "> <a class="nav-link nav-blog" href="https://www.screenly.io/ose/">OSE</a> </li> </ul> <ul class="nav navbar-nav navbar-right"> <li class="nav-item"> <a class="nav-link nav-log-in" href="https://login.screenlyapp.com/login">Sign In</a> </li> </ul> </div> <div class="order-1 order-lg-2 d-flex cmy-2 cmy-lg-0"> <a class="nav-link nav-schedule-demo btn" href="https://login.screenlyapp.com/sign-up"> Free Trial </a> <button class="navbar-toggler" type="button" data-toggle="collapse" data-target="#navbarSupportedContent" aria-controls="navbarSupportedContent" aria-expanded="false" aria-label="Toggle navigation"> <span class="navbar-toggler-icon"></span> </button> </div> </div> </nav> </header> <main id="main"> <div class="container"> <div class="row"> <section class="col-12 text-center cmt-7 cmb-8"> <div class="row"> <div class="col-12"> <h3 class="text-center cmb-2">Something went wrong</h3> <div class="title-description text-center font-weight-normal">We can’t find the page you’re looking for </div> </div> <div class="col-12 text-center cmt-8"> <img src="/assets/screenly-tv-bf22a0a87740f5667829e7adf06fb7111bdd4cfd2f9dfdace75527c4ee88a1fa.svg"> </div> <div class="col-12 text-center cmt-6"> <a class="btn btn-primary mb-0" href="/">Back to Home</a> </div> </div> </section> </div> </div> </main> </div> <script defer src="https://static.cloudflareinsights.com/beacon.min.js/v652eace1692a40cfa3763df669d7439c1639079717194" integrity="sha512-Gi7xpJR8tSkrpF7aordPZQlW2DLtzUlZcumS8dMQjwDHEnw9I7ZLyiOj/6tZStRBGtGgN6ceN6cMH8z7etPGlw==" data-cf-beacon='{"rayId":"751c6a4b9d85b7e8","token":"eb43a2127fb3449da3c27a8cc23b2334","version":"2022.8.1","si":100}' crossorigin="anonymous"></script>'
```

vpetersson, 2022-09-28:
Please try again. It was a temporary glitch related to some website changes we made today.

tynie, 2022-09-28:
okay that worked. After installation and reboot i expected to see the screenly splash screen. Instead it shows the console without any information about the screenly installation. Also the webinterface seems not to be started. Anything i can do?

vpetersson, 2022-09-28:
Strange. Try 

```
$ cd ~/screenly
$ docker compose ps
[...]
$ docker compose logs
[...]
```

And post the output.

tynie, 2022-09-28:
It seems not to be installed, what do you think?

```
root@raspberrypi:~# cd ~/screenly
-bash: cd: /root/screenly: No such file or directory
root@raspberrypi:~# cd /screenly
-bash: cd: /screenly: No such file or directory
```

I ran the installation again. The only error i can see was:
```
+ sudo -E ansible-playbook site.yml
ERROR! the playbook: site.yml could not be found
```

vpetersson, 2022-09-28:
You're logged in as root. I presume this is not the intended runtime user. Did you install it as a different user? If not, you really ought to wipe the card and start over with a regular runtime user (e.g. `pi`).

tynie, 2022-09-28:
okay made a fresh install of bullseye minimal and ran the setup.

```
config@raspberrypi:~ $ cd ~/screenly
-bash: cd: /home/config/screenly: No such file or directory
```

So same behavior i think

vpetersson, 2022-09-28:
Something went very wrong during the installation. Try re-running the installer as the runtime user.

tynie, 2022-09-28:
okay. Is it ok to run the setup via putty or is it better to run it directly on the PI?

vpetersson, 2022-09-28:
Shouldn't really matter, but it's probably easier to install over an SSH session if you want to extract logs etc.

tynie, 2022-09-28:
okay i ran the setup again, same problem.... Any other idea?

vpetersson, 2022-09-28:
Please post the installation logs. The only things that I can think of is that the installation script assumes the installation is running using the user `pi` (which was the default until Bullseye). There might be hard coded settings for that. Hence, the best way might be reflashing the device and create the user `pi` until this has been resolved.

wer1000, 2022-11-16:
I have the same problem. Is there a solution?