Screenly/Anthias #1619: I cannot use this to flash an SD card for a Pi 4. Error = missing partition table

mikekorncnx, 2022-09-30:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version:
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
* Screenly OSE Version:

mikekorncnx, 2022-10-04:
This is the full error message from balenaEtcher:

Missing Partition Table

It looks like this is not a bootable image. The image does not appear to contain a partition table, and might not be recognized or bootable by your device.

If I proceed to burn flash the image anyway, the Pi 4 will not boot. 

Any help would be greatly appreciated

vpetersson, 2022-10-04:
Please see the updated release notes. TLDR don't use the images. They're too old. Use the manual installer for now. New disk images will be available in the near future.

closed 2022-10-04 by vpetersson

mikekorncnx, 2022-10-04:
After installing screenly manually as instructed, it boots directly into raspbain lite and not screenly

vpetersson, 2022-10-04:
We identified a bug last week that is triggered when the runtime use is anything other than "pi" (which was the default until Bullseye). Try doing another install with the user "pi".

mikekorncnx, 2022-10-05:
Thanks - that worked. After I got it up and running I discovered that I cannot display video of any filetype - it displays a black screen for the duration. I have posted this elsewhere but still no response. Wondering if this is a bug.