Screenly/Anthias #1621: Allow Screenly to be installed with a user other than 'pi'

vpetersson, 2022-10-04:
Previous to Rasbian OS/Bullseye, the installer had the user 'pi' hard coded into the system, and we made assumptions around this. Since this is no longer true, there are a number of ways the installer will break if it is installed using a different user.

Breakage has been reported in a few issues, including in #1619.

MangNPB, 2022-10-19:
I tried the latest bash script install and 'pi' user is still required. is there a different installing .sh or method to install it on other user?

vpetersson, 2022-10-19:
Yes, this ticket is still opened, so it isn't resolved.

ealmonte32, 2022-11-03:
@MangNPB ,

This is a very big task because you would need to have part of the install script get the current username and then replace every single ansible task or code that mentions `/home/pi` and replace it with the variable that will obtain the name of the current user so that it would always be dynamic.

vpetersson, 2022-11-03:
Yes, and not only that, we would also make sure all `sudoers` is configured properly etc.

MangNPB, 2022-11-22:
Big thanks to all those who are in the works for making this change as it's one of the brute force first choice of username.
Any estimations of when will this be done?

vpetersson, 2022-11-22:
It's not super highly prioritized given how easy it is to work around.

vpetersson, 2022-11-29:
Resolved in https://github.com/Screenly/screenly-ose/pull/1662