Screenly/Anthias #1648: Resolve issue with splash_screen

vpetersson, 2022-10-28:
When running on Balena, the splash screen is crashing with the following error:

```
28.10.22 13:05:16 (+0000)  srly-ose-server  Exception on /splash-page [GET]
28.10.22 13:05:16 (+0000)  srly-ose-server  Traceback (most recent call last):
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 2447, in wsgi_app
28.10.22 13:05:16 (+0000)  srly-ose-server      response = self.full_dispatch_request()
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1952, in full_dispatch_request
28.10.22 13:05:16 (+0000)  srly-ose-server      rv = self.handle_user_exception(e)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask_restful/__init__.py", line 271, in error_router
28.10.22 13:05:16 (+0000)  srly-ose-server      return original_handler(e)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask_cors/extension.py", line 165, in wrapped_function
28.10.22 13:05:16 (+0000)  srly-ose-server      return cors_after_request(app.make_response(f(*args, **kwargs)))
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1821, in handle_user_exception
28.10.22 13:05:16 (+0000)  srly-ose-server      reraise(exc_type, exc_value, tb)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1950, in full_dispatch_request
28.10.22 13:05:16 (+0000)  srly-ose-server      rv = self.dispatch_request()
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1936, in dispatch_request
28.10.22 13:05:16 (+0000)  srly-ose-server      return self.view_functions[rule.endpoint](**req.view_args)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "server.py", line 1812, in splash_page
28.10.22 13:05:16 (+0000)  srly-ose-server      my_ip = get_node_ip()
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/src/app/lib/utils.py", line 89, in get_node_ip
28.10.22 13:05:16 (+0000)  srly-ose-server      ), headers=headers)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 75, in get
28.10.22 13:05:16 (+0000)  srly-ose-server      return request('get', url, params=params, **kwargs)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 61, in request
28.10.22 13:05:16 (+0000)  srly-ose-server      return session.request(method=method, url=url, **kwargs)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 529, in request
28.10.22 13:05:16 (+0000)  srly-ose-server      resp = self.send(prep, **send_kwargs)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 645, in send
28.10.22 13:05:16 (+0000)  srly-ose-server      r = adapter.send(request, **kwargs)
28.10.22 13:05:16 (+0000)  srly-ose-server    File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 519, in send
28.10.22 13:05:16 (+0000)  srly-ose-server      raise ConnectionError(e, request=request)
28.10.22 13:05:16 (+0000)  srly-ose-server  ConnectionError: HTTPConnectionPool(host='10.114.104.1', port=48484): Max retries exceeded with url: /v1/device?apikey=e2446068f12647a691cb7a946151f3d4 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x720bd130>: Failed to establish a new connection: [Errno 111] Connection refused',))
```

Thee error appears to happen on both Pi 3 and Pi 4.

nicomiguelino, 2022-11-02:
I've encountered the same issue on my end. Just leaving this here for reference.

[issue-1648-balena-2022-02-11-0358.log](https://github.com/Screenly/screenly-ose/files/9924697/issue-1648-balena-2022-02-11-0358.log)

vpetersson, 2022-11-04:
I can confirm that this works for me on both Pi 3 and Pi 4.

nicomiguelino, 2022-11-29:
I'll close this issue when I've verified the results on the BalenaHub images.

nicomiguelino, 2022-11-30:
Here's a more recent log...

```
srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 515, in request
 srly-ose-viewer      prep = self.prepare_request(req)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 453, in prepare_request
 srly-ose-viewer      hooks=merge_hooks(request.hooks, self.hooks),
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 318, in prepare
 srly-ose-viewer      self.prepare_url(url, params)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 392, in prepare_url
 srly-ose-viewer      raise MissingSchema(error)
 srly-ose-viewer  MissingSchema: Invalid URL 'None/v1/device?apikey=None': No scheme supplied. Perhaps you meant http://none/v1/device?apikey=None?
 srly-ose-viewer  Traceback (most recent call last):
 srly-ose-viewer    File "viewer.py", line 481, in <module>
 srly-ose-viewer      main()
 srly-ose-viewer    File "viewer.py", line 459, in main
 srly-ose-viewer      retry_call(get_supervisor_api_response, tries=5, delay=1)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/retry/api.py", line 101, in retry_call
 srly-ose-viewer      return __retry_internal(partial(f, *args, **kwargs), exceptions, tries, delay, max_delay, backoff, jitter, logger)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/retry/api.py", line 33, in __retry_internal
 srly-ose-viewer      return f()
 srly-ose-viewer    File "/usr/src/app/lib/utils.py", line 81, in get_supervisor_api_response
 srly-ose-viewer      ), headers=headers)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 75, in get
 srly-ose-viewer      return request('get', url, params=params, **kwargs)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 61, in request
 srly-ose-viewer      return session.request(method=method, url=url, **kwargs)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 515, in request
 srly-ose-viewer      prep = self.prepare_request(req)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 453, in prepare_request
 srly-ose-viewer      hooks=merge_hooks(request.hooks, self.hooks),
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 318, in prepare
 srly-ose-viewer      self.prepare_url(url, params)
 srly-ose-viewer    File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 392, in prepare_url
 srly-ose-viewer      raise MissingSchema(error)
 srly-ose-viewer  requests.exceptions.MissingSchema: Invalid URL 'None/v1/device?apikey=None': No scheme supplied. Perhaps you meant http://none/v1/device?apikey=None?
 srly-ose-viewer  ./bin/start_viewer.sh: line 54: kill: (37) - No such process
```

nicomiguelino, 2022-12-01:
This will remain open until the splash screen issue works using a BalenaHub image.

nicomiguelino, 2022-12-02:
I can confirm that this works on my Pi 3 and Pi 4 devices.

PR #1653 (2022-11-03): Fix splash screen issue on Balena.
#### General

Might fix #1648.
PR #1655 (2022-11-29): Add follow-up fixes to Balena splash screen error.
Fixes #1648.

#### Errors

- The screen keeps displaying a page with the following **_Internal Server Error_** message &ndash; "The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

#### Logs

The latest logs come first.
- [screenly-ose-pi4_weathered-smoke-21.11.22_09_53_53_(-0800).txt](https://github.com/Screenly/screenly-ose/files/10059302/screenly-ose-pi4_weathered-smoke-21.11.22_09_53_53_.-0800.txt)
- [screenly-ose-pi4_weathered-smoke-21. [... cut]
PR #1677 (2022-12-01): Add follow-up fixes to Balena splash screen error.
Can fix #1648 (for good)...
closed 2022-12-02 by nicomiguelino