Screenly/Anthias #1649: No shutdown, no reboot and loss of asset.

titis14, 2022-10-28:
Hello, I am testing the system and I have 2 problems.
In the Settings menu the 2 Reboot or turn off buttons do not work, no reaction on the raspberry.

Another problem when I restart the Raspberry it does not reload my web page I return to the default screenly image,
I have to go to Schedule overview and I have to deactivate and reactivate my asset.
Thanks for your help.

#### Environment

* Raspberry Pi Hardware Version: Pi 4 1Go Ram SD 32Go
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version:  master@b1b84fd

vpetersson, 2022-11-01:
These are two separate issue and should be treated as such. 

I'll repurpose this issue for the reboot/shutdown. 

This is expected behavior. This currently only works when running on Balena.

The reason being that we are unable to issue a shutdown/reboot command from within the container. This needs to be reworked as we lost this support with the move to running the workload within containers.

ealmonte32, 2022-11-16:
@titis14 
Since I can estimate for you that it will be a long time before any coding work is done to allow a container to pass on a simple command to the host, such as shutdown or reboot (since this is specifically for the Screenly OSE container, not the Pi), I would suggest you get yourself a terminal like putty on Windows or terminal on Mac, and just ssh into the Pi to turn it off/restart it when necessary, it is a simple process and if you need help doing that just ask.

Since the other topic you already sent to another Issue, if my suggestion above solves your issue, pls close this Issue as I mentioned before we want to clean up the issues that have been answered.

vpetersson, 2022-11-29:
I just remember that some work was done on this in [host_agent.py](https://github.com/Screenly/screenly-ose/blob/master./host_agent.py). It hasn't been touch in some time, but the concept of using Redis for this isn't bad.

nicomiguelino, 2022-12-13:
I ran `journalctl -u screenly-host-agent` in host (Raspberry Pi OS Lite, Bullseye, 64-Bit). Here's a console output for reference:

```
-- Journal begins at Wed 2022-09-21 20:06:10 PDT, ends at Tue 2022-12-13 09:49:57 PST. --
Dec 08 20:50:05 raspberrypi4 systemd[1]: Started Screenly Host Agent.                                                                                         Dec 08 20:50:05 raspberrypi4 systemd[15014]: screenly-host-agent.service: Failed to determine user credentials: No such process
Dec 08 20:50:05 raspberrypi4 systemd[15014]: screenly-host-agent.service: Failed at step USER spawning /usr/bin/python3: No such process
Dec 08 20:50:05 raspberrypi4 systemd[1]: screenly-host-agent.service: Main process exited, code=exited, status=217/USER
Dec 08 20:50:05 raspberrypi4 systemd[1]: screenly-host-agent.service: Failed with result 'exit-code'.                                                         Dec 08 20:50:15 raspberrypi4 systemd[1]: screenly-host-agent.service: Scheduled restart job, restart counter is at 1.                                         Dec 08 20:50:15 raspberrypi4 systemd[1]: Stopped Screenly Host Agent.                                                                                         Dec 08 20:50:15 raspberrypi4 systemd[1]: Started Screenly Host Agent.                                                                                         Dec 08 20:50:15 raspberrypi4 systemd[15492]: screenly-host-agent.service: Failed to determine user credentials: No such process
Dec 08 20:50:15 raspberrypi4 systemd[15492]: screenly-host-agent.service: Failed at step USER spawning /usr/bin/python3: No such process
Dec 08 20:50:15 raspberrypi4 systemd[1]: screenly-host-agent.service: Main process exited, code=exited, status=217/USER
Dec 08 20:50:15 raspberrypi4 systemd[1]: screenly-host-agent.service: Failed with result 'exit-code'.                                                         Dec 08 20:50:25 raspberrypi4 systemd[1]: screenly-host-agent.service: Scheduled restart job, restart counter is at 2.                                         Dec 08 20:50:25 raspberrypi4 systemd[1]: Stopped Screenly Host Agent.                                                                                         Dec 08 20:50:25 raspberrypi4 systemd[1]: Started Screenly Host Agent.                                                                                         Dec 08 20:50:25 raspberrypi4 systemd[16015]: screenly-host-agent.service: Failed to determine user credentials: No such process
Dec 08 20:50:25 raspberrypi4 systemd[16015]: screenly-host-agent.service: Failed at step USER spawning /usr/bin/python3: No such process
Dec 08 20:50:25 raspberrypi4 systemd[1]: screenly-host-agent.service: Main process exited, code=exited, status=217/USER
Dec 08 20:50:25 raspberrypi4 systemd[1]: screenly-host-agent.service: Failed with result 'exit-code'.                                                         Dec 08 20:50:36 raspberrypi4 systemd[1]: screenly-host-agent.service: Scheduled restart job, restart counter is at 3.                                         Dec 08 20:50:36 raspberrypi4 systemd[1]: Stopped Screenly Host Agent.                                                                                         Dec 08 20:50:36 raspberrypi4 systemd[1]: Started Screenly Host Agent.                                                                                         Dec 08 20:50:36 raspberrypi4 systemd[16197]: screenly-host-agent.service: Failed to determine user credentials: No such process
Dec 08 20:50:36 raspberrypi4 systemd[16197]: screenly-host-agent.service: Failed at step USER spawning /usr/bin/python3: No such process
Dec 08 20:50:36 raspberrypi4 systemd[1]: screenly-host-agent.service: Main process exited, code=exited, status=217/USER
Dec 08 20:50:36 raspberrypi4 systemd[1]: screenly-host-agent.service: Failed with result 'exit-code'.                                                         Dec 08 20:50:46 raspberrypi4 systemd[1]: screenly-host-agent.service: Scheduled restart job, restart counter is at 4.                                         Dec 08 20:50:46 raspberrypi4 systemd[1]: Stopped Screenly Host Agent.
Dec 08 20:50:46 raspberrypi4 systemd[1]: Started Screenly Host Agent.                                                                                         Dec 08 20:50:46 raspberrypi4 systemd[16464]: screenly-host-agent.service: Failed to determine user credentials: No such process                               Dec 08 20:50:46 raspberrypi4 systemd[16464]: screenly-host-agent.service: Failed at step USER spawning /usr/bin/python3: No such process                      Dec 08 20:50:46 raspberrypi4 systemd[1]: screenly-host-agent.service: Main process exited, code=exited, status=217/USER
Dec 08 20:50:46 raspberrypi4 systemd[1]: screenly-host-agent.service: Failed with result 'exit-code'.
Dec 08 20:50:56 raspberrypi4 systemd[1]: screenly-host-agent.service: Scheduled restart job, restart counter is at 5.
```

closed 2022-12-14 by nicomiguelino