Screenly/Anthias #1651: Resolve video playback issue on Pi4 and Balena

vpetersson, 2022-11-01:
#### Overview of the Issue

A paragraph or two about the issue you're experiencing. This is somewhat related to https://github.com/Screenly/screenly-ose/issues/1590

#### Reproduction Steps

* Schedule a video asset

```
01.11.22 09:31:47 (+0000)  srly-ose-viewer  [ece223e8] mmal_codec decoder: VCSM init succeeded: Legacy
01.11.22 09:31:47 (+0000)  srly-ose-viewer  [ece223e8] main decoder error: buffer deadlock prevented
01.11.22 09:31:47 (+0000)  srly-ose-viewer  error: XDG_RUNTIME_DIR not set in the environment.
01.11.22 09:31:47 (+0000)  srly-ose-viewer  error: XDG_RUNTIME_DIR not set in the environment.
01.11.22 09:31:56 (+0000)  srly-ose-viewer  Showing asset foobar.mp4 (video)
01.11.22 09:31:56 (+0000)  srly-ose-viewer  Current url is null
01.11.22 09:31:56 (+0000)  srly-ose-viewer  [f290c398] mmal_codec decoder: VCSM init succeeded: Legacy
01.11.22 09:31:56 (+0000)  srly-ose-viewer  [f290c398] main decoder error: buffer deadlock prevented
01.11.22 09:31:56 (+0000)  srly-ose-viewer  error: XDG_RUNTIME_DIR not set in the environment.
01.11.22 09:31:56 (+0000)  srly-ose-viewer  error: XDG_RUNTIME_DIR not set in the environment.
```

#### Environment

* Raspberry Pi Hardware Version: Pi 4
* Raspberry Pi Network Setup: Unrelated
* Screenly OSE Version: Latest

nicomiguelino, 2022-11-16:
@vpetersson, I still have to figure out how to (programmatically) set `gpu_mem` and `dtoverlay`. It seems that setting `gpu_mem` to a higher value will fix the playback issue.

Setting `gpu_mem` to something higher might do the trick.

vpetersson, 2022-11-16:
In balena you do those using the dashboard on a per fleet or per device. Some of them you can set using the cli.

nicomiguelino, 2022-11-16:
Efforts to automate modification of config parameters like `dtoverlay` and `gpu_mem` on Balena will be done on #1661.

closed 2022-11-16 by nicomiguelino