Screenly/Anthias #1652: Default page on restart and not my web page

titis14, 2022-11-02:
Hello,
When I restart the Raspberry it does not reload my web page I revert to the default screenly image.
I need to access the schedule overview and I need to deactivate and reactivate my asset.
Thank you for your help.

Environment:
    Raspberry Pi Hardware Version: Pi 4 1Go Ram SD 32Go
    Raspberry Pi Network Setup: Ethernet
    Screenly OSE Version: master@b1b84fd

ealmonte32, 2022-11-03:
Does it show the screenly splash page and then after a while it loads your assets? or it just stays showing the black screen with screenly logo?
Do you know the IP of the Pi to manage the Assets via the interface?

titis14, 2022-11-09:
Hi,
I have the purple page with the screenly logo telling me to go to the link to do my scenes.
Yes I know because as indicated at the beginning I configured a web page and it does not come back on restart.

ealmonte32, 2022-11-10:
Sorry I am having a little hard time understanding, just confirm for me:
1. You restart the Pi and Screenly logo with IP of the Pi appears.
2. This Screenly logo stays showing the IP address of the Pi indefinitely.
3. You have already gone to the IP of the Pi and Schedule Overview and configured assets and they are active.
4. The only way the assets you have scheduled will show after a restart of the Pi is when you go into the Schedule Overview page and deactivate and re-activate the asset?

Is this correct flow of issue?
You need to show the logs while this is happening, enter the console and type: `journalctl -f` and wait a minute or more seeing everything that is going on live, and also if you are using the latest version then also do `docker logs screenly-srly-server` (you should press tab on this last docker logs command to make sure you are putting the exact container name of your installation), so after typing `docker logs screenly-` press tab and it will autocomplete for you or show you the available options of names for typing/finalizing the command.

When you have some logs to show come and paste them here, that will help better understand what the situation is.

titis14, 2022-11-11:
> 
    You restart the Pi and Screenly logo with IP of the Pi appears.
    This Screenly logo stays showing the IP address of the Pi indefinitely.
    You have already gone to the IP of the Pi and Schedule Overview and configured assets and they are active.
    The only way the assets you have scheduled will show after a restart of the Pi is when you go into the Schedule Overview page and deactivate and re-activate the asset?

Is this correct flow of issue?

Hello,
Yes that's exactly it, I see to look at the logs and I come back to you.
Thanks

titis14, 2022-11-12:
Hello,
Finally everything is good, it's just that when switching on, the loading of the scenes takes a little time. Indeed once on the purple screen with the ip you have to wait, then I have a black screen with the logo then my web page comes back. You have to wait about 2/3 min.
Sorry for the disturbance.

vpetersson, 2022-11-14:
I think I know the root cause here. We had a bug in the splash-screen. This has now been fixed. It's either that or for whatever reason, the IP wasn't set properly during the installation. Either scenario would yield in the splash-screen not working.

ealmonte32, 2022-11-14:
> Hello, Finally everything is good, it's just that when switching on, the loading of the scenes takes a little time. Indeed once on the purple screen with the ip you have to wait, then I have a black [...]

If you dont want to wait the time it takes from the splash screen to your assets and your IP won't change (and if it does you know how to easily get it), just go into the OSE settings and disable the splash screen so that when it restarts it will always go directly to show your assets.

titis14, 2022-11-14:
Good evening,
Thanks for the advice, I'll look into finding that option.

titis14, 2022-11-15:
Thank you very much for the capture, I will take a look.

ealmonte32, 2022-11-16:
> Thank you very much for the capture, I will take a look.

Once you confirm this helps with the splash screen waiting, pls close this issue so that we can clean up the Issues list.

ealmonte32, 2023-02-08:
@titis14 
Did you ever disable this setting and get this working to your liking?
Just wanting to close stale Issues.

titis14, 2023-02-09:
Hello, Yes it's good you can close

closed 2023-02-09 by titis14