Screenly/Anthias #1654: Install fails - Pi Zero W - Buster & Bullseye

AnchorHRG, 2022-11-05:
When using the install script the install fails on both Buster & Bullseye on Raspberry Pi Zero Ws, unsure about other Pi models.
Install fails on Buster due to Python not being version 3.8, Ansible fails to install.
Here is the log for the Bullseye failed installation:
<details>
<summary>Log</summary>

 ```
      _____                           __         ____  _____ ______
      / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
      \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
     ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
    /____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                     /____/
Screenly OSE requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Do you want Screenly OSE to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to perform a full system upgrade as well? (y/N)
+ REPOSITORY=https://github.com/screenly/screenly-ose.git
+ sudo mkdir -p /etc/ansible
+ sudo tee /etc/ansible/hosts
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt update -y
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
33 packages can be upgraded. Run 'apt list --upgradable' to see them.
+ sudo apt-get install -y --no-install-recommends git libffi-dev libssl-dev python3 python3-dev python3-pip whois
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.30.2-1).
libffi-dev is already the newest version (3.3-6).
python3 is already the newest version (3.9.2-3).
python3-dev is already the newest version (3.9.2-3).
whois is already the newest version (5.5.10).
libssl-dev is already the newest version (1.1.1n-0+deb11u3+rpt1).
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 33 not upgraded.
+ '[' n == y ']'
+ export MANAGE_NETWORK=false
+ MANAGE_NETWORK=false
+ '[' master = master ']'
++ curl -s https://raw.githubusercontent.com/Screenly/screenly-ose/master/requirements/requirements.host.txt
++ grep ansible
+ ANSIBLE_VERSION=ansible-core==2.12
+ sudo pip install ansible-core==2.12
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible-core==2.12
  Using cached https://www.piwheels.org/simple/ansible-core/ansible_core-2.12.0-py3-none-any.whl (2.1 MB)
Collecting jinja2
  Using cached https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting packaging
  Using cached https://www.piwheels.org/simple/packaging/packaging-21.3-py3-none-any.whl (40 kB)
Collecting resolvelib<0.6.0,>=0.5.3
  Using cached https://www.piwheels.org/simple/resolvelib/resolvelib-0.5.4-py2.py3-none-any.whl (12 kB)
Collecting PyYAML
  Using cached https://www.piwheels.org/simple/pyyaml/PyYAML-6.0-cp39-cp39-linux_armv6l.whl (45 kB)
Collecting cryptography
  Using cached cryptography-38.0.3.tar.gz (599 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting cffi>=1.12
[... 248 lines cut ...]
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' '!' -name locale.alias -exec rm -r '{}' ';'
+ sudo chown -R pi:pi /home/pi
+ '[' '!' -f /etc/sudoers.d/010_pi-nopasswd ']'
+ check_defaultpw
+ '[' master = master ']'
+ '[' false = false ']'
+ set +x
Wrong salt length: 3 bytes when 8 <= n <= 16 expected.
The default raspberry pi password was not detected, continuing with installation...
++ git rev-parse --abbrev-ref HEAD
fatal: not a git repository (or any of the parent directories): .git
++ git rev-parse --short HEAD
fatal: not a git repository (or any of the parent directories): .git
++ lsb_release -a
No LSB modules are available.
+ echo -e 'Screenly version: @\nDistributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye'
+ '[' false = false ']'
+ set +x
Installation completed.
You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)
```
</details>

vpetersson, 2022-11-05:
The error appears to be that the package cryptography fail to build, which in turn is likely due to lack of RAM in the Pi Zero. 

Not sure what the easiest fix for this, as this package is needed by ansible.

AnchorHRG, 2022-11-05:
@vpetersson Is there an older version of the install script that can be used for now for Pi Zero Ws, compatible with Buster? Or is there a pre built image based on Raspbian OS (Is [2022-11-04-raspberry-pi.zip](https://github.com/Screenly/screenly-ose/releases/download/v0.18.4/2022-11-04-raspberry-pi.zip) based on Raspbian OS?)? We need to install additional software alongside Screenly OSE.
Thanks for taking a look.

vpetersson, 2022-11-05:
Try the pre made image for pi 1. It's not based on raspbian but should in theory work.

ScahR45, 2022-11-07:
Hi
I've exactly the same error on Raspberry Pi 4 B 2GB with Raspbian Bulleyes lite 32bit

```
...
ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
```

Thank you for you work !

vpetersson, 2022-11-07:
> Hi I've exactly the same error on Raspberry Pi 4 B 2GB with Raspbian Bulleyes lite 32bit
>
> ```
> ...
> ERROR: Failed building wheel for cryptography
> Failed to build cryptography
> ERROR: Could not build [...]

That's odd. @nicomiguelino are you able to reproduce this?

rebootit, 2022-11-07:
Exactly the same thing on pi 2 and pi 3 when trying a fresh install

vpetersson, 2022-11-07:
So we're getting `error: can't find Rust compiler` in the logs. Not sure why this would have changed, since we have a pinned version of ansible. 

Can someone try just installing rust (`curl https://sh.rustup.rs -sSf | sh`) and then re-run the installer.

nicomiguelino, 2022-11-07:
Hi @rebootit & @vpetersson. Yes, I'm getting the same issue. I've ran the command for installing Rust (and I've sourced my `.bashrc` as well), but still getting the same error.

vpetersson, 2022-11-08:
This is odd though. I wonder what introduced this. It ran just fine a few weeks ago and I don't recall bumping up anything related to this.

ScahR45, 2022-11-08:
Hi,
I ran installation many times 2 weeks ago with success too
I just retrying this morning, even with rustc installed, cryptography installation always KO on 32bit raspbian bullseye
```
pi@raspberrypi:~ $ rustc -V
rustc 1.65.0 (897e37553 2022-11-02)
```

If i make a pip install cryptography, it block here (tested 2 times with fresh install) : 

```
pi@raspberrypi:~ $ pip install cryptography
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cryptography
  Downloading cryptography-38.0.3.tar.gz (599 kB)
     |████████████████████████████████| 599 kB 5.9 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting cffi>=1.12
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting pycparser
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (PEP 517) ... /

```

vpetersson, 2022-11-08:
I think i see what's going on here. Ansible might depend on a non-pinned version of cryptography and cryptography just released [a new version](https://pypi.org/project/cryptography/#history) (38.0.3) on November 1st. This release doesn't have any pre-compiled wheels yet. This is likely what broke things.

Assuming this is the case, the workaround is easy: just install cryptography first (`sudo pip install cryptography==38.0.2`) and then run the installer.  

Assuming that does the trick, we can easily port this to the installer.

ScahR45, 2022-11-08:
Yes  with cryptography 38.0.2 it works ! Thanks

nicomiguelino, 2022-11-08:
It works on my Pi 4 with 32-bit OS installed.

PR #1656 (2022-11-08): Fixes installer
Manually pins cryptography library to 38.0.2.

Closes #1654
closed 2022-11-08 by vpetersson