Screenly/Anthias #1657: how to get a .xml in my screenly content

TeunvanderCruijsen, 2022-11-08:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version:
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
* Screenly OSE Version:

TeunvanderCruijsen, 2022-11-08:
We use your service in our building in the Netherlands. We would like to share an real time overview of the bus/tramstop next to our building.
9292 offers both api and an .xml file/app to show this real time information.
Now is my questing how do I get this on my screens? I’m not a programmer so I don’t now how to use the ose or any code language.
I hope you can help me.

vpetersson, 2022-11-08:
Well, XML is not really something that we support. You would have to write some kind of custom parser. 

We could probably help you with this if you decide to use the commercial version, but it's not something we have on the roadmap for Screenly OSE.

TeunvanderCruijsen, 2022-11-08:
> ATTENTION: This mail is from outside our organisation. Do not click on links or attachments if you do not know the sender or if you are not sure that the content is safe.

Hi guys,

I am using the commercial version of Screenly.
What would be possible then?

Met vriendelijke groet / Kind regards,
- - - - - - - - - - - - - - - - - - - - - - - - - - - -
Teun van der Cruijsen
Service Excellence Manager Winthontlaan 1 via Vebego Facility Solutions
Working flexible hours but I check my mail frequently
Follow us on LinkedIn<https://www.linkedin.com/company/winthontlaan1/>

Mobile: 06-13278785
Reception: 030-2074157
Winthontlaan 1, 3526 KV Utrecht
The Netherlands
www.winthontlaan1.nl<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.proofpoint.com%2Fv2%2Furl%3Fu%3Dhttp-3A__www.winthontlaan1.nl%26d%3DDwMFAg%26c%3DjozbAXBGpZCeJmn-Q9SThA%26r%3DdREKm1caPK2Xn4UiV1QgKgrPxCN_rkM_zjkUPNS1eXU%26m%3Ds4u6mBvdOWNGYMZk-Z09_a-4jjlbY2wRt0Qjy011On4%26s%3DPLyMYuPK7knwIE9Yz96ixJflfz2iovJBSlK1XhFzlZU%26e%3D&data=05%7C01%7Cteun.vandercruijsen%40yask.nl%7C96738a9dbf314ad0f15308da58e53dba%7C66f0a000775c40de8eff942c9ca690c8%7C0%7C0%7C637920038880918328%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=u9vYNmAN5eOWi9YoPIso062lTEzpnFFxls4DO9wAKgY%3D&reserved=0>
***@***.***

vpetersson, 2022-11-08:
Reach out to our support. We have something on the roadmap that we that could in theory solve this.

vpetersson, 2022-11-08:
Closing this issue out in favor of support ticket.

closed 2022-11-08 by vpetersson