Screenly/Anthias #1659: Older system assets fail to display, screen flashes

bmcwr, 2022-11-10:
Since 4th November have had problem with 2 older instances of Screenly - Mr Brownstone I think.
- System boots correctly, splash screen shows
- First asset runs at least for a time
- Then screen display is mainly black with partial flashes of white.

TOP shows python consuming lots of CPU - which I assume prevents the display function running.

pi: $ ps -eo pid,ppid,cmd,%mem,%cpu --sort=-%cpu | head
  PID  PPID     CMD                                      %MEM      %CPU
 3822     1     /usr/bin/python /home/pi/sc         2.1          78.5

PID of problem script is not consistent, keeps cycling upwards

**Removing network connection removes the problem.**

Have you changed something centrally that the clients are attempting to access over the last week?

**Environment**
Both systems on the same release (Mr Brownstone?) and both running on Pi3B. 
Systems are normally connected over WiFi but a test box also shows effect with wired Ethernet.

vpetersson, 2022-11-10:
I can't say with certainty, but with any installation older than a few years, there is a very high probability that the SD card has worn out. That can cause a ton of strange issues that are hard to pin down. 

As such, the first step would be to get a brand new SD card and start all over.

bmcwr, 2022-11-10:
Understand you comment but not in this case. Both systems failed on same day. (From the outset I installed log2ram to minimise writes).

Also I have written an old saved image to a new card and the same fault is seen, also when using that imaged card in a different Pi3B.

It seems to me that some Screenly python script isn't getting what it expects from the network interface.
No network = no problem

There is a wait.py in /bin that seems to check the network by going to a default location - not a SW person.

vpetersson, 2022-11-10:
If the code hasn't changed, then it logically must be related to some 3rd party resource. 

Could you try the latest beta disk image to see if that solves the issue?

bmcwr, 2022-11-10:
I wanted to try a latest image to see if that fixed everything but the curl method didn't work for me (on Buster lite). 
What is the latest install method to use?

vpetersson, 2022-11-10:
There are disk images posted. That would be the easiest way. If not, try Bullseye Lite.

ealmonte32, 2022-11-16:
> I wanted to try a latest image to see if that fixed everything but the curl method didn't work for me (on Buster lite). What is the latest install method to use?

@bmcwr , try Bullseye Lite and bash install method, this would give you the latest version and it would fix your issue.. if you experience an error at the end of the install script similar to [this](https://github.com/Screenly/screenly-ose/pull/1658), simply reboot the Pi and when back on run `./screenly/bin/upgrade_containers.sh` once and it should work.

Mr Brownstone is from 2017 and it was using the uzbl browser which would explain the issue with asset loading and memory.
And i am highly certain that when you unplug the network the issue goes away because the browser cant load anything thus not affecting the memory or running processes of the Pi.

P.S. not sure why you would want to be loading an image of screenly OSE from 2017...