Screenly/Anthias #1660: Terminal access

Tharrington86, 2022-11-16:
#### Overview of the Issue
With the latest release for RPi4 is there a way to step out of Screenly OSE to the terminal? I have read users having success on older versions by using Ctrl+Alt+F1 but that does not seem to function for my instance. I am trying to configure a static IP for my device but am unable since I cannot reach the terminal.

#### Environment

* Raspberry Pi Hardware Version: RPi 4B
* Raspberry Pi Network Setup:  Ethernet
* Screenly OSE Version: master@f6311f9

vpetersson, 2022-11-16:
That's expected. You can't use TTY1 as that is occupied by the player. Try Ctrl+Alt+F2.

closed 2022-11-16 by vpetersson

Tharrington86, 2022-11-16:
I tried F2 and got a black screen only. F1 resulted in my asset reappearing.

institut-agro-ead, 2023-03-31:
Hello,
Same problem here with Anthias (pi3) freshly installed on RPI3B via Raspberry Pi Imager.

vpetersson, 2023-03-31:
That's not a bug. It's expected. If you need terminal access, you need to use the manual installation on raspbian.

vpetersson, 2023-03-31:
Strange. Please do that installer again and see if any errors show up in the logs. We've tested it and it should work just fine.