Screenly/Anthias #1661: Set device and/or fleet configs programmatically.

nicomiguelino, 2022-11-16:


nicomiguelino, 2023-06-20:
@vpetersson, here's a sample command for setting fleet configs programmatically (via Balena CLI):

```bash
# balena env add $CONFIG_VARIABLE $VALUE --fleet $FLEET_NAME
balena env add RESIN_HOST_CONFIG_gpu_mem 512 --fleet anthias-pi4
balena env add RESIN_HOST_CONFIG_dtoverlay vc4-fkms-v3d --fleet anthias-pi4
```

For automation, these commands can be used in a shell script.

vpetersson, 2023-06-20:
> @vpetersson, here's a sample command for setting fleet configs programmatically (via Balena CLI):
> 
> ```shell
> # balena env add $CONFIG_VARIABLE $VALUE --fleet $FLEET_NAME
> balena env add RESIN_HOST_CONFIG_gpu_mem 512 --fleet anthias-pi4
> balena env add RESIN_HOST_CONFIG_dtoverlay vc4-fkms-v3d --fleet anthias-pi4
> ```
> 
> For automation, these commands can be used in a shell script.
> 
> I'll be closing this issue now, as this is a one-time setup, esp. for new fleets.

Yes, but these values are different for different devices.

nicomiguelino, 2023-06-20:
Thanks for pointing that one out, @vpetersson. But can the values be the same for devices of the same fleet?
(I can include docs for setting config values for a specific device).

closed 2024-07-27 by nicomiguelino