Screenly/Anthias #1666: Installation fails using the installation script: Docker fails to start

NassimBentarka, 2022-11-25:
#### Overview of the Issue

I'm experiencing the following issue on a fresh Raspbian Bullseye installation on Raspberry Pi Zero W. The issue also happens on a fresh Raspbian Buster installation as well.

When the [Installation Script](https://install-ose.srly.io) finishes, Screenly doesn't run after the reboot. When we check on the status of `docker.service` we get the following output:
```
pi@raspberrypi:~ $ sudo systemctl status docker
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com

Nov 25 13:17:30 raspberrypi systemd[1]: Dependency failed for Docker Application Container Engine.
Nov 25 13:17:30 raspberrypi systemd[1]: docker.service: Job docker.service/start failed with result 'dependency'.
Nov 25 13:24:57 raspberrypi systemd[1]: Dependency failed for Docker Application Container Engine.
Nov 25 13:24:57 raspberrypi systemd[1]: docker.service: Job docker.service/start failed with result 'dependency'.
```
After futher investigating, it looks like `docker.service` fails because it depends on `containerd.service`, whose `ExecStart=/usr/bin/containerd` command itself fails.

When we run `sudo /usr/bin/containerd` we get the following:
```
pi@raspberrypi:~ $ sudo /usr/bin/containerd
Segmentation fault
```

Now I found out that the docker fails to start because `/usr/bin/containerd` fails with a Segmentation Fault.

According to [Issue Comment 615203782](https://github.com/containerd/containerd/issues/4112#issuecomment-615203782) and [Issue Comment 615193596](https://github.com/moby/moby/issues/40733#issuecomment-615193596), the reason `/usr/bin/containerd` fails with a Segmentation Fault is that `docker-ce` using a Debian source instead of a Raspbian source to compile.

#### How to Fix (temporarily as of 11/24/2022 / or at least until my Pull Request is merged)

I have been able to have Containerd and Docker, and Screenly-OSE (of course) running by replacing the the source to be `raspbian` rather than `debian` on `/etc/apt/sources.list.d/docker.list` as you can see below:

From this:
```
pi@raspberrypi:~ $ cat /etc/apt/sources.list.d/docker.list 
deb [arch=armhf] https://download.docker.com/linux/debian bullseye stable
```
To this:
```
pi@raspberrypi:~ $ cat /etc/apt/sources.list.d/docker.list 
deb [arch=armhf] https://download.docker.com/linux/raspbian bullseye stable
```
After the replacement, make sure to double check the source is set to raspbian and NOT debian using `pi@raspberrypi:~ $ apt-cache policy containerd.io`

Then, run `sudo apt update && sudo apt upgrade`

Then, re-run the Screenly-OSE Installation Script using `bash <(curl -sL https://install-ose.srly.io)` and you should have Screenly running beautifully!

I am moving forward to creating a pull request to implementing the fix on the [Screenly-OSE Installation Script](https://install-ose.srly.io) right before `docker-ce` is installed, that way Screenly-OSE doesn't fail. I'll make sure the patch I'm adding is fail-safe and environment-independent.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Zero W
* Raspberry Pi Network Setup: Ethernet via USB Ethernet Adapter
* Screenly OSE Version: Screenly WebView 0.2.1 Commit # 0b6d493 (Stable)

Jayex007, 2023-06-17:
Please @NassimBentarka how do i change the source to raspbian on pi zero W

closed 2023-11-03 by NassimBentarka

nicomiguelino, 2023-11-03:
@NassimBentarka, thanks for the solution that you've provided! I'm yet to test that one out. I don't have a Pi Zero W, but I have a Pi 2 sitting around.

NassimBentarka, 2023-11-03:
It's my pleasure! Just make sure to use the right image on [WebView-v0.2.1](https://github.com/Screenly/Anthias/releases/tag/WebView-v0.2.1)

nicomiguelino, 2023-11-06:
@NassimBentarka, how were you able to flash the image that you've provided into the SD card. Did you use Raspberry Pi imager?

NassimBentarka, 2023-11-06:
> @NassimBentarka, how were you able to flash the image that you've provided into the SD card. Did you use Raspberry Pi imager?

Yes, I downloaded the image file from [WebView-v0.2.1](https://github.com/Screenly/Anthias/releases/tag/WebView-v0.2.1) and used Raspberry Pi imager to flash it as a custom image on 32GB SD Card. I booted on the RPi Zero W. Wifi Access Point works perfectly. MP4 1080p 30FPS resolves perfectly as well. If anyone is seeing difficulties decoding video, I recommend converting any files that are harder to decode to MJPEG even that the file size might get bigger, but it will be much simpler to decode before uploading them to Screenly.

I have been using it and it's very stable. I can state that [WebView-v0.2.1](https://github.com/Screenly/Anthias/releases/tag/WebView-v0.2.1) is the only latest version of Screenly that is stable and lightweight enough to run on RPi Zero W.

Hope that helps!

nicomiguelino, 2023-11-06:
Thanks for providing more details. I tried imaging the [WebView-0.2.1 release image](https://github.com/Screenly/Anthias/releases/download/WebView-v0.2.1/qt5-5.15.2-buster-pi1.tar.gz) using the Pi Imager, but I got the following error ("MBR does not have valid signature"):

[image]

NassimBentarka, 2023-11-06:
@nicomiguelino
Try this:
1. Open Gparted and create new partition table as GPT
2. Format your SD card first using the RPi Imager formatting feature.
3. Write image.

Also, since you have a RPi 2, use the pi2 image:
[image]

NassimBentarka, 2023-11-06:
@nicomiguelino

I forgot to add. It's very important to decompress the tar.gz before feeding it into Imager. Please only feed .img files to Imager just for consistency.

nicomiguelino, 2023-11-06:
Thanks, @NassimBentarka. I don't know specifically how to create a boot partition on a micro-SD card before imaging it. I tried GParted on WSL (my machine's running Windows), but it doesn't detect my micro-SD card. (Guess, I'll have to do some extra steps, or do some alternatives.)

I've only tried decompressing the archive using WinRAR, but I can try decompressing it via the command-line (WSL).

nicomiguelino, 2026-01-28:
Thank you, @NassimBentarka. That helpful piece of information could be included in the documentation.