Screenly/Anthias #1670: Not all Screenly ose is operationnal after installation via balena on a raspberry 3

forum6691, 2022-11-28:
#### Overview of the Issue
I installed the screenly OSE on a raspberry 3 in folowing this procedure
https://www.balena.io/blog/deploy-free-digital-signage-software-screenly-ose/
All seems OK, but at the end, nothing appears ont the screen
I try to connect screenly web server directly on the local network with http://localadress:8080, but no answer.
I activate via balena the public adress and I succeed to access to the screenly main screen !
I chose 2 actives assets like weather and clock and wait for the display but nothing appears again !
[image]

[image]

#### Environnent

* Raspberry Pi Hardware Version:
* Raspberry Pi Network Setup:  built in Ethernet and USB WiFi Adapter installed, but I use the copper ethernet for this first test.
* Screenly OSE Version: last source get on git and then compile by balena processus
* micro SD card 16 Go, speed 10 model intenso

**I have these logs  and it seems ose-server has a problem: "unable to get local git branch and unable to get latests version from GitHub"**

An idea of the problem ? Thank you.

<redis> 1:M 28 Nov 2022 17:51:28.926 # Server initialized

<redis> 1:M 28 Nov 2022 17:51:28.929 * Ready to accept connections

<srly-ose-celery> [2022-11-28 17:52:21,989: INFO/Beat] beat: Starting...

<srly-ose-celery> [2022-11-28 17:52:22,276: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 17:52:22,424: INFO/Beat] Scheduler: Sending due task cleanup (server.cleanup)

<srly-ose-celery> [2022-11-28 17:52:22,433: INFO/MainProcess] Received task: server.get_display_power[2b323b85-4ab5-458b-9778-2b7a46f73678]  

<srly-ose-celery> [2022-11-28 17:52:22,442: INFO/Beat] Scheduler: Sending due task cleanup_usb_assets (server.cleanup_usb_assets)

<srly-ose-celery> [2022-11-28 17:52:22,469: INFO/MainProcess] Received task: server.cleanup[91e466fb-a95f-4d45-873e-0e47c2083140]  

<srly-ose-celery> [2022-11-28 17:52:22,500: INFO/MainProcess] Received task: server.cleanup_usb_assets[5d28386a-d484-4a88-a919-1a36b41a26f4]  

<srly-ose-celery> [2022-11-28 17:52:22,617: INFO/ForkPoolWorker-2] Task server.cleanup_usb_assets[5d28386a-d484-4a88-a919-1a36b41a26f4] succeeded in 0.075033016s: None

<srly-ose-celery> [2022-11-28 17:52:23,601: INFO/ForkPoolWorker-5] Task server.cleanup[91e466fb-a95f-4d45-873e-0e47c2083140] succeeded in 1.099758138s: None

<srly-ose-celery> [2022-11-28 17:57:22,266: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 17:57:22,292: INFO/MainProcess] Received task: server.get_display_power[a1351413-0c1a-4cc4-b43a-249d445c2b14]  

<srly-ose-celery> [2022-11-28 18:02:22,364: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:02:22,390: INFO/MainProcess] Received task: server.get_display_power[b946b203-e72b-409b-8ee4-b3c4485b2478]  

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-server> Unable to get local Git branch

<srly-ose-server> Unable to get latest version from GitHub

<srly-ose-celery> [2022-11-28 18:07:22,399: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:07:22,426: INFO/MainProcess] Received task: server.get_display_power[2cda96bf-448f-402c-ba68-fd467c9d893d]  

<srly-ose-celery> [2022-11-28 18:12:22,431: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:12:22,459: INFO/MainProcess] Received task: server.get_display_power[e73bd279-a9b1-4428-bbca-1b065ccd0f92]  

<srly-ose-celery> [2022-11-28 18:17:22,529: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:17:22,552: INFO/MainProcess] Received task: server.get_display_power[c77f11ff-967a-400d-87fa-706fe86525b4]  

<srly-ose-celery> [2022-11-28 18:22:22,626: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:22:22,650: INFO/MainProcess] Received task: server.get_display_power[3a67b84e-e763-4556-ad9d-e8e88e309f68]  

<srly-ose-celery> [2022-11-28 18:27:22,642: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:27:22,667: INFO/MainProcess] Received task: server.get_display_power[05179d00-2dc6-490b-bca5-2f4540adb08d]  

<srly-ose-celery> [2022-11-28 18:32:22,739: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:32:22,764: INFO/MainProcess] Received task: server.get_display_power[dff377e1-353f-4773-bede-adfe17bc5e22]  

<srly-ose-celery> [2022-11-28 18:37:22,836: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:37:22,861: INFO/MainProcess] Received task: server.get_display_power[098c8ee6-cb7a-4bcb-af5b-786c49ea260b]  

<srly-ose-celery> [2022-11-28 18:42:22,893: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:42:22,919: INFO/MainProcess] Received task: server.get_display_power[cc60f3f4-54cb-442a-8dd0-2da3f823b239]  

<srly-ose-celery> [2022-11-28 18:47:22,986: INFO/Beat] Scheduler: Sending due task display_power (server.get_display_power)

<srly-ose-celery> [2022-11-28 18:47:23,014: INFO/MainProcess] Received task: server.get_display_power[ec283423-1be4-49f0-ab20-dc743e7e878a]

vpetersson, 2022-11-28:
Those instructions are a bit out of date. The recommended installation method can be found [here](https://github.com/Screenly/screenly-ose/#disk-images).

closed 2022-11-28 by vpetersson

forum6691, 2022-11-28:
Hi Victor, please to read again on this project.  We exchange 4 years ago n always on sceeenly OSE. As you see, I continue to use it.

Before to choose to test with Balena, I tried the disk images for RPI3 on your link and transfer directly on the sd card, but I can open the local adress and screenly did'nt started.
So I try balena following the instructions out of date. 

I will try your new method and come back to tell you. Thank you very much for your quick and precise answer.

forum6691, 2022-11-28:
I installed via the link https://hub.balena.io/organizations/screenly_ose/fleets/screenly-ose-pi3 and choosed button "Get started".
I create the image automtically and push on the sd card with balenaEtcher.
I put the sdcard in the raspberry 3, started it. I get the balena picture on the screen,... and after what must I do to link this raspberry in my balena dashboard ?  How to do to recognized the instance on my raspberry with my account on balena ? I didn't find any documentation on this ?
Thanks for help.