Screenly/Anthias #1671: Screenly installed with Balena doesn't show anything

forum6691, 2022-11-29:
Me again.

#### Overview of the Issue
Following advice given yesterday on this site ( #1670), I installed Screenly use link found in this page:
https://github.com/Screenly/screenly-ose
and in clicking on the button "Deploy with Balena" https://dashboard.balena-cloud.com/deploy?repoUrl=https://github.com/screenly/screenly-ose&defaultDeviceType=raspberrypi3

I choose my fleet and the release appeared on my balena dashboard
Then I add a device via balena dashboard, created an image for the Sdcard with balena etcher , put the card in the raspberry and booted it. The balena picture appeared.

After many minutes of dowloading the different screenly services on the raspberry, All services started.
[image]

At this stage of progress, I try to access to screenly on my local lan without success.
So I active the "Public device URL" and open the link given. 
I succeeded to display screenly controls. I added some active assets (webpages and an jpeg picture), set the time and date and wait , nothing appeared on the display of the raspberry except always the balena picture.

[image]

#### Environment
    Raspberry Pi Hardware Version:
    Raspberry Pi Network Setup: built in Ethernet and USB WiFi Adapter installed, but I use the copper ethernet for this first test.
    Screenly OSE Version: last source get on git and then compile by balena processus
    micro SD card 16 Go, speed 10 model intenso

I copy the log from balena below:

[balena log-29.11.22_06 31 42_(+0100).txt](https://github.com/Screenly/screenly-ose/files/10110060/balena.log-29.11.22_06.31.42_.%2B0100.txt)

**Any help will be appreciated to know where is the problem in this configuration ?**

forum6691, 2022-11-29:
Following operations to choose RPI 2 instead RPI 3 , I have exactly the same problem, only the picture Balena is displayed on screen
See the log below:

[valette_holy-tile-29.11.22_07 46 11_(+0100).txt](https://github.com/Screenly/screenly-ose/files/10110420/valette_holy-tile-29.11.22_07.46.11_.%2B0100.txt)

An idea of the problem

vpetersson, 2022-11-29:
Probably related to some settings. This method isn't fully supported yet. Currently, we only support using Balena Hub or installing from source on Raspberry Pi OS. 

You might be able to get it to work by tweaking `gpu_mem` in Balena, but the deployment to Balena requires a fair bit of work. You can see how we deploy to Balena Hub [here](https://github.com/Screenly/screenly-ose/blob/master/.github/workflows/docker-build.yaml) for inspiration.

forum6691, 2022-11-29:
Hi Victor.
You said: _"his method isn't fully supported yet. Currently, we only support using Balena Hub,"_ ,but I deployed with balena hub to create the release, then added the RPI  also with balena hub.
 
Is there an other method with balena hub different from this I used ?

vpetersson, 2022-11-29:
No, you used Balena. That's not the same as Balena Hub. Use [these](https://github.com/Screenly/screenly-ose#disk-images) disk images or go to [hub.balena.io](https://hub.balena.io) and pick the corresponding version there.

forum6691, 2022-11-29:
Weird subtleties, but why not.
So I went to https://hub.balena.io/organizations/screenly_ose/fleets/screenly-ose-pi2 and click to the button "Get started". Then propose the copy of the distribution and creation automatically with balena etcher on my sdcard. 
This stage was OK. 
I put the sd car in the RPI 2 and booted. After some seconds, the balena picture appeared, some seconds after, it disappeared after seconds again, it appeared again and stays definitively.

How to access screenly ? I try the local IP adress:8080 without success (I know my ip adress is the same given by my DHCP).
 
I look in my fleet on balena cloud, but nothing appears (it's normal because no association with my fleet was done I think.)

Maybe there is a know bug with this procedure ?

BrendonKoz, 2023-01-13:
@forum6691 I just noticed an update pending for my balena-controlled devices, and also tried updating one of our two devices via the BalenaHub. It worked, and like you I can access it locally on the same network, but there's no visibility into its remote IP (if one exists). After finding and reading [this page](https://blog.balena.io/introducing-open-fleets-and-self-submitted-apps-and-blocks-on-balenahub/), it appears as though what we're both looking for is not available in the current iteration.

I've gone back to the balena CLI method of pushing code updates to my devices. I never figured out how to properly update that way though, so although I've properly rebuilt and assigned to my device, the running version is still telling me there's an update available (and the UI is not updated to Anthias). Work-in-progress. I mostly wanted to reply to clarify how the Balena Hub seems to work (and doesn't work).

forum6691, 2023-01-17:
> @forum6691 I just noticed an update pending for my balena-controlled devices, and also tried updating one of our two devices via the BalenaHub. It worked, and like you I can access it locally on the [...]

Thanks Brendon for your answer and explanation.
I decided to stop try to use Balena to deal with my OSE screenly and install a classical version (OS then Anthias). At the moment this is the surely method for me.

BrendonKoz, 2023-01-18:
I stepped through the instructions from the original [Balena/Screenly-OSE setup instructions](https://blog.balena.io/deploy-free-digital-signage-software-screenly-ose/) for a manual Fleet setup and install. Unfortunately, it seems that the current iteration of Anthias' setup uses bind mounts which Balena doesn't allow. Attempting to use `balena push` from within a cloned/downloaded repo results in an error:
> bind mounts are not allowed
I have to imagine that there must be a way to make it run though since the current BalenaHub fleet is up-to-date with the main repo branch.
I'd (personally) be fine using the BalenaHub deployment method so long as there was still a way to attempt a safe reboot/shutdown if the system becomes unstable, *and* if a public URL could be enabled (and known).

vpetersson, 2023-01-23:
> I stepped through the instructions from the original [Balena/Screenly-OSE setup instructions](https://blog.balena.io/deploy-free-digital-signage-software-screenly-ose/) for a manual Fleet setup and [...]

These instructions were removed because they were outdated. If you really want to run it yourself, you would currently have to reverse engineer the process we use to deploy to Balena Hub, take a look at [docker-build.yaml](https://github.com/Screenly/Anthias/blob/master/.github/workflows/docker-build.yaml).

> I'd (personally) be fine using the BalenaHub deployment method so long as there was still a way to attempt a safe reboot/shutdown if the system becomes unstable, and if a public URL could be enabled (and known).

The shut down/reboot feature should be working fine. It is using native Balena features.

As far as the public URL goes, this is disabled by default. Looking over the [Device API](https://docs.balena.io/reference/supervisor/supervisor-api/#http-api-reference), there doesn't seem to be a way to toggle this from the device directly unfortunately.

BrendonKoz, 2023-01-24:
> These instructions were removed because they were outdated. If you really want to run it yourself, you would currently have to reverse engineer the process we use to deploy to Balena Hub, take a [...]

Ah, I guess if they were removed, they were removed from the Screenly/Anthias side. I wasn't aware they were outdated; thank you for clarifying that - and thank you for providing the docker-build.yaml file as a reference!

> The shut down/reboot feature should be working fine. It is using native Balena features.

Unfortunately, at least when I tried the BalenaHub/Fleet option, the device was added to the community fleet, not to my own account's interface, so I had no access to controls to reboot or shutdown the device from Balena; at least none that I could find. For most scenarios I'd imagine this isn't a concern and having a constantly up-to-date version is far more preferred to any other limitation(s).

Unless I have further progress to report back with, I'll stop commenting as I feel this answers all of my questions. Thank you all!

vpetersson, 2026-05-08:
Pre-rebrand report against Screenly OSE on balena Pi 2B. Anthias has since dropped Pi 2 entirely and split the webview into Qt5 (Pi 3/4) and Qt6 (Pi 5/x86). Closing as stale.