Screenly/Anthias #1679: The viewer container keeps restarting due to issues related to VLC.

nicomiguelino, 2022-12-06:
#### Overview of the Issue

The display shows a black screen.

Here's what I got upon checking the viewer container:

```
./bin/start_viewer.sh: line 36: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
vlc: unknown option or missing mandatory argument `--mmal-display=HDMI-2'
Try `vlc --help' for more information.
Traceback (most recent call last):
  File "viewer.py", line 60, in <module>
    media_player = VLCMediaPlayer() if 'Raspberry Pi 4' in get_raspberry_model() else OMXMediaPlayer()
  File "/usr/src/app/lib/media_player.py", line 35, in __init__
    self.player = self.instance.media_player_new()
AttributeError: 'NoneType' object has no attribute 'media_player_new'
./bin/start_viewer.sh: line 54: kill: (37) - No such process
./bin/start_viewer.sh: line 36: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
vlc: unknown option or missing mandatory argument `--mmal-display=HDMI-2'
Try `vlc --help' for more information.
Traceback (most recent call last):
  File "viewer.py", line 60, in <module>
    media_player = VLCMediaPlayer() if 'Raspberry Pi 4' in get_raspberry_model() else OMXMediaPlayer()
  File "/usr/src/app/lib/media_player.py", line 35, in __init__
    self.player = self.instance.media_player_new()
AttributeError: 'NoneType' object has no attribute 'media_player_new'
./bin/start_viewer.sh: line 54: kill: (38) - No such process
```

#### Reproduction Steps

```bash
cd ~/screenly
./bin/upgrade_containers.sh # pulls the images and start the containers
docker logs -f screenly-srly-ose-viewer-1
```

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B
* Raspberry Pi Network Setup: Ethernet
* OS: Raspberry Pi OS Lite (Bullseye), 64-bit

nicomiguelino, 2022-12-06:
@vpetersson

https://github.com/Screenly/screenly-ose/blob/09a31df5b11fa0c8a4cf874c38ae2c2c7cd14dfe/lib/media_player.py#L29-L37

`self.instance` becomes null probably because the `--mmal-output` and `--vout` are not recognized as options.

https://github.com/Screenly/screenly-ose/blob/09a31df5b11fa0c8a4cf874c38ae2c2c7cd14dfe/lib/media_player.py#L39-L46

Here's a temporary workaround:
- As a workaround, we can remove `--mmal-output` and `--vout` from the options. Doing so will resolve the issue and unblock other features. However, #1590 and #1651 needs to be re-opened.

I can't exactly remember when, but weeks ago, I can still see the `--mmal-output` and `--vout` from the help info (via `vlc -H`). Could it be that VLC made breaking changes? Here's what can be done next...
- Research recent VLC release notes and relevant forums. If the root cause is relevant to VLC, we can
  - Use an older version of the `vlc` package, or
  - compile VLC from source &ndash; We can use an older commit for this one

vpetersson, 2022-12-06:
Maybe we just revert back to using `omxplayer` on Pi 4. The initial reason for using `vlc` was because `omxplayer` didn't support Pi 4 at all.

The trade-off is that `omxplayer` still doesn't support 4k, but that's a trade-off I'm willing to live with for now.

Some relevant reading:

* https://github.com/popcornmix/omxplayer/issues/716

nicomiguelino, 2022-12-07:
Nice find, thanks! I still can't make the `OMXMediaPlayer` play a video asset, but *I'm almost there*.

I was able to make an MP4 video playback work (by hand) by doing the following:
- Get into the viewer container's shell by running `docker exec -it screenly-srly-ose-viewer-1 bash`.
- Go to the path where a `.mp4` file is located.
- Run `omxplayer --display=2 [filename.mp4]`.

But it's surprising that it's not playing [`/tests/assets/asset.mov`][1], for instance.

[1]: https://github.com/Screenly/Anthias/blob/c85dddd783f71a6a425b7b2217d0a2a9c3a97f19/tests/assets/asset.mov

nicomiguelino, 2023-12-27:
Re-opening this since the issue persists again...

PR #1684 (2022-12-09): Fix VLC video playback issues on Raspberry Pi 4. :wrench:
Fixes #1679 (temporarily).
closed 2023-12-28 by nicomiguelino