Screenly/Anthias #1691: Installation Problem on Raspberry PI 3 - Docker Timeout

michaelortnersanube, 2022-12-13:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Installation failure on ansible step "Install Docker"

#### Reproduction Steps

Steps to reproduce this issue, eg:

I have installed raspberry pi image over the raspberry pi imager, then i insert it into the raspberry and fill out the installation process, keyboard language german, than i have to add a user, name "pi" and then the password for the user. Then i have activated over raspi-config the ssh deamon and connect over ssh.

At least i have make a sudo apt update && sudo apt upgrade -y

Now i have used the bashscript and select all three questions with "Y"

The installation begin. And then on the Task " Install Docker "

this error come:

[image]

#### Environment

* Raspberry Pi Hardware Version:
Raspberry PI 3
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
LAN
* Screenly OSE Version:
From GitHub Repo the latest with Debian 11 Bullseye no desktop environment

vpetersson, 2022-12-13:
Can you please paste that as text rather than a screenshot as it's almost impossible to read?

michaelortnersanube, 2022-12-13:
yes i can - but a very strange thing: I have rebooted the raspberry pi, and now it pulls the docker images. Is this maybe a onetime bug? If you want, i can make a second sd card and try again a fresh install like i have wrote. Then i can send you the text. If you need more details of the hardware please send me all things what you need. :)

vpetersson, 2022-12-13:
Well, if it's a timeout error, it's definitely possible that the SD cards is too slow. We only use industrial Class 10 SD cards for all our players. If you're using anything other than a Class 10 SD card, then you might want to get one of these.

willworkforicecream, 2022-12-13:
I've been able to replicate this (or some other issue, hard to read) on a few different Pis and a few different SD cards over the last week or two. 

Running the script a second time consistently resolves the issue.

vpetersson, 2022-12-13:
@willworkforicecream this likely a function of the docker images being rather big at the moment. As such, they will require a decent amount of data to be written to disk. If you're using a slower SD card, it's likely going to time out. 

That said, the error you posted isn't from the docker image pull -- it's from the installation of docker.

A few questions:

* What else is installed on the device (or is it a fresh install)?
* What SD card are you using?

willworkforicecream, 2022-12-13:
Fresh install. This specific time I'm installing it on a SanDisk Ultra Plus class 10 that I tested a write speed to 22 MB/s earlier this week. I had the same thing happen on a Patriot class 10.  I'm going to try it again with a SSD to see if I can rule out any SD card shenanigans.

willworkforicecream, 2022-12-13:
It does look like a case of not all SD cards being created equally, even those [claiming](https://pibenchmarks.com/benchmark/64939/) to be class 10. No issues with when installing on an [SSD](https://pibenchmarks.com/benchmark/64937/).

vpetersson, 2022-12-13:
Oh definitely. SD cards very a ton in performance in quality. We've had a lot of conversations about this with the folks over at Balena over the years and we've come to the same conclusion. The most reliable ones are SanDisk's (now Western Digital) Industrial cards. Short of that, the Samsung ones are also decent.

ealmonte32, 2023-02-08:
@michaelortnersanube 
Just checking to see if you can close this Issue.

closed 2023-02-09 by michaelortnersanube