Screenly/Anthias #1694: Hung on Screenly Logo

issacboil, 2022-12-14:
We are running the latest version of Anthias and whenever we plug it into Ethernet, It works and gets a IP. But then we unplug it from Ethernet, It just gets hung on the Black Display with the Screenly Logo. We need to have this Pi work on Wifi.

MisterAnderson91, 2023-01-17:
#1628 seems to be resolved but I am still seeing the problem, both on the pre-built image and on a fresh install with the script.

Assuming the image doesn't have the fix yet, but the install via the script fails to prompt for any wifi setup, it just shows the last known (and disconnected) LAN IP address and never prompts for wifi setup.

Also #1628 talks about Balena, is that relevant to the other install methods?

vpetersson, 2023-01-17:
@MisterAnderson91 what hardware are you using?

MisterAnderson91, 2023-01-17:
A raspberry pi 4 4GB. I used the latest image of raspberry pi os lite, downloaded yesterday, then ran the script-installer. I used USB tethering from a phone to get it internet during the script install to avoid any interference from existing wifi settings and unplugged just before accepting the Y/n reboot option at the end of installation. On reboot, it acts like that previous connection still exists and never starts the wifi AP, just showing the previous IP address.

This is the same behaviour seen throughout the last year, no apparent difference.

vpetersson, 2023-01-18:
So Anthias is using Network Manager (NM) for managing the wireless connectivity. The best way to debug this is to use `nmtui` from the command line (or `nmcli` if you're more advantageous). You should be able to add/edit your various network connections using that.

MisterAnderson91, 2023-01-23:
The problem is the configuration of this enterprise network. nmtui doesn't support it and it's a nightmare to get right in nmcli. It is super easy to connect with the old Screenly OSE wizard though. The easy wifi setup was the primary attraction of Screenly OSE.

vpetersson, 2023-01-23:
I don't get this part. The WiFi Connect used only supports very basic scenarios to my knowledge. For anything enterprise-y (e.g. 802.1x or similar), you would still have had to resort to configuring Network Manager directly. Could you provide some details on this?

MisterAnderson91, 2023-01-23:
In the last major release before the hiatus (2020, worked on pi3 and below) after first boot with no ethernet, you could connect to the wifi AP. When it scanned it definitely picked up enterprise wifi networks and allowed you to connect with a username and password and it must have autconfigured the rest. It was fantastic until pi3 stocks sold out and the lack of pi4 compatibility

vpetersson, 2023-01-23:
OK, that does indeed sounds like 802.1x was supported. In any case, nothing has changed. The code is still there. In the last month or so @nicomiguelino has been working hard on restoring a lot of functionality. We're committed to restoring the former glory (and then some). 

This functionality should in theory already have been restored, so it's likely a bug somewhere. We'd need to get some steps/logs for reproducing it in order to fix it.

MisterAnderson91, 2023-01-23:
I saw in another issue that the current build only shows the previous IP address until upgrade_containers.sh is run. That may also mean that the wifi AP script is also getting ignored and not activated? Because neither the initial install script, nor upgrade_containers can be run without an active network connection.

Maybe a new pre-built image with the recent fixes could help?

vpetersson, 2023-01-23:
> I saw in another issue that the current build only shows the previous IP address until upgrade_containers.sh is run. That may also mean that the wifi AP script is also getting ignored and not [...]

No these are completely unrelated. The IP variable is only tracked to show it on the splash screen.

> Maybe a new pre-built image with the recent fixes could help?

Good idea. I've triggered a [new disk image build](https://github.com/Screenly/Anthias/actions/runs/3985171858).

MisterAnderson91, 2023-01-24:
I've imaged with 2023-01-23-raspberrypi4-64.zip and tested. On first boot, you get the Anthias full-screen logo, followed shortly by a white page with "Internal Server Error", followed by the black Screenly logo screen.

Left it for 30 minutes and rebooted by unplugging the power. Second boot gives the Anthias logo, now followed by the "to manage content on this screen, just point your browser to: http://192.168.42.1", followed by the black Screenly logo screen again.

I hadn't thought to check previously without the onscreen prompt but at least with this image, I can see the wifi network "Anthias WiFi Connect" appear and can connect to it. My android phone detects no internet connection and attempts to load a captive portal page by connecting to connectivitycheck.gstatic.com but it just gives a page that says "Sorry, this page does not exist!". Not sure if that's an android page or not. 

However I can select "Use this network as-is", and in the browser 192.168.42.1 loads the management interface immediately. I'm able to add an asset and have it display (although I don't know what time the OS thinks it is; I had to just set the start time back a year). Unfortunately, I see no way to set the time, and no way to access the wifi setup wizard (or any wifi settings really) from the web interface. So I'm unable to choose an existing wifi network to connect to.

I am going to redo the script-installation method and see if the wifi AP appears as I never checked for it with that method.