Screenly/Anthias #1698: Failed install on RaspberryPi 2

och29, 2022-12-15:
#### Overview of the Issue

I had no success with the Balena image so tried via the script with a clean install of Raspian 
The script failed too. I think, first because Docker wasn't installing (an issue's been raised already) and then the same but for Ansible.
Essentially the script would complete within a couple of minutes but not pull anything

Log after I manually installed Docker but before Ansible:

>       _____                           __         ____  _____ ______
>       / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
>       \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
>      ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
>     /____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
>                                      /____/
> Screenly OSE requires a dedicated Raspberry Pi / SD card.
> You will not be able to use the regular desktop environment once installed.
> 
> Do you still want to continue? (y/N)
> Do you want Screenly OSE to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)
> 
> Would you like to perform a full system upgrade as well? (y/N)
> + REPOSITORY=https://github.com/screenly/anthias.git
> + '[' '!' -d /home/pi/screenly ']'
> + sudo mkdir -p /etc/ansible
> + echo -e '[local]\nlocalhost ansible_connection=local'
> + sudo tee /etc/ansible/hosts
> + '[' '!' -f /etc/locale.gen ']'
> + sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
> + sudo apt update -y
> Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
> Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
> Hit:3 https://download.docker.com/linux/raspbian bullseye InRelease
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> All packages are up to date.
> + sudo apt-get install -y --no-install-recommends git libffi-dev libssl-dev python3 python3-dev python3-pip whois
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> git is already the newest version (1:2.30.2-1).
> libffi-dev is already the newest version (3.3-6).
> python3 is already the newest version (3.9.2-3).
> python3-dev is already the newest version (3.9.2-3).
> whois is already the newest version (5.5.10).
> libssl-dev is already the newest version (1.1.1n-0+deb11u3+rpt1).
> python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> + '[' y == y ']'
> + export MANAGE_NETWORK=true
> + MANAGE_NETWORK=true
> + sudo apt-get install -y network-manager
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> network-manager is already the newest version (1.30.6-1+deb11u1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> + '[' master = master ']'
> ++ curl -s https://raw.githubusercontent.com/screenly/anthias/master/requirements/requirements.host.txt
> ++ grep ansible
> + ANSIBLE_VERSION=ansible-core==2.12
> + sudo pip install cryptography==38.0.2
> Traceback (most recent call last):
>   File "/usr/bin/pip", line 33, in <module>
>     sys.exit(load_entry_point('pip==20.3.4', 'console_scripts', 'pip')())
>   File "/usr/bin/pip", line 25, in importlib_load_entry_point
>     return next(matches).load()
>   File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
>     module = import_module(match.group('module'))
>   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
[... 112 lines cut ...]
> + sudo find /usr/share/locale -type f '!' -name en '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -delete
> + sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' '!' -name locale.alias -exec rm -r '{}' ';'
> + sudo chown -R pi:pi /home/pi
> + '[' '!' -f /etc/sudoers.d/010_pi-nopasswd ']'
> ++ git rev-parse --abbrev-ref HEAD
> fatal: not a git repository (or any of the parent directories): .git
> ++ git rev-parse --short HEAD
> fatal: not a git repository (or any of the parent directories): .git
> ++ lsb_release -a
> No LSB modules are available.
> + echo -e 'Screenly version: @\nDistributor ID: Raspbian
> Description:    Raspbian GNU/Linux 11 (bullseye)
> Release:        11
> Codename:       bullseye'
> + '[' false = false ']'
> + set +x
> Installation completed.
> You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)
> 

#### Environment

* Raspberry Pi Hardware Version: Pi 2, class 10 Samsung SD
* Raspberry Pi Network Setup: USB WiFI adapter
* Screenly OSE Version:

vpetersson, 2022-12-15:
(reformatted the logs with code block for improved readability)

So the issue really starts with `sudo pip install cryptography==38.0.2`. This is a pretty heavy library to compile, and it might be the case that it can't be compiled on a Pi 2 due to lack of RAM. Hence, if there is now Python Wheel available, it will choke.

Could you try running this command stand-alone to make sure it works? If that still fails, try `sudo pip install cryptography`

vpetersson, 2023-05-22:
@vordenken Answering `yes` to this question switches the network to use Network Manager. It is recommended to use Network Manager (using `nmtui` or `nmcli`) to manage networking these days as it is a lot more flexible than alternatives. 

In theory, if you used WiFi before, the device should pick up on this and spin up a hotspot for WiFi configuration upon the next boot.

nicomiguelino, 2024-07-23:
Feel free to reopen this when the issue persists again.

closed 2024-07-23 by nicomiguelino