Screenly/Anthias #1701: Screen only shows Anthias logo

FabianH8, 2022-12-25:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

My screen just shows the Anthias logo with a purple background.

#### Reproduction Steps

* Start the Raspberry Pi

#### Environment

* Raspberry Pi Hardware Version: 
* Raspberry Pi Network Setup: Ethernet
* Screenly OSE Version: Anthias

nicomiguelino, 2022-12-25:
Hi @FabianH8, I would like to ask some more questions about the issue:
- How did you install Anthias (formerly Screenly OSE)?
  - Did you install it via the [Raspberry Pi Imager](https://github.com/Screenly/Anthias#disk-images), via [BalenaHub](https://github.com/Screenly/Anthias#disk-images), or via the [install command](https://github.com/Screenly/Anthias#installing-on-raspbianraspberry-pi-os) `bash <(curl -sL https://install-ose.srly.io)`?

FabianH8, 2022-12-26:
I tried it via Balena and via the Raspberry Pi Imager. When I installed it via Balena, the Balena logo was still there after a long time. When I installed it via the Raspberry Pi Imager, it just showed the Anthias logo with a purple background.

nicomiguelino, 2022-12-29:
Hi @FabianH8. I just tried installing Anthias via the Raspberry Pi Imager on my Raspberry Pi 4 three times. On average, it took about 5 to 7 minutes for the splash screen (with the IP address of the device) to show up.

There could be other factors why it's taking too long. It could be because of Internet connectivity. It could be because of other reasons.

I'll test the installation on a Raspberry Pi 3 (Model B+). Stay tuned for more updates.

nicomiguelino, 2022-12-29:
@FabianH8, what Pi version are you using?
The SD card used might be a factor as well. It's recommended to use a Class 10 micro SD card. More details could be found [here](https://github.com/Screenly/Anthias#installing-on-raspbianraspberry-pi-os).

FabianH8, 2022-12-30:
@nicomiguelino I use the Raspberry Pi B+ with a Class 10 micro SD card.

vpetersson, 2022-12-30:
If it's the original pi b+ it might be a bit before we get that to work as it has even less resources than a pi zero IIRC.

godfather55v, 2023-02-22:
I'm having the same issues w/ the pi configurator.  I've tried several SD card's and raspberry pi b+ boards all just giving me the logo on boot.  I am able to load the management interface and add assets, but nothing ever displays.

ndkp52, 2023-02-23:
Hello,

I have the same problem with a Pi4.
Everything seems to work except the display itself.
Only the Anthias By Screenly logo appears 

Thanks for your work

vpetersson, 2023-02-23:
We've pushed a fix for this that *should* resolve this issue. Unfortunately this time Balena is [acting up](https://forums.balena.io/t/frequently-stuck-in-build-in-progress/366916), preventing this release to be pushed out to the fleets.

ndkp52, 2023-02-23:
Ok thank's ! 
I hope you will be able to push this maj

ndkp52, 2023-02-24:
small precision: 
Server installed with an image from GitHub "2022-12-12-raspberrypi4-64" .
SD Card flash with Raspberry Pi Imager

Thank's

godfather55v, 2023-02-24:
fyi, i'm new to github...

i grabbed a fresh .zip of the 0.18.6 version the file was 2023-02-17-raspberrypi3.  I flashed my memory card w/ Pi imager.  It is still doing the same thing, am i doing something wrong, or are we still waiting for your change to propagate?

many thanks!

brunomr88, 2023-02-28:
I tried to install Anthias on a Raspberry PI 3 using PI Manager, and it keeps stopping at the Anthias Logo.

vpetersson, 2023-03-06:
This should no longer be an issue as the underlying issue has been resolved.

closed 2023-03-06 by vpetersson

brunomr88, 2023-03-06:
> Isso não deve mais ser um problema, pois o problema subjacente foi resolvido.

Here I am still on the screen with the Anthias logo, we downloaded the v0.18.6 version from Github, without success.
`https://github.com/Screenly/Anthias/releases/download/v0.18.6/2023-03-06-raspberry-pi2.zip`
Installed via Raspberry Pi Imager, SD 32gb class 10, Raspberry Pi 2

vpetersson, 2023-03-06:
There might still be issues on legacy Pis, like the Pi 2. We've only confirmed everything on Pi 3 and Pi 4.

brunomr88, 2023-03-07:
> There might still be issues on legacy Pis, like the Pi 2. We've only confirmed everything on Pi 3 and Pi 4.

Installed on Raspberry Pi 3, same problem, does not leave the screen with the Anthias logo

vpetersson, 2023-03-07:
We're unable to reproduce this. Flashed out the latest release to a Pi 3 yesterday without issues.

brunomr88, 2023-03-07:
> We're unable to reproduce this. Flashed out the latest release to a Pi 3 yesterday without issues.

Installing via Raspberry Pi Imager, where can you find Anthias on another special purpose OS?

vpetersson, 2023-03-07:
Perhaps a caching issue. Pull the image directly from our release here for now. That said, the device should self-update if it has ethernet.