Screenly/Anthias #1713: Splash Screen Displays old IP

cedgeit, 2023-01-17:
#### Overview of the Issue
We set up an Anthis OSE installation in our office.  All went well.  We took it to the site where it will be deployed (different subnet and it connects fine and works.  However, the splash screen that says how to connect still has the IP address from our office even though it is invalid and a completely different subnet.  We are able to connect to the device using the IP that was assigned by their DHCP server on the LAN and Wifi.  This does not affect the functionality of screenly.

#### Reproduction Steps

Steps to reproduce this issue, eg:
- Burn SD card using Raspberry Pi Imager 1.7.3 on Fedora Linux with Advanced Options enabling SSH (although this did not work..needed touch ssh in boot.  Other settings were there via firstboot,sh)
- Initial setup on one subnet loading OS Lite Bullseye
- Full Upgrade of the OS
- Install screenly using: bash <(curl -sL https://www.screenly.io/install-ose.sh)
- Power Off and move to a different network and subnet.
- Splash Screen shows the IP of the original subnet.
-
#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Model Raspberry Pi 3 Model B Rev 1.2
* Raspberry Pi Network Setup: LAN and WIFI
* Screenly OSE Version:  master@2a0e51e
* OS Raspberry Pi OS Lite Bullseye

nicomiguelino, 2023-01-17:
Hello @cedgeit, thank you for raising this issue!

The IP address of the captive portal is currently hardcoded to `192.168.42.1`, which might have something to do with the issue as well.

Have you tried running `bin/upgrade_containers.sh` at least once (after the upgrade)?
Anyways, I've recently checked the source code, and it seems that the IP address(es) (via the `MY_IP` environment variable) doesn't change on startup. The value of `MY_IP` only changes when `bin/upgrade_containers.sh` was run.

vpetersson, 2023-01-17:
> Anyways, I've recently checked the source code, and it seems that the IP address(es) (via the MY_IP environment variable) doesn't change on startup

We should probably change this now that we pull the value from Redis.

cedgeit, 2023-01-17:
Hello @nicomiguelino ..Thank you.  Running bin/upgrade_containers.sh worked perfectly.  Much appreciated

closed 2023-01-17 by cedgeit