Screenly/Anthias #1717: Not booting from »Other specific-purpose OS« with Raspberry Pi 3+

sheydin, 2023-01-23:
After flashing the SD card with Anthias from »Other specific-purpose OS« it doesn't boot up. The screen remains black, the green light flashes quickly at beginning. An SD card flashed with Raspberry PI OS boots up properly. Any idea?

vpetersson, 2023-01-23:
Hmm that's odd. Are you able to reproduce this @nicomiguelino?

sheydin, 2023-01-23:
Just installed full Raspberry PI OS with GUI and then executed:
> $ bash <(curl -sL https://install-anthias.srly.io)

vpetersson, 2023-01-23:
> Just installed full Raspberry PI OS with GUI and then executed:

You need to use the Raspberry Pi OS Lite, not the desktop version.

nicomiguelino, 2023-01-23:
@vpetersson, I've not replicated the issue. It currently takes a while for the Anthias splash screen to finish ('cause during that time, images are being downloaded). After 11 minutes, the splash screen (with the IP address) showed up. After a minute, the black Screenly loading page (with Screenly logo at the center) showed up.

@sheydin, I was not able to access the web interface as well. I'll investigate the issue further.

nicomiguelino, 2023-01-23:
@sheydin & @vpetersson, the issue where I'm not able to access the web UI via browser is a false positive on my end. My Raspberry Pi and my laptop (which I use for development) are connected to different networks, that's why I can't access the web interface.

@sheydin, is your display still stuck on a black screen?

sheydin, 2023-01-25:
@nicomiguelino I screen is stuck black after installing Pi OS Lite and installing Anthias over it. Next I'll try to flash Anthias directly into the SD via »Other specific-purpose OS« and give it more time to boot.

ealmonte32, 2023-01-27:
> @nicomiguelino I screen is stuck black after installing Pi OS Lite and installing Anthias over it. Next I'll try to flash Anthias directly into the SD via »Other specific-purpose OS« and give it more time to boot.

Dont need to do all that, if you installed the OS Lite and anthias on it via bash but get a black screen, try to just recreate the containers again with this script: `./screenly/bin/upgrade_containers.sh` , run that and see if after it finishes you get the anthias screen with IP.. it has been an issue with many other users and myself included, cant really pinpoint why sometimes the containers get built and sometimes they dont, but it definitely has to do with docker.

sheydin, 2023-01-27:
The initial problem is solved basically by being more patient. Flashed the SD card again via »Other specific-purpose OS« and it boots now. During booting the screen remains black. A loading indicator would be helpful to recognize it's booting. 

After booting I see the Screenly logo only. The Raspberry Pi is not connected via Network. How can I give it Wifi credentials so that it connects to Wifi and I am able to access to web interface?

Much thanks for your help.

ealmonte32, 2023-01-30:
> After booting I see the Screenly logo only. The Raspberry Pi is not connected via Network. How can I give it Wifi credentials so that it connects to Wifi and I am able to access to web interface?

In the console, type in `sudo raspi-config` and in there you have all the options for the Pi, including connecting to wireless in the network options.

sheydin, 2023-01-30:
> In the console, type in sudo `raspi-config` and in there you have all the options for the Pi, including connecting to wireless in the network options.

Sorry, this might be a stupid question, bit I do not find any hint how to open console when Anthias is up and running.

ealmonte32, 2023-01-30:
> > In the console, type in sudo `raspi-config` and in there you have all the options for the Pi, including connecting to wireless in the network options.
> 
> Sorry, this might be a stupid question, bit I do not find any hint how to open console when Anthias is up and running.

on a keyboard connected directly on the Pi, press `ctrl+alt+f1`

agawley, 2023-07-11:
This is my issue. https://github.com/Screenly/Anthias/issues/1734

No idea how to set the wifi though? How do I get the image to boot to terminal?

ealmonte32, 2023-07-11:
> This is my issue. #1734
> 
> No idea how to set the wifi though? How do I get the image to boot to terminal?

Are you saying that pressing `control+alt+f1` on your keyboard does not bring up the console screen?

agawley, 2023-07-11:
Correct. The pi is unresponsive to keyboard input.

ealmonte32, 2023-07-11:
WiFi can be set in multiple ways.. one way that does not require the console is explained here:
https://forums.screenly.io/t/wifi-not-working/817/2

Please read it thoroughly, that way you will have connectivity without troubleshooting why your `control+alt+f1` doesnt seem to work, which is extremely rare and it is not specifically an Anthias thing, but an OS/window manager thing. Did you try ctrl+alt+f2/f3/f7/etc?

Did you add the blank `ssh.txt` or `ssh` file to the `/boot/` directory of the SD card to enable SSH? which then allows you to ssh into the Pi to check if one the containers is not built and that is why you are getting the 500 error, since that would be one of the main reasons for that message..

I also recommend the alternative way to get up and running with Anthias when the Pi-Imager doesn't work as expected which you can find by searching in the screenly.io forums, you also didn't mention if you're using the Raspbian OS Lite supported version? or are you using the default balenaOS image? etc, more specific details would help..

ealmonte32, 2023-07-11:
> Looks like the OS from RPi Imager is BalenaOS

There we go, this is why, balenaOS does not have ctrl+alt+f1, that's a BusyBox type operating system (as far as I know), and doesnt have this shortcut to console.
I have heard nico tell other users that the reason Anthias takes a long time to load is because it is downloading stuff or building the containers in the background and some users have had to wait upwards of an hour maybe, depending their internet speed, to get it fully working.
This is why I only test and work on anthias with Raspbian OS lite: https://forums.screenly.io/t/i-simply-cant-get-anthias-to-work/1005/4

agawley, 2023-07-13:
Got it working by installing PiOS Lite and following your instructions. Thanks.

ealmonte32, 2023-07-13:
@agawley , good to hear.

@nicomiguelino , since this is only related to and happens with balena, shouldn't the Anthias page show some sort of progress bar or heading that lets the user know that they have loaded `Anthias (balenaOS version)` and that it will take a while to load, so that we don't have these back and forth with users not knowing why their screen is black and are sitting there waiting for something to happen and then go and open an issue here or post in forums.. I think all that would be avoided with such heading and/or progress bar.

agawley, 2023-07-13:
FWIW I left the BalenaOS image running for several hours with no result.

nicomiguelino, 2024-05-30:
@agawley, @sheydin, @ealmonte32 Are you still having problems with this issue?

agawley, 2024-05-30:
I followed these instructions to avoid the issue https://forums.screenly.io/t/i-simply-cant-get-anthias-to-work/1005/4