Screenly/Anthias #1719: Raspberry Pi Zero W Rev 1.1

outrank-james, 2023-01-26:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Docker fails to install / run when installing screenly via a fresh install of Raspbian lite image. 

`TASK [system : Install Docker] ***************************************************************************************************************
fatal: [localhost]: FAILED! => {"cache_update_time": 1674739188, "cache_updated": true, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"       install 'docker-ce:armhf' 'docker-ce-cli:armhf' 'docker-compose-plugin:armhf' -o APT::Install-Recommends=no' failed: E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "rc": 100, "stderr": "E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stderr_lines": ["E: Sub-process /usr/bin/dpkg returned an error code (1)"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following package was automatically installed and is no longer required:\n  dc\nUse 'sudo apt autoremove' to remove it.\nThe following additional packages will be installed:\n  containerd.io\nSuggested packages:\n  cgroupfs-mount | cgroup-lite\nRecommended packages:\n  apparmor docker-ce-rootless-extras libltdl7\nThe following NEW packages will be installed:\n  containerd.io docker-ce docker-ce-cli docker-compose-plugin\n0 upgraded, 4 newly installed, 0 to remove and 54 not upgraded.\nNeed to get 80.1 MB of archives.\nAfter this operation, 335 MB of additional disk space will be used.\nGet:1 https://download.docker.com/linux/debian bullseye/stable armhf containerd.io armhf 1.6.15-1 [20.4 MB]\nGet:2 https://download.docker.com/linux/debian bullseye/stable armhf docker-ce-cli armhf 5:20.10.23~3-0~debian-bullseye [38.4 MB]\nGet:3 https://download.docker.com/linux/debian bullseye/stable armhf docker-ce armhf 5:20.10.23~3-0~debian-bullseye [13.1 MB]\nGet:4 https://download.docker.com/linux/debian bullseye/stable armhf docker-compose-plugin armhf 2.15.1-1~debian.11~bullseye [8234 kB]\nFetched 80.1 MB in 1min 18s (1028 kB/s)\nSelecting previously unselected package containerd.io.\r\n(Reading database ... \r(Reading database ... 5%\r(Reading database ... 10%\r(Reading database ... 15%\r(Reading database ... 20%\r(Reading database ... 25%\r(Reading database ... 30%\r(Reading database ... 35%\r(Reading database ... 40%\r(Reading database ... 45%\r(Reading database ... 50%\r(Reading database ... 55%\r(Reading database ... 60%\r(Reading database ... 65%\r(Reading database ... 70%\r(Reading database ... 75%\r(Reading database ... 80%\r(Reading database ... 85%\r(Reading database ... 90%\r(Reading database ... 95%\r(Reading database ... 100%\r(Reading database ... 45476 files and directories currently installed.)\r\nPreparing to unpack .../containerd.io_1.6.15-1_armhf.deb ...\r\nUnpacking containerd.io (1.6.15-1) ...\r\nSelecting previously unselected package docker-ce-cli.\r\nPreparing to unpack .../docker-ce-cli_5%3a20.10.23~3-0~debian-bullseye_armhf.deb ...\r\nUnpacking docker-ce-cli (5:20.10.23~3-0~debian-bullseye) ...\r\nSelecting previously unselected package docker-ce.\r\nPreparing to unpack .../docker-ce_5%3a20.10.23~3-0~debian-bullseye_armhf.deb ...\r\nUnpacking docker-ce (5:20.10.23~3-0~debian-bullseye) ...\r\nSelecting previously unselected package docker-compose-plugin.\r\nPreparing to unpack .../docker-compose-plugin_2.15.1-1~debian.11~bullseye_armhf.deb ...\r\nUnpacking docker-compose-plugin (2.15.1-1~debian.11~bullseye) ...\r\nSetting up containerd.io (1.6.15-1) ...\r\nCreated symlink /etc/systemd/system/multi-user.target.wants/containerd.service -> /lib/systemd/system/containerd.service.\r\nJob for containerd.service failed because a fatal signal was delivered causing the control process to dump core.\r\nSee \"systemctl status containerd.service\" and \"journalctl -xe\" for details.\r\nSetting up docker-compose-plugin (2.15.1-1~debian.11~bullseye) ...\r\nSetting up docker-ce-cli (5:20.10.23~3-0~debian-bullseye) ...\r\nSetting up docker-ce (5:20.10.23~3-0~debian-bullseye) ...\r\nCreated symlink /etc/systemd/system/multi-user.target.wants/docker.service -> /lib/systemd/system/docker.service.\r\nCreated symlink /etc/systemd/system/sockets.target.wants/docker.socket -> /lib/systemd/system/docker.socket.\r\nA dependency job for docker.service failed. See 'journalctl -xe' for details.\r\ninvoke-rc.d: initscript docker, action \"start\" failed.\r\n* docker.service - Docker Application Container Engine\r\n     Loaded: loaded (\u001b]8;;file://raspberrypi/lib/systemd/system/docker.service\u0007/lib/systemd/system/docker.service\u001b]8;;\u0007; enabled; vendor preset: enabled)\r\n     Active: inactive (dead)\r\nTriggeredBy: \u001b[0;1;32m*\u001b[0m docker.socket\r\n       Docs: \u001b]8;;https://docs.docker.com\u0007https://docs.docker.com\u001b]8;;\u0007\r\n\r\nJan 26 13:26:31 raspberrypi systemd[1]: \u001b[0;1;38;5;185m\u001b[0;1;39m\u001b[0;1;38;5;185mDependency failed for Docker Application Container Engine.\u001b[0m\r\nJan 26 13:26:31 raspberrypi systemd[1]: \u001b[0;1;39m\u001b[0;1;31m\u001b[0;1;39mdocker.service: Job docker.service/start failed with result 'dependency'.\u001b[0m\r\ndpkg: error processing package docker-ce (--configure):\r\n installed docker-ce package post-installation script subprocess returned error exit status 1\r\nProcessing triggers for man-db (2.9.4-2) ...\r\nErrors were encountered while processing:\r\n docker-ce\r\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following package was automatically installed and is no longer required:", "  dc", "Use 'sudo apt autoremove' to remove it.", "The following additional packages will be installed:", "  containerd.io", "Suggested packages:", "  cgroupfs-mount | cgroup-lite", "Recommended packages:", "  apparmor docker-ce-rootless-extras libltdl7", "The following NEW packages will be installed:", "  containerd.io docker-ce docker-ce-cli docker-compose-plugin", "0 upgraded, 4 newly installed, 0 to remove and 54 not upgraded.", "Need to get 80.1 MB of archives.", "After this operation, 335 MB of additional disk space will be used.", "Get:1 https://download.docker.com/linux/debian bullseye/stable armhf containerd.io armhf 1.6.15-1 [20.4 MB]", "Get:2 https://download.docker.com/linux/debian bullseye/stable armhf docker-ce-cli armhf 5:20.10.23~3-0~debian-bullseye [38.4 MB]", "Get:3 https://download.docker.com/linux/debian bullseye/stable armhf docker-ce armhf 5:20.10.23~3-0~debian-bullseye [13.1 MB]", "Get:4 https://download.docker.com/linux/debian bullseye/stable armhf docker-compose-plugin armhf 2.15.1-1~debian.11~bullseye [8234 kB]", "Fetched 80.1 MB in 1min 18s (1028 kB/s)", "Selecting previously unselected package containerd.io.", "(Reading database ... ", "(Reading database ... 5%", "(Reading database ... 10%", "(Reading database ... 15%", "(Reading database ... 20%", "(Reading database ... 25%", "(Reading database ... 30%", "(Reading database ... 35%", "(Reading database ... 40%", "(Reading database ... 45%", "(Reading database ... 50%", "(Reading database ... 55%", "(Reading database ... 60%", "(Reading database ... 65%", "(Reading database ... 70%", "(Reading database ... 75%", "(Reading database ... 80%", "(Reading database ... 85%", "(Reading database ... 90%", "(Reading database ... 95%", "(Reading database ... 100%", "(Reading database ... 45476 files and directories currently installed.)", "Preparing to unpack .../containerd.io_1.6.15-1_armhf.deb ...", "Unpacking containerd.io (1.6.15-1) ...", "Selecting previously unselected package docker-ce-cli.", "Preparing to unpack .../docker-ce-cli_5%3a20.10.23~3-0~debian-bullseye_armhf.deb ...", "Unpacking docker-ce-cli (5:20.10.23~3-0~debian-bullseye) ...", "Selecting previously unselected package docker-ce.", "Preparing to unpack .../docker-ce_5%3a20.10.23~3-0~debian-bullseye_armhf.deb ...", "Unpacking docker-ce (5:20.10.23~3-0~debian-bullseye) ...", "Selecting previously unselected package docker-compose-plugin.", "Preparing to unpack .../docker-compose-plugin_2.15.1-1~debian.11~bullseye_armhf.deb ...", "Unpacking docker-compose-plugin (2.15.1-1~debian.11~bullseye) ...", "Setting up containerd.io (1.6.15-1) ...", "Created symlink /etc/systemd/system/multi-user.target.wants/containerd.service -> /lib/systemd/system/containerd.s [... cut ...]
PLAY RECAP ***********************************************************************************************************************************
localhost                  : ok=27   changed=19   unreachable=0    failed=1    skipped=2    rescued=0    ignored=0   

`

#### Reproduction Steps

Steps to reproduce this issue, eg:

Install Anthias (Screenly) via bash command  bash <(curl -sL https://install-ose.srly.io)

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Zero W Rev 1.1 - 32 Big
* Raspberry Pi Network Setup: standard
* Screenly OSE Version: latest install

vpetersson, 2023-01-26:
Are you able to install Docker using [these instructions](https://docs.docker.com/engine/install/debian/)?

outrank-james, 2023-01-26:
> 

Giving it a try now - will report back ASAP

outrank-james, 2023-01-26:
Looks to be an issue with Docker - doesn't seem to be able to run on the Raspberry Pi Zero W

`Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit containerd.service has entered the 'failed' state with result 'core-dump'.
Jan 26 14:04:22 raspberrypi systemd[1]: Failed to start containerd container runtime.
░░ Subject: A start job for unit containerd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit containerd.service has finished with a failure.
░░ 
░░ The job identifier is 4666 and the job result is failed.
Jan 26 14:04:27 raspberrypi systemd[1]: containerd.service: Scheduled restart job, restart counter is at 61.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ Automatic restarting of the unit containerd.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Jan 26 14:04:27 raspberrypi systemd[1]: Stopped containerd container runtime.
░░ Subject: A stop job for unit containerd.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A stop job for unit containerd.service has finished.
░░ 
░░ The job identifier is 4734 and the job result is done.
Jan 26 14:04:27 raspberrypi systemd[1]: Starting containerd container runtime...
░░ Subject: A start job for unit containerd.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit containerd.service has begun execution.
░░ 
░░ The job identifier is 4734.
Jan 26 14:04:28 raspberrypi systemd[1]: containerd.service: Main process exited, code=dumped, status=11/SEGV
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit containerd.service has exited.
░░ 
░░ The process' exit code is 'dumped' and its exit status is 11.
Jan 26 14:04:28 raspberrypi systemd[1]: containerd.service: Failed with result 'core-dump'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit containerd.service has entered the 'failed' state with result 'core-dump'.
Jan 26 14:04:28 raspberrypi systemd[1]: Failed to start containerd container runtime.
░░ Subject: A start job for unit containerd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit containerd.service has finished with a failure.
░░ 
░░ The job identifier is 4734 and the job result is failed.`

vpetersson, 2023-01-26:
I don't have a Pi Zero here, but what do you get from running `arch`? Docker is trying to install the `armhf` version, so if you get anything other than that, that would explain it.

outrank-james, 2023-01-26:
Arch = armv6l

vpetersson, 2023-01-26:
Interesting. Ok, can't really look into this now. I wouldn't really recommend using a Pi Zero for Anthias regardless due to it's low resources. A Pi 3/4 would be much more suitable for the use case. It's certainly *possible* to run it on a Pi Zero (and people have done this), but the performance will be unsuitable for anything other than very basic image playback.

outrank-james, 2023-01-26:
This was just a test to get a single image on a screen - we've used this perfectly well on a Pi3/4 on other little projects - thought I'd give it a test on a Zero. 

Definitely looks to be a Docker issue on the armv6l - would probably work just fine on the newer Pi Zero's

NassimBentarka, 2025-12-31:
Just in case anyone needs to run Screenly on Raspberry Pi Zero W, see https://github.com/Screenly/Anthias/issues/1666#issuecomment-3702350165