Screenly/Anthias #1721: Set zoom level for Assets

TiWe1998, 2023-01-27:
Hello, I'm looking for a way to set the zoom level. Unfortunately I found an old manual. I cannot find the folder or file in the current image.

And another small question: Is it correct that the various websites are completely updated again and again or do they remain as they were when they were first started?

BR

TiWe1998, 2023-01-27:
Yes!

ealmonte32, 2023-01-27:
Thanks.
You should close this issue on here after you post on forums.

TiWe1998, 2023-02-01:
I posted it in the forum. It has yet to be released. Thanks! Can be closed here.

ealmonte32, 2023-02-03:
> I posted it in the forum. It has yet to be released. Thanks! Can be closed here.

can you just close this yourself? otherwise i have to bother someone else to close.

closed 2023-02-03 by vpetersson