Screenly/Anthias #1722: Pi lockups after a day

cedgeit, 2023-01-31:
#### Overview of the Issue

Loaded the latest screenly on a (3) Pi 3 devices.  We initially had (2) assets scheduled.  One is a web page that displays for 60 seconds and then it transitions to a JPG image for 10 seconds and cycles from there.  What happens is that after a day or so the pi will lock up.  We changed the assets to only display the web page and it still locked up.  So as a test we rebooted the pi and monitored the memory throughout the day and it appears that throughout the course of the day, the memory drops to the point where the pi locks up.  So it appears there might be a memory leak with the web viewer function.  

The web page above displays a google calendar with scrolling content along the side.

#### Reproduction Steps

Steps to reproduce this issue, eg:

Press Add Asset
Enter the Asset URL ...
Press "On"
Wait for the asset to show up on the screen
Wait for a day to see if the pi locks up.

#### Environment
Raspberry Pi Model	Raspberry Pi 3 Model B Rev 1.2
Raspberry Pi Network Setup: Wifi and Ethernet

Screenly OSE Version: master@42b975e

vpetersson, 2023-01-31:
Interesting. In theory, this should be impossible since we set cgroup limits in Docker. 

Does this happen if you *only* rotate through a few images?

cedgeit, 2023-02-01:
Thank you.  So as a test we disabled the web page asset and setup (2) images that display every 10 seconds.  So far the pi's seem to be fine.  No lockups, etc.  Memory hangs right around 38 free.   It only appears to be happening when we use a web page display.  Thanks again

vpetersson, 2023-02-01:
Without having all the information, I would think that the issue is that the website is very heavy. The quickest fix for this is probably to swap out your Pi 3 for a Pi 4 with say 4GB of RAM.

cedgeit, 2023-02-01:
Thanks...We will dig deeper into this issue.  We were running screenly OS from 2019 with the exact configuration on all (3) Pi's for a couple years with no issues.  Perhaps the new version requires more resources.  I thought I read somewhere with the commercial version of screenly that there was an issue with the web viewer component that required a cron job to restart it periodically.  I will see if if we can find that information.  Again, Thanks!

ealmonte32, 2023-02-02:
@cedgeit 

Can you run the following commands on your Pi3 and post output:
`grep -i memory /var/log/messages`
&
`uname -a`

___

>I thought I read somewhere with the commercial version of screenly that there was an issue with the web viewer component that required a cron job to restart it periodically. 

As far as I remember, this is probably incorrect and related to OSE only, where it was suggested to users that experienced the scenario you mention to create a cronjob of restarting the screenly viewer once a day or once a week so that it would not run into the OOM error and freeze the Pi or leave the screen black after OOM killed the process.

ealmonte32, 2023-02-02:
@cedgeit 

Also..
Is your google calendar private or is it public? if the latter, can you share the URL so that I can test it and see if my Pi3 freezes like yours because I did experience this before with some websites but I forgot if it was fixed with kernel/firmware update since in 2020 cgroups were broken in the 64-bit kernel from what I've read, thus in my testing scenarios and the low risk of updating firmware on the Pi because it is specifically for Anthias, I run `rpi-update` to always make sure I am running latest version of the kernel firmware and I am not experiencing this issue while having all these assets (it has two srly.io websites so it's not heavy on the web use), although I do experience random memory numbers written on the side of the screen from time to time when images are displayed.. but not Pi freezing..

[image]

vpetersson, 2023-02-02:
> I thought I read somewhere with the commercial version of screenly that there was an issue with the web viewer component that required a cron job to restart it periodically.

No the commercial version is a lot more tested and stable. 

It is however true that Anthias did switch browser relatively recently to a more modern one. This likely increased the memory consumption.

ealmonte32, 2023-02-06:
@cedgeit 
No need to post the site you are adding as I was able to reproduce by adding the BBC news website and another 2 for 1 min each to loop over the weekend and the Pi is freezing for me as well.
This tells me there is a need to adjust the cgroup memory limit to invoke OOM killer sooner.

cedgeit, 2023-02-06:
Hello...Thank you very much for taking the time to test...We ran some tests over the weekend on different websites with little content and it locked up as well.  We will adjust the memory settings and see if this helps..I believe we saw that information posted.  Much appreciated.

ealmonte32, 2023-02-07:
@vpetersson 
@nicomiguelino 

So after some testing, even though we set limit on viewer container to 70% of total memory, this doesn't really seem to be having the desired effects because if we add up all containers mem usage during normal operations, the limit set at 70% just for the viewer will not be reached while the total memory of the Pi will be reached and cgroup will not invoke OOM killer because there is no `memory.limit_in_bytes` set up for the Pi.

As you can see in the screenshot example, the celery container is taking up `114 MB` of ram, plus the `52 MB` from server container, and when we add the rest of them, we already have an estimate of `~190 MB` of memory in use without even including the viewer.

This test Pi3B has been set to only use 128MB of GPU memory, thus it gives it a total mem capacity of 860MB, the OS without any containers running is about 80-100MB, this all calculates to the system having a viewer `mem_limit` that will never be reached because the Pi itself will reach the memory limit before the viewer does.

I think aside from recompiling QtWebEngine with some non-cache/non-persistent memory configuration, the two things we can do for now is look into: [celery memory leak or optimization](https://docs.celeryq.dev/en/stable/userguide/optimizing.html) if there is anything that can be done there, and reducing the viewer container `mem_limit` to 55% of total memory, which should be plenty for being able to use the QtWebEngine while preventing resource-heavy websites from rendering the Pi inoperable which requires physical intervention to reboot.

I will be testing the proven list of website assets that freeze this Pi3B with this 55% modification to see how well it works.

Thoughts?

[image]

vpetersson, 2023-02-07:
Very good research, @ealmonte32. It's obvious that the logic we have used here (70% of RAM) for each container is flawed. We need to be much more restrictive and calculate what the total *available* RAM is (i.e. RAM - GPU allocated RAM) and then divvy up *that* across all containers. We might oversubscribe a tad, but not to the crazy level we are right now.

All this being said, I would like to hold this a bit until we've finalized the [Django](https://github.com/Screenly/Anthias/issues/558)/[Python 3](https://github.com/Screenly/Anthias/issues/1091) migration, since we do not know what this will do to the memory consumption yet. Hopefully by merely upgrading our building blocks (such as Celery) will help reduce the memory usage.

ealmonte32, 2023-02-08:
@cedgeit 

After testing with different amounts I am getting the opposite of what I thought, so strangely making the container mem_limit higher for the viewer allowed it to run longer and hasn't frozen the Pi, while lowering it did not freeze the Pi but IPv4/dhcpcd stopped working and viewer was showing black screen.

So... I would like you to try changing this number to what I have now and see if it freezes after two days as you stated before, then report back:

Edit the file: `nano ./screenly/bin/upgrade_containers.sh` or use vi, whatever you prefer.

Change the `.70` to `.80` and save it.

Then, stop all containers with this command: `sudo docker stop $(sudo docker ps -a -q)`

Then after they are all stopped, run the script: `./screenly/bin/upgrade_containers.sh`

[image]

cedgeit, 2023-02-09:
I apologize for the long delay...Thank you.  We made the changes you mentioned to (2) Pi's and enabled the web site site again.  We will monitor this and post back.  If it hangs again is there specific logs you would like us to collect for you?

ealmonte32, 2023-02-09:
> I apologize for the long delay...Thank you. We made the changes you mentioned to (2) Pi's and enabled the web site site again. We will monitor this and post back. If it hangs again is there specific logs you would like us to collect for you?

Well, we know what the issue is so the viewer or host logs won't really show anything we don't already know.

I am having good results with `.80` so I hope it also works for you, and you can probably even increase to `.85`... only thing I recently noticed which might not even be related to this is that when mixing heavy sites with also showing assets of images, it now after a while shows a black screen instead of the images, no idea why the viewer is doing that, but the Pi is not freezing any longer with heavy sites and the websites themselves are showing.. this images thing is probably for a different topic because it only started happening when I upgraded the Anthias version..

cedgeit, 2023-02-10:
Hello...We made the change to (2) PI's.  One of them locked up and the other one is really slow right now.  We will try to bump up the value to .85 and see if that helps.

cedgeit, 2023-02-13:
Hello..Unfortunately, the (2) Pi's lockup after 48 hours changed it to .85 as well.  The page being displayed is a single google calendar in monthly view under a google account.  Is there anything else I should try? Thanks for your help in trying to resolve this issue.

ealmonte32, 2023-02-13:
@cedgeit 
I added a bunch of websites that load many things and use lots of resources that before I changed to .80 it would freeze the Pi after maybe a day or so, and now it doesn't, so I know it helps.

Did you run the upgrade containers script after and then with command `docker stats` made sure that your .85 was in fact taking effect in the container?
I would try `.99` just to see because how can one simple google calendar page be so heavy that crashes the Pi? sounds like a memory leak..

cedgeit, 2023-02-15:
@ealmonte32 Thank you very much...We did bump them up to .90 and .95 and it still locks up.  docker stats does show the changes are working as expected.

I hate to ask but is it possible to back rev to a previous screenly OSE version?  As I initially indicated we were using Screenly for a couple of years and all worked perfectly well.  It could be that the Pi3 is underpowered for the current OSE version and we need to upgrade and we can do that but we would like to get the existing Pi's up if possible.  Again thank you again for trying to assist.

ealmonte32, 2023-02-15:
> I hate to ask but is it possible to back rev to a previous screenly OSE version? As I initially indicated we were using Screenly for a couple of years and all worked perfectly well. It could be that [...]

Well, in terms of you being able to get an older version of Anthias like Screenly OSE before it used QtWebview, sure, you can go into the Releases and try to get those older images working but know you will get no support at all since those are deprecated and also since the older ones use the UZBL browser it would even be worse since you won't be able to load some of the most basic websites.

I am trying to understand how a single Google calendar page being set as the asset can lock up the Pi, this doesn't make sense unless there is some memory leak specifically with this combination of website and Qt browser version.
I will test by publishing my own personal google calendar just to test and see, unless you want to publicly post the exact one you're using.
Have you ran the `rpi-update` on your Pi3 btw?
How much memory did you assign the Pi for graphics?

cedgeit, 2023-02-15:
Hi Thanks!  I am going to continue down this path to fix this issue on the pi3 and not back-rev.  You have been super helpful and it might help others.  Not gonna give up.  

So I did NOT run the rpi-update and am doing that right now.  I must confess it is coming from 5.15 to 6.1 so I should have done this
[image]
 from the beginning.  Many apologies.  So please see the attached docker stats and see if they look correct.  Also 196 is allocated to GPU memory. 

I am going to see if I can get the site published publically to check it out.  There are google creds in there for the calendar so I will see what I can do.  
Again, thanks!
[image]

cedgeit, 2023-02-15:
Hi Thanks! I am going to continue down this path to fix this issue on the pi3 and not back-rev. You have been super helpful and it might help others. Not gonna give up.

So I did NOT run the rpi-update and am doing that right now. I must confess it is coming from 5.15 to 6.1 so I should have done this
Uploading pi.png…
from the beginning. Many apologies. So please see the attached docker stats and see if they look correct. Also 196 is allocated to GPU memory.

I am going to see if I can get the site published publically to check it out. There are google creds in there for the calendar so I will see what I can do.
Again, thanks!
[image]

ealmonte32, 2023-02-15:
@cedgeit 
The Pi3 I was testing actually finally, after more than a week of having 8 assets with multiple resource heavy sites added, "froze", but the Pi was actually still locally able to function but very unstable and was timing out.
I use 128MB of GPU memory so maybe that is why mine was lasting a bit longer, not sure, can you tell me if your calendar looks like this:
https://calendar.google.com/calendar/u/0/embed?src=en.usa%23holiday@group.v.calendar.google.com&ctz=America/New_York

Because that is a public google calendar that anyone can use, and if you're using google calendar and yours just has private events listed, then it should be the same page load thus not necessary to use your calendar for testing.

cedgeit, 2023-02-15:
Thanks...Yes...That is how the calendar looks with a monthly view.  But there are multiple calendars feeding the view that are all color-coded, etc.  9 calendars to be exact.  Perhaps the google related function or managing all of those events from multiple calendars is causing an issue.

I can change the graphics memory setting from 196MB to 128MB if that will help.

When our pi's freeze (usually after 24-48 hours) they lock up completely.  Not even pingable, no ssh (we did enable that), etc.

I am hoping to get a better baseline in a day or 2 with the new firmware.

cedgeit, 2023-02-17:
@ealmonte32 Thanks for the info.  Unfortunately, the firmware updates etc did not help.  When the google calendar is running as an asset we barely get 10 hours now out of the PI 3's before they lock up.  Again not pingable on the local network or accessible in any way.  So you are thinking that the pi is still alive even though the network services are not accessible?  I could develop a cron script to restart docker.service and docker.socket if the pi can not ping out to the web.  And if the pi is still alive but networking is not working then perhaps this might provide a workaround.  

I am going to take a detailed look at the actual calendar this weekend and see if there is something out of the ordinary there.  Gonna review the pi logs as well.

Again, Thank you