Screenly/Anthias #1723: WiFi docker service breaks normal WiFi function

iDazai, 2023-02-02:
#### Overview of the Issue

When restarting the wlan0 interface, the real wlan0 interface won't get a connection. Most likely because of the WiFi docker service which acts as an access point.

#### Reproduction Steps

Enable WiFi before installing Anthias, restart the wlan0 interface, comes up with the IP from DHCP.
Install Anthias and restart the wlan0 interface, interface comes up with another MAC address and a 192.168.42.1 IP.
After commenting out the lines of the _srly-ose-wifi-connect_ service inside the _docker-compose.yml_ and restarting the Pi, it works as it should.

#### Environment

* Raspberry Pi Hardware Version: 3b+
* Raspberry Pi Network Setup: built-in WiFi
* Screenly OSE Version: master@42b975e pulled on 02.02.2023

Is it possible to add this as a question at the beginning of the installation script if the user wants that function or not?
Or am I the only one with this problem, when the Pi is connected with WiFi?

vpetersson, 2023-02-02:
In theory, this use case should be handled. If there is connectivity, the Active Portal shouldn't be launched. 

@nicomiguelino could you please check if you can reproduce this?

nicomiguelino, 2023-02-02:
On it, @vpetersson. Yes, it's supposed to work, but I'll check the scripts (especially `start_wifi_connect.sh`), there might be something with the flags used in the while loop.

nicomiguelino, 2023-02-06:
@iDazai, @vpetersson 
I'm still not able to reproduce the issue.
I already checked the flags/conditionals inside the while loop.

I want to ask for more details @iDazai.

- Is your Pi running Raspberry Pi OS Lite?
- Is your an Ethernet cable connected to your Pi when it happened?
  - If it's connected via an Ethernet cable, it might be that you're connected to the Ethernet, but doesn't have Internet connectivity.
- How often does this issue occur? (Is it intermittent, or is it consistent?)

I'm trying to replicate the issue by starting the WiFi Connect service (at startup) via `systemd` (by writing a new `systemd` service).

iDazai, 2023-02-08:
@nicomiguelino 

- Yes, I'm running the Lite version 09/2022 from the official website upgraded to the latest version.
- I'm not sure if I still had it connected, will try without the cable next time
- It was consistent, if I remember correctly on boot I would get the reserved IP address for the MAC-address but after a restart of the wlan0 interface I got the 192.168.42.1 IP address and the MAC always changed after the interface restart

I'll try to reproduce the issue when I have time this week.

iDazai, 2023-02-14:
I couldn't reproduce it, but I also changed some steps.
Here's what I did:

- Changed everything under **raspi-config** > **localisation** to German settings
- **apt-get update && apt-get upgrade**
- **apt-get install fail2ban**
- **rfkill block 1** (bluetooth)
- **nano /etc/systemd/timesyncd** for internal NTP server
- Enable ssh and vnc via **raspi-config**
- **wget https://raw.githubusercontent.com/Screenly/screenly-ose/master/bin/upgrade_containers.sh** and change to internal DNS Server, uncommented the four pull lines
- **wpa_passphrase “Netzwerk-SSID” | tee –a /etc/wpa_supplicant/wpa_supplicant.conf > /dev/null** and **wpa_cli -i wlan0 reconfigure**
- **bash <(curl -sL https://install-anthias.srly.io/)** with the options **y, n, n**
- **./upgrade_containers.sh** after the installation is done

Here's what I didn't do, but did the last time:
- added **deb [arch=armhf] https://download.docker.com/linux/raspbian bullseye stable** to **/etc/apt/sources.list.d/docker.list**  and added the public key to the **keyrings** I did that because I had problems with the install script, it always failed at docker with a big red error output. I think it was similar to this issue [#1691](https://github.com/Screenly/Anthias/issues/1691#issue-1494397345) and I read somewhere that adding it before the Anthias install should help but I had mixed results. Mostly it worked when I started the script for a second time after a reboot. But this time without changing anything it worked on the first try.
- And normally I also did change some settings in server.py and viewer.py according to issue [#1494](https://github.com/Screenly/Anthias/issues/1494#issue-805341251) to make it work without internet and copied them in the docker containers. But I guess it also works without doing this now? This last issue can also be closed.

And this issued can also be closed if you don't have anything more to ask or add.

vpetersson, 2023-02-14:
What really stands out here is `vnc`. You really shouldn't install that. It eats up resources that Anthias desperately needs, but perhaps more importantly, it will most likely not even work given that we are relying a fair bit on framebuffer rendering.

iDazai, 2023-02-14:
> What really stands out here is `vnc`. You really shouldn't install that. It eats up resources that Anthias desperately needs, but perhaps more importantly, it will most likely not even work given that we are relying a fair bit on framebuffer rendering.

Yes, it doesn't work anyways, it was just in my initial installation routine for other raspberries where I don't use Anthias.

extremeskillz, 2023-03-01:
Same issue here. the install script is bricking the networking on the anthias installs. In raspi-config you can set the networking back to dhcpd then you have the add in the wireless lan settings again but this goes away again after a reboot. I have a screenly that is bricked at the moment as I can't recover. This was a upgrade to the latest screenly version using the script install.