Screenly/Anthias #1725: Not booting from latest image, getting 4 flashes on green light.

gljones2001, 2023-02-03:
### Overview of the Issue

When I install the latest complete image on a Pi 4, the pi gives me a 4 flash error code on the green LED when it tries to boot. Seems to be the `loader.bin` failed to boot. I have tried several SD cards (all high-quality class 10 or above), and even two different Pi 4's. Same error. I have tried loading with either the image itself from this GitHub, or the "Other specific OS" in the Raspberry Pi Imager. Same error.  Raspbian Lite loads just fine, no issue.

### Reproduction Steps

1. Get an SD card and a Pi 4
2. Erase SD card with SD Card Formatter
3. Flash the latest image (either from GitHub, or with Raspberry Pi Flashing Tool)
4. Insert SD into Pi 4
5. Plug Pi 4 into monitor/tv and power
6. watch green light. It will flash sporadically for ~30sec then start doing 4 blinks then off, 4 blinks then off, 4 blinks then off, etc. pattern with the light. Splash screen will show on TV, but nothing else.

### Environment

- Pi 4 4gb model (tried several of the same model)
- Anthias 18.5
- SanDisk ImageMate Pro UHS-I 64GB (V30, U3, C10, A2)
- PNY 64GB Class 10 SD
- Walmart Onn branded Micro/mini HDMI to HDMI adapter
- Sufficient Power supply (anker nano 20W) w/ belkin USB-C cable
- Wifi 5 (ubiquiti unifi)
- Ethernet (1gbps, same unifi network)
- Logitech K400+ Keyboard/Touchpad
- Walmart Onn Branded 4k 42" Roku TV (had to set HDMI to safe mode to get this working)

Hope that's everything you need 😁

vpetersson, 2023-02-06:
This is weird. Lots of people are using these images. You've already done all the basic things that we'd normally ask you to check for, so I'm a bit lost here tbh.

Could you try the [v0.18.6](https://github.com/Screenly/Anthias/releases/tag/v0.18.6) pre-release to see if that changes anything?

gljones2001, 2023-02-08:
I installed the v0.18.6 image, but now it will not show the wifi config page. All I get is text in the top left saying `Sorry, this page does not exist!`
What do I do now?

gljones2001, 2023-02-08:
Yeah I just tried the latest release again with Etcher (IT DID FINALLY BOOT) but now it won't let me configure wifi! Same `Sorry, this page does not exist!` error when I go to `http://<ip address>/wifi`. Any ideas?

vpetersson, 2023-02-08:
Hold tight. There's [an issue](https://forums.balena.io/t/error-no-matching-manifest-for-linux-arm64-v8-in-the-manifest-list-entries-on-github-action/366625/5) with deploying the latest updates that we are working on.

gljones2001, 2023-02-08:
Ok, so I was now trying to get this to work with Balena Cloud, and it keeps saying "Bind mounts aren't allowed", even if I use your own fork on Balena Cloud. It won't let me install it properly.

vpetersson, 2023-02-09:
Deploying this yourself to Balena requires a bit more work than just a `balena deploy`. You can see how this is done [here](https://github.com/Screenly/Anthias/blob/master/.github/workflows/docker-build.yaml#L109).

gljones2001, 2023-02-09:
> Deploying this yourself to Balena requires a bit more work than just a `balena deploy`. You can see how this is done [here](https://github.com/Screenly/Anthias/blob/master/.github/workflows/docker-build.yaml#L109).

So how exactly do I run this? I can see it's a GitHub workflow CI build YAML config, but what would I run on a Linux dev machine to achieve the same result?

gljones2001, 2023-02-09:
Ah. Ok. I think I figured it mostly out. It keeps saying `Error: no matching manifest for linux/arm64/v8 in the manifest list entries` when I run `balena push`. Do I need to choose a different release? I set it to the latest release in the docker-compose.yml generated in the balana-deploy folder.

gljones2001, 2023-02-09:
Oh I think I just found it. It seems screenly is not built for 64 bit arm yet. I will try using an armv7 fleet instead.

vpetersson, 2023-02-09:
So you've run into the same bug as we've run into as well actually. There seems to be a bug somewhere in Balena. We've [opened forum post](https://forums.balena.io/t/error-no-matching-manifest-for-linux-arm64-v8-in-the-manifest-list-entries-on-github-action/366625/5) outlining it.

With regards to 64bit -- yes, this is true. But there is no issue running a 32bit container on a 64bit OS. this is what we've done all along in the past.

gljones2001, 2023-02-09:
Huh. Interesting.
Well do you have a workaround? Like an older release that works on balena with a pi 4?

vpetersson, 2023-02-09:
I haven't had any bandwidth myself to try this out, but yes, rolling back to previous Balena CLI versions would be a good experiment to see if that changes anything.

gljones2001, 2023-02-09:
Ok, what Anthias release would you recommend trying?

vpetersson, 2023-02-09:
This isn't about the Anthias release, it's about the [balena CLI](https://github.com/balena-io/balena-cli) version. Our CI pipeline pins to `@master` of balena CLI.

gljones2001, 2023-02-09:
Ahhhhh ok so its an issue with the Balena CLI application on my linux desktop. Very interesting.

gljones2001, 2023-02-09:
So then what previous CLI version worked with Anthias? I can try that out and let you know how it goes.

gljones2001, 2023-02-09:
Tried Balena CLI v15.0.0 and same error.

gljones2001, 2023-02-09:
14.5.15, same error. I wonder if there is an error on their end, not with the CLI on the PC's end. Cause 14.5.15 came out beginning of December last year, well before you started having issues.

gljones2001, 2023-02-15:
I just saw the issue on Balena.io was updated, as well as a few commits here in this repo. Is the issue with Balena CLI fixed? Do I need to test it out for you?

gljones2001, 2023-02-15:
Ah. Just saw that you updated it like 10 minutes ago.

gljones2001, 2023-02-15:
Please let me know when it is building successfully. I will be glad to test it out. Thanks!

gljones2001, 2023-02-17:
@vpetersson Is there a build I can test out for you? I noticed that you've been running the GitHub actions for the past couple days. Maybe one of them will work.

vpetersson, 2023-02-17:
So we've overhauled how `bin/build_containers.sh` works a fair bit to make it easier to use directly both on the Pi as well as in our CI/CD pipeline. I've also kicked off a new [disk image build](https://github.com/Screenly/Anthias/actions/runs/4206202635) with all latest changes. Maybe give those images a try (assuming the build process goes through).

gljones2001, 2023-02-17:
Cool. Thanks! I'll let you know how it goes, once there is a good image to use.

gljones2001, 2023-02-23:
@vpetersson DUDE!!! The last build WORKED. You are AWESOME. Thanks! If you need me to test anything else, let me know!

vpetersson, 2023-02-24:
Just keep kicking the tires. We're still in a massive overhaul phase so just keep reporting in bugs.

closed 2023-02-24 by vpetersson