Screenly/Anthias #1731: Image Assets vanishes withing couple of hours from rotation in RPi3

Kallio, 2023-02-09:
#### Overview of the Issue

Image Assets vanishes withing couple of hours from rotation in RPi3. 
Tested with 4x 253KB jpeg files and couple of web assets in same loop. Rebooting system helps 
 
#### Reproduction Steps

Steps to reproduce this issue, eg:

* Add jpg assets, lets say 4 or more
* Press "On"
* Wait for couple of hours and assets will be blank pages on rotation. ( showing time will eg. 10sec is still there) 
#### Environment

* Raspberry Pi Hardware Version: Pi 3 Model B Plus Rev 1.3
* Raspberry Pi Network Setup:  Ethernet
* Anthias Version:  master@5d930ae

vpetersson, 2023-02-09:
Try swapping out the SD card. My bets would be that the SD card is about to give up.

Kallio, 2023-02-09:
I'll try. If SD card would be the case, why I can write and read from console without errors and get ok checksums?

vpetersson, 2023-02-09:
Well, it could be some sporadic errors. We've seen all sort of strange read/write errors on SD cards over the years.

Kallio, 2023-02-10:
I suspect that there were some kind of powersupply issue after all . Swapped SDcard to other Pi with other powersupply and it has been running 18 hours without issues.

closed 2023-02-10 by Kallio