Screenly/Anthias #1732: Raspberry Pi 4B with RTMP

Pellmart1, 2023-02-09:
#### Overview of the Issue

When adding a RTMP as an asset, it will not play.

#### Reproduction Steps

-access device using browser and IP
-click Add Asset
-enter url  rtmp://x.x.x.x:(port)/location
-add check to "Skip asset check" check box.
-asset gets added to list of assets
Get confirmation pop up : Asset has been successfully uploaded.
-New added assets in Inactive Assets list.  it will show the link but disabled.
-Change to enable
-Will show up in Active Assets list.
-Screen will flash, but nothing shows up but a random flashing white screen every minute or so.

If asset is added but no check is added to the Skip asset check box, then you get error pop up
"Server Error: global name 'mplayer' is not defined" 
It will not add the RTMP feed.
nothing changes on the screen.  simpl stays blank and accessible IP disapears.

#### Environment

Using Ethernet
Used Raspberry Pi Imager where you can find Anthias 

Option | Value
Load Average | 0.17
Free space | 22G
Memory | Total: 1731 \| Used: 412 \| Free: 801 \| Shared: 23 \| Buff: 38 \| Available: 1227
Uptime | 0 days and 0.16 hours
Monitor info | state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
Display Power (CEC) | None
Raspberry Pi Model | Raspberry Pi 4 Model B Rev 1.1
Screenly OSE Version | master@42b975e
MAC Address | Unable to retrieve MAC address.

Thanks for your time and consideration

nicomiguelino, 2023-02-09:
Hello @Pellmart1. This feature (along with RSTP assets) has not been working and tested for a long time.

vpetersson, 2023-02-09:
Yeah this is a known issue. Or rather, it was removed in the Pi 4 work. We plan to reinstate this in the future. But this won't happen until we're done with the Python 3/Django migration, which is the current priority.

vpetersson, 2023-02-09:
Lol. You beat me to it @nicomiguelino.