Screenly/Anthias #1736: How-to : upgrade from screenly to Anthias

Draky50110, 2023-02-15:
Hello.

I have installed Screenly-OSE "Spring 10" release.

How can I update to last Anthias release 18.5 ?
SSH/cli or else ?

I tried the UPGRADE button in admin backoffice but it gives me an error : 

`UnicodeDecodeError('utf8', "\x1b[33m\r0% [En cours]\x1b[0m\r             \rR\xe9ception de:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15,0 kB]\n\x1b[33m\r0% [1 InRelease 15,0 kB/15,0 kB 100%] [Attente des fichiers d'en-t\xeate]\x1b[0m\x1b[33m\r                                                                      \r0% [Attente des fichiers d'en-t\xeate]\x1b[0m\r                                   \rAtteint:2 http://archive.raspberrypi.org/debian stretch InRelease\n", 39, 40, 'invalid continuation byte')
Screenly-OSE update was finished

Thanks`

vpetersson, 2023-02-15:
Yeah we're still working on the user experience for this. The quickest way is likely to re-run the installer over ssh.

Draky50110, 2023-02-15:
Without formatting ?
Will it keep assets and config ?

Or fresh re-install from scratch ?

vpetersson, 2023-02-15:
If you've running your installation for a while, I would suggest starting afresh on a brand new SD card as they tend to wear out after few years.

closed 2023-02-15 by Draky50110