Screenly/Anthias #1747: Issue iusing wifi-connect on Raspbian

vpetersson, 2023-02-22:
Just did a fresh install on Raspbian and ended up with the following error:

```
screenly-anthias-wifi-connect-1  | Checking internet connectivity ...
screenly-anthias-wifi-connect-1  | Your device is not connected to the internet.
screenly-anthias-wifi-connect-1  | Starting up Wifi-Connect.
screenly-anthias-wifi-connect-1  | Connect to the Access Point and configure the SSID and Passphrase for the network to connect to.
screenly-anthias-wifi-connect-1  | [network_manager::device:WARN] Undefined device type: 30
screenly-anthias-wifi-connect-1  | WiFi device: wlan0
screenly-anthias-wifi-connect-1  | [network_manager::dbus_api:ERROR] Get org.freedesktop.NetworkManager.AccessPoint::RsnFlags property failed on /org/freedesktop/NetworkManager/AccessPoint/20: wrong property type
screenly-anthias-wifi-connect-1  | Error: Getting access points failed
screenly-anthias-wifi-connect-1  |   caused by: D-Bus failure: Get org.freedesktop.NetworkManager.AccessPoint::RsnFlags property failed on /org/freedesktop/NetworkManager/AccessPoint/20: wrong property type
screenly-anthias-wifi-connect-1  | Checking internet connectivity ...
screenly-anthias-wifi-connect-1  | Your device is not connected to the internet.
screenly-anthias-wifi-connect-1  | Starting up Wifi-Connect.
screenly-anthias-wifi-connect-1  | Connect to the Access Point and configure the SSID and Passphrase for the network to connect to.
screenly-anthias-wifi-connect-1  | [network_manager::device:WARN] Undefined device type: 30
screenly-anthias-wifi-connect-1  | WiFi device: wlan0
screenly-anthias-wifi-connect-1  | [network_manager::dbus_api:ERROR] Get org.freedesktop.NetworkManager.AccessPoint::RsnFlags property failed on /org/freedesktop/NetworkManager/AccessPoint/20: wrong property type
screenly-anthias-wifi-connect-1  | Error: Getting access points failed
screenly-anthias-wifi-connect-1  |   caused by: D-Bus failure: Get org.freedesktop.NetworkManager.AccessPoint::RsnFlags property failed on /org/freedesktop/NetworkManager/AccessPoint/20: wrong property type
```

ealmonte32, 2023-02-28:
Oh, I see that this seems to be the same issue we are discussing on here: https://github.com/Screenly/Anthias/issues/1751

nicomiguelino, 2023-03-28:
I'll close this task once I tested the changes.

harmakhis971, 2023-03-30:
hello @nicomiguelino , first thanks for your time in this project ..
i do have a strange issue on my raspeberry model 3+ .
If i want to use Anthias Connect Wifi hotspot to configure my actual wifi network , it create the nmconnection file but it still force to use the hotspot. 
I do have the same issue if i change the file into resin-boot/system-connections/mysupadupaWifiAccessFile.nmconnection

Do you have any idea to avoid that ..

vpetersson, 2026-05-08:
The `wifi-connect` captive portal was removed entirely in #2763 (merged 2026-04-28) — the current flow expects you to set up Wi-Fi via Raspberry Pi Imager or `nmcli`/`nmtui`. Closing.

PR #1778 (2023-03-28): Fix `wifi-connect` issues.
#### Overview

- Fixes #1746
- Fixes #1747

#### Tasks

- [x] Fix relevant blocking issues.
- [x] The `anthias-wifi-connect` service must exit early if no Wi-Fi devices are found, esp. for Pi 1 and Pi 2 devices.
- [x] Create a `systemd` unit file on RPi OS so that the `anthias-wifi-connect` service will start during startup.
- [x] Add documentation on how to configure Wi-Fi on the device.
- [x] Add a disclaimer in the docs that it's not yet possible to customize the access point settings &ndash; e.g., SSID, password, and gateway (the IP address and port, which defaul [... cut]