Screenly/Anthias #1750: wlan0 pulls 192.168.46 address

jefflnester, 2023-02-24:
I have attempted to install Anthias dozens of times with slight variations. The current installation it finishes and the splash screen comes up but the web page is not accessible. 

At the command line if I do a ifconfig it shows me a 192.168.46.x address. The splash screen shows me the IP address that I had assigned to the pi (192.168.1.212). Of course the real IP address is not accessible and the 46 address is not reachable from my network. 

#### Reproduction Steps

In order to remove things the script had to do I have made sure that the OS was fully updated before beginning.  I then did:

$ bash <(curl -sL https://install-anthias.srly.io/)
y to the first question
n to the network management (if I say y to manage network the pi doesn't come back up after a reboot)
n to updates 

#### Environment

* Raspberry Pi Hardware Version: 4B
* Raspberry Pi Network Setup: built-in WIFI
* Anthias Version:  What ever the bash command returns

jefflnester, 2023-02-27:
So I figured a hack around this. I added to the root crontab an action on reboot to run a script. I configured using nmcli my wireless access point. I then created a shell script that crontab runs on reboot. This scripts sleeps for 45 seconds then brings up my wifi interface. Seems to work every time! If I ping the pi during reboot I see the ping work at first then stop working (That is when the IP address gets changed to the 46 address. Once my script executes the pings starts working again and Anthias works as designed.  The installer is someone toasting the real network configuration

extremeskillz, 2023-03-01:
@jefflnester In general this is a bug. Its been reported by several posts and its on the latest release when install by script with the option N on the networking part. Also the latest release bricks Image based assets. Hopefully a fix comes soon.

vpetersson, 2023-03-01:
We will tackle this once we've merged in https://github.com/Screenly/Anthias/pull/1743