Screenly/Anthias #1752: NGINX Host dont found

JhonathanJJGS, 2023-02-24:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

NGINX restarting all time in docker
Log:
nginx: [emerg] host not found in upstream "anthias-server:8080" in /etc/nginx/sites-enabled/srly-ose.conf 

#### Reproduction Steps

Brand new instalation via .sh script. 

#### Environment

* Raspberry Pi Hardware Version: 3
* Raspberry Pi Network Setup:Ethernet
* Anthias Version: Last version from install script 

Thanks!

nicomiguelino, 2023-02-24:
Ideally, running `bin/upgrade_containers.sh` should work. Changes were added to the script so that services will be renamed from `srly-ose-*` to `anthias-*`. Here's the snippet:

```python
if [[ -n $(docker ps | grep srly-ose) ]]; then
    # @TODO: Rename later
    set +e
    docker container rename srly-ose-wifi-connect anthias-wifi-connect
    docker container rename srly-ose-server anthias-server
    docker container rename srly-ose-viewer anthias-viewer
    docker container rename srly-ose-celery anthias-celery
    docker container rename srly-ose-websocket anthias-websocket
    docker container rename srly-ose-nginx anthias-nginx
    set -e
fi
```

Thanks for raising this issue @JhonathanJJGS. What would the console output look like if you try running `docker compose ps` or `docker ps`?

JhonathanJJGS, 2023-02-24:
Hmm maybe the file nginx.conf is not being copied inside de nginx container when i upgrade_container.sh, what do you think?

closed 2023-02-24 by JhonathanJJGS