Screenly/Anthias #1754: Anthias v0.18.6 wifi-ap working, no wizard to connect to existing network, no other way to configure

MisterAnderson91, 2023-02-27:
#### Overview of the Issue

Anthias v0.18.6 image does not proceed past the splash screen when no initial network connection provided.
Wifi-AP does show up. It can be connected to. When connected, Anthias splash screen still remains.
Once connected, going to 192.168.42.1 gives the Anthias settings pages but does not show any wizard to get the device connected to an existing wifi network.

There is also no option to enable SSH access to manually configure wifi access, and using a keyboard connected directly, their seems to be no way to bring up a TTY to sign in and configure.

#### Reproduction Steps

* Apply v0.18.6 image from releases to microSD
* Boot Raspberry pi and leave on anthias splash screen until Anthias WiFi Connect AP is available on another device
* Connect to the AP, manually go to 192.168.42.1 in a browser.

#### Environment

* Raspberry Pi Hardware Version: 4B
* Raspberry Pi Network Setup: No network connection (wifi connection desired)
* Anthias Version: 0.18.6 pre-release image

vpetersson, 2023-02-27:
> Anthias v0.18.6 image does not proceed past the splash screen when no initial network connection provided.
Wifi-AP does show up. It can be connected to. When connected, Anthias splash screen still remains.
Once connected, going to 192.168.42.1 gives the Anthias settings pages but does not show any wizard to get the device connected to an existing wifi network.

This has been resolved. However, there's an issue with Balena where the changes aren't being deployed properly https://forums.balena.io/t/frequently-stuck-in-build-in-progress/366916/3

> There is also no option to enable SSH access to manually configure wifi access, and using a keyboard connected directly, their seems to be no way to bring up a TTY to sign in and configure.

This will likely never be supported in the Balena based disk images. If you need SSH access, the recommended way is to use the Raspbian based installation method.

MisterAnderson91, 2023-02-27:
I'll try building from the script again, but raspberry pi OS prebuilt images would be really useful and appreciated. It seems like BalenaOS is limiting the ability to use workarounds for problems with its lack of TTY access, and building the raspberry pi OS image from the script every time is painful in an environment with restricted internet access.

ealmonte32, 2023-02-28:
@MisterAnderson91 

If you need to give the Pi Wi-Fi access before it even gets connected to a monitor/keyboard, simply generate the `wpa_supplicant.conf` file by following the instructions I gave here: https://forums.screenly.io/t/wifi-not-working/817/2

This is in response to you saying `there is no other way to configure`, when in reality you can use `ctrl+alt+f1` after anthias installs (even when screen is black) and log into the Pi and run `sudo raspi-config` and set up your wireless that way, while my method is the normal headless method to deploy Pi's with Wi-Fi already "preconfigured".

MisterAnderson91, 2023-02-28:
> This has been resolved. However, there's an issue with Balena where the changes aren't being deployed properly https://forums.balena.io/t/frequently-stuck-in-build-in-progress/366916/3

@vpetersson I've reimaged with raspberry pi OS and installed from the script. Seems to behave the same way; the wifi hotspot and everything activates perfectly, but still no wifi setup wizard to connect it to an existing network.

The only difference is that instead of sitting on the splash screen, it shows an IP address it had during the script install (not its current one). Asset display doesn't seem to work either, even if scheduled for 2020-2029 (the time should be correct though from the script installation when it had internet access).

I think I can make this work for now, now that the wifi AP opens.

> If you need to give the Pi Wi-Fi access before it even gets connected to a monitor/keyboard, simply generate the `wpa_supplicant.conf` file by following the instructions I gave here: [...]

@ealmonte32 You seem to be talking about the script installation on raspberry pi OS; this issue is logged against the prebuilt image with balenaOS.

I tried using raspberry pi imager to set the wifi settings (pretty sure it just writes a wpa_supplicant file) and it had no effect with balenaOS.

MisterAnderson91, 2023-02-28:
@vpetersson I've also noticed that installing from the script on rpiOS, after reboot the wifi-AP is initiated and seems to takeover the wifi connection. However, even adding another available wifi network with nmtui, it fails to connect to that network on reboot and the wifi-AP is present again. I don't see how to get this working on another wifi network.

extremeskillz, 2023-03-01:
Same here: Script builds are breaking on fresh rpi installs. I say no to networking and it installs it anyway. I already broke my upgraded screenly install with the latest upgrade. I am not upgrading my other screens. Also the latest upgrades breaks all Image based assets.

ealmonte32, 2023-03-01:
@vpetersson 
I can confirm more of this now as I tested clean install..
The wifi-connect is breaking network stuff, it removes my already configured wpa_supplicant.conf passed onto dhcpcd and it starts anthias connect wifi service without needing to since the Pi had network connectivity..
I had to force remove network-manager for it to not take over my wifi connection and I am able to actually use it normally now.. and I also did say no to managing my network during initial install.
I think wifi connect service and that start script needs to be temporarily rolled-back/disabled until fixed.

vpetersson, 2023-03-01:
We're just wrapping up the Python 3 PR -- @nicomiguelino will be looking into this today.

briancasas, 2023-03-01:
> 

Muchas gracias, realizando esta eliminación pude solucionar este problema. Incluso reiniciando el Raspberry.
pudiste visualizar activos? me esta dejando visualizar solo videos...

ealmonte32, 2023-03-01:
> > 
> 
> Muchas gracias, realizando esta eliminación pude solucionar este problema. Incluso reiniciando el Raspberry. pudiste visualizar activos? me esta dejando visualizar solo videos...

Webpages work, but I believe Images are still an issue that others have reported.

ealmonte32, 2023-03-01:
** Update **

Removing network-manager will start creating a lot of error logs on the wifi-connect container, thus you should also stop it:
```
 sudo docker stop screenly-anthias-wifi-connect-1
```

nicomiguelino, 2023-03-01:
You can check out #1762, which disables the `wifi-connect` service until a fix is applied.

nicomiguelino, 2023-03-28:
Hello, everyone. I've merged Screenly/Anthias#1718, which hopefully fixes issues like the following:
- `wifi-connect` starting even if it shouldn't (e.g., there's Internet connectivity)
- `wifi-connect` starting on Raspberry Pi 1 & 2 devices &ndash; These devices don't have a Wi-Fi adapter built-in

However, there are still some issues that I need to resolve, for which I might take some time. That's why I've decided to merge the PR (mentioned above) as soon as possible. More details could be found [here][1].

[1]: https://github.com/Screenly/Anthias/blob/master/docs/wifi-setup.md#limitations-and-known-issues

nicomiguelino, 2023-03-28:
By "no Internet connectivity", I mean not beang to access the Internet itself. For instance, not being able to ping google.com or 1.1.1.1.

Thanks for pointing that one out. I'll include that in the succeeding changes.

MisterAnderson91, 2023-03-28:
It would be great to avoid the need for an internet connection to have the wifi-connect functionality working, whether that's a selectable option or the default. The old Screenly OSE build didn't need one after the initial boot.

ealmonte32, 2023-03-29:
@MisterAnderson91 

Remember that your situation might be a unique one out of thousands of users, thus programming something (whether it's a feature, a function, etc.) for the majority of users will always (in my opinion) be the way to go vs changing something for few use-cases.

I do agree that requiring internet connectivity is not the best option, some networks might be closed networks that don't have internet access and Anthias is only being used for video + images display, but you still need it with a network connection to manage it.

MisterAnderson91, 2023-03-29:
@ealmonte32 That is fair enough, except this project has worked perfectly fine with no internet connection in the past, and there's no particularly good alternative that I've found (even when this project was sitting dead for multiple years). I'm just asking in case it was an accidental oversight, instead of an intentional change that's truly required. It may require no extra work to do it another way that doesn't exclude offline use, I'm just asking just in case so everyone's aware that it's a use-case.

I'd also say it's presumptuous too to just assume that it's the minority of use-cases, considering there's no statistics coming from anyone deploying these in that type of environment. But that's just me.

I'd happily keep using the 2020 image of Screenly OSE as it would still work, except it never got updated to rpi4 compatibility, and you just can't buy 3s anymore.

ealmonte32, 2023-03-29:
@MisterAnderson91 

The statistics I know are that when this was changed, around beginning of ~2021 🤔 until now (don't remember exactly), I've only heard of less than 5 issues be raised about needing to use it offline only, thus I am going based off that.

Regardless, in the end of it all, it should function as we all agree it should, which is the wifi-connect service running on devices that have a wlan interface, and whenever it detects no network on the host just like resin-wifi used to do.
I'm sure nico will figure it out, it's just that since containers are being used, there's many things that need to be reworked to make it function properly.

vpetersson, 2023-03-30:
The best solution for this is probably to have some kind of "manual override" for offline mode. Meaning, perhaps a file on disk or similar.

Offline mode in general works poorly on the Raspberry Pi (due to the lack of RTC). In order to make it work, you'd need to either add an RTC module to the board, or have it run on a network where the Pi is configured to set the time from an NTP server on the network. Thus, this requires a rather sophisticated user and thus a "manual override" solution is fine IMHO.