Screenly/Anthias #1755: No way to see or set time in web interface

MisterAnderson91, 2023-02-27:
#### Overview of the Issue

Anthias v0.18.6 image has no way to see the host's current time. As such, there is no way to know what time/date assets need to activate them.

Setting assets to display for "1 day" etc. seems to only use the local time of the guest computer to set the asset start/end times.

There seems to be an assumption that the anthias host will have direct unfiltered internet access constantly and can track the time via NTP using whatever NTP server is added by default, in whatever timezone is set by default.

Requesting the following fixes:

* Display the host's current time in the web interface
* Use the host's local time to calculate asset schedules when using the premade "1 day" etc. instead of the time from the guest machine, as an immediate workaround for systems without the correct time set 
* Add the ability to change the host's current time directly in the web interface 
* And if possible, add the ability to configure an alternate NTP server (and set the correct timezone) for networks with one locally available

#### Reproduction Steps

* Apply v0.18.6 image from releases to microSD
* Boot Raspberry pi and leave on anthias splash screen until Anthias WiFi Connect AP is available on another device
* Connect to the AP, manually go to 192.168.42.1 in a browser.

#### Environment

* Raspberry Pi Hardware Version: 4B
* Raspberry Pi Network Setup: No network connection (wifi connection desired)
* Anthias Version: 0.18.6 pre-release image

vpetersson, 2023-02-27:
Great observation, @MisterAnderson91. Yes, the time on the device is set using NTP as you correctly assumed. However, as things stand, there's no way to set the timezone when running on Balena. This is obviously a problem when trying to schedule assets.

We'll have to look into this and probably need to add a timezone setting.

I did just have a look however, and the timezone is currently set to UTC by BalenaOS (which is a sensible default of course).

MisterAnderson91, 2023-02-27:
> 

A timezone option would be great, but just being able to see the current system time and set it is arguably more important because at least you could still workaround an incorrect time. It also needs the option to pick the NTP server for environments where there's a local one, or only specific URLs are whitelisted for internet access, etc. At the moment it's all just invisible.

extremeskillz, 2023-03-01:
CUrrently I have bricked installs with the latest version after an upgrade with the install script but what I do to maintain time on the pi since they don't do that on their own when powered off hard is to run ntpd. I point ntpd to my internal domain controller and pfsense firewall running ntp as well which in turn is pointed to my domain controller as well. This keeps time in check on reboots and hard power power cut offs. Just a suggestion. This is done in terminal with linux commands. Also you can still run raspi-config to get most of the system configuration done before running the installation script. Just make sure the sytem time is correct before running it as it will error out.

MisterAnderson91, 2023-03-15:
@extremeskillz That is a solution for script-installs on Raspberry Pi OS, but the prebuilt images have switched to balenaOS and those options are unavailable.

Even if it was available, it's not useful unless someone is checking via ssh or similar that the time is always staying correct. It would be much more convenient to see the system time in the web interface somewhere.