Screenly/Anthias #1758: Install script ignoring network preference option?

MisterAnderson91, 2023-02-28:
#### Overview of the Issue

Installing on rpios, install script seems to ignore the y/n choice about anthias managing the network connection, and takes over the network connections anyway. The wifi-AP starts after first reboot, and any previously configured wifi connection stops auto-connecting. Network-Manager is also now installed and has taken-over the wlan0 interface apparently.

#### Reproduction Steps

* Image with latest rpios lite image (wifi set from rpi imager)
* Sign in and run bash <(curl -sL https://install-anthias.srly.io)
* Choose options with Y, N, N (Yes continue, No don't manage the network connection, No don't do a full upgrade)
* Wait for install, type Y to reboot
* After reboot, a heap of extra network interfaces are present, and the previously configured wifi auto-connect is gone (the wifi-AP starts instead)

See the image below for a freshly installed and rebooted device, showing the network interfaces. I would have expected the previously-working auto-connected wifi network to be connected to again, but it is not. 

[image]

Not sure if it's relevant, but there's also a "permission denied" error at the pull stage and a bunch of platform errors near the end:

[image]

[image]

P.S. Assets don't display on this install. I've repeated the install multiple times to make sure it wasn't just a failure in the script.

extremeskillz, 2023-03-01:
Same issue here. Glad you posted it. I rebuilt my pi's 3 times to replay this issue and I can confirm it continues to install the wifi manager, overwriting my networking. 

My hardware is PI 3b+, with latest Bullseye. 

No matter what I do the networking is getting managed, killing my current networking. If I go into raspi-config to fix it it, it gets overwritten again after a reboot. The latest update also stopped displaying Image based assets, in which mine are all .png files. URL based assets and youtube downloads work fine when the networking works. 

This is definitely a bug in the latest updates of OSE.

MisterAnderson91, 2023-03-03:
Trying the install script today, saying "N" to network management worked as expected. I was able to manually enable network-manager, connect a wifi network, and add image and video assets and they worked immediately. I will close off this issue.

closed 2023-03-03 by MisterAnderson91