Screenly/Anthias #1764: error in setting auth in web gui

beccobunsen, 2023-03-03:
installed from bash <(curl -sL https://install-anthias.srly.io)

try to set auth on web gui

Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

Raspberry Pi Model | Raspberry Pi 3 Model B Rev 1.2

master@df00592

vpetersson, 2023-03-06:
Please provide some logs using:

```bash

$ cd ~/screenly
$ docker compose logs 
[...]
```

Without this we can't move this forward.

ealmonte32, 2023-03-06:
@vpetersson 

Just confirming this is an issue.
Below is the server container log after pressing Save Settings when trying to enable basic auth.
Same fresh bash install from last week.

[image]

ealmonte32, 2023-03-06:
> @ealmonte32 nit but please use [code snippets](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) in issues, as it makes it a lot easier to work with.
> 
> But yes, that does look very much related to the Python3 migration.

Ah, sorry it was a quick screenshot to show confirmation of issue.

```
Exception on /settings [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.7/dist-packages/flask_restful/__init__.py", line 271, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.7/dist-packages/flask_cors/extension.py", line 165, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.7/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/src/app/lib/auth.py", line 250, in decorated
    return settings.auth.authenticate_if_needed() or orig(*args, **kwargs)
  File "server.py", line 1679, in settings_page
    next_auth_backend.update_settings(current_pass_correct)
  File "/usr/src/app/lib/auth.py", line 132, in update_settings
    new_pass = hashlib.sha256(new_pass).hexdigest() if new_pass else None
TypeError: Unicode-objects must be encoded before hashing
Exception on /settings [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.7/dist-packages/flask_restful/__init__.py", line 271, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.7/dist-packages/flask_cors/extension.py", line 165, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.7/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/src/app/lib/auth.py", line 250, in decorated
    return settings.auth.authenticate_if_needed() or orig(*args, **kwargs)
  File "server.py", line 1679, in settings_page
    next_auth_backend.update_settings(current_pass_correct)
  File "/usr/src/app/lib/auth.py", line 132, in update_settings
    new_pass = hashlib.sha256(new_pass).hexdigest() if new_pass else None
TypeError: Unicode-objects must be encoded before hashing
```

beccobunsen, 2023-03-07:
Sorry for the delay....

screenly-anthias-server-1  | Exception on /settings [POST]
screenly-anthias-server-1  | Traceback (most recent call last):
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2447, in wsgi_app
screenly-anthias-server-1  |     response = self.full_dispatch_request()
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1952, in full_dispatch_request
screenly-anthias-server-1  |     rv = self.handle_user_exception(e)
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask_restful/__init__.py", line 271, in error_router
screenly-anthias-server-1  |     return original_handler(e)
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask_cors/extension.py", line 165, in wrapped_function
screenly-anthias-server-1  |     return cors_after_request(app.make_response(f(*args, **kwargs)))
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1821, in handle_user_exception
screenly-anthias-server-1  |     reraise(exc_type, exc_value, tb)
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask/_compat.py", line 39, in reraise
screenly-anthias-viewer-1  | get_next_asset
screenly-anthias-viewer-1  | refresh_playlist
screenly-anthias-viewer-1  | refresh: counter: (0) deadline (2023-03-03 11:44:00) timecur (2023-03-06 12:50:34.909172)
screenly-anthias-viewer-1  | update_playlist
screenly-anthias-viewer-1  | Generating asset-list...
screenly-anthias-viewer-1  | generate_asset_list deadline: 2023-03-03 11:44:00
screenly-anthias-viewer-1  | get_next_asset after refresh
screenly-anthias-viewer-1  | get_next_asset counter 0 returning asset 5 of 5
screenly-anthias-viewer-1  | Showing asset Escuela 751 en la Expo Educativa Trelew 2022.mp4 (video)
screenly-anthias-viewer-1  | Asset URI /data/screenly_assets/381df78ffed24b87bba3c73229877232.mp4
screenly-anthias-viewer-1  | Displaying video /data/screenly_assets/381df78ffed24b87bba3c73229877232.mp4 for 192
screenly-anthias-viewer-1  | Reading config-file...
screenly-anthias-viewer-1  | Current url is null
screenly-anthias-server-1  |     raise value
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1950, in full_dispatch_request
screenly-anthias-server-1  |     rv = self.dispatch_request()
screenly-anthias-server-1  |   File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1936, in dispatch_request
screenly-anthias-server-1  |     return self.view_functions[rule.endpoint](**req.view_args)
screenly-anthias-server-1  |   File "/usr/src/app/lib/auth.py", line 250, in decorated
screenly-anthias-server-1  |     return settings.auth.authenticate_if_needed() or orig(*args, **kwargs)
screenly-anthias-server-1  |   File "server.py", line 1679, in settings_page
screenly-anthias-server-1  |     next_auth_backend.update_settings(current_pass_correct)
screenly-anthias-server-1  |   File "/usr/src/app/lib/auth.py", line 132, in update_settings
screenly-anthias-server-1  |     new_pass = hashlib.sha256(new_pass).hexdigest() if new_pass else None
screenly-anthias-server-1  | TypeError: Unicode-objects must be encoded before hashing

beccobunsen, 2023-03-07:
we can fix this :

new_pass = hashlib.sha256(new_pass).hexdigest()

whith this ?

hashlib.sha256(.sha256(new_pass).encode('utf-8')).hexdigest()

vpetersson, 2023-03-07:
@nicomiguelino can you take over this one?

nicomiguelino, 2023-03-07:
@vpetersson, on it...

nicomiguelino, 2023-03-07:
Thanks, @beccobunsen for raising this issue (and for posting a possible fix). Yes, it's because of the migration. By default, strings default to `unicode` in Python 3. A PR is on its way.

nicomiguelino, 2023-03-07:
Here's a PR that I've created &ndash; #1771. I still have to quickly test the changes before marking it as ready for review, so it's still a work in progress. I'll keep you posted.

nicomiguelino, 2023-03-08:
I'll be re-opening this issue due to issues raised via [this Anthias forum][1]. I'll continue to investigate things after a fresh install.

[1]: https://forums.screenly.io/t/can-change-all-settings-but-basic-authentification/1036/2

nicomiguelino, 2023-03-08:
I did a fresh install on Pi 3 and 4 and haven't reproduced the issue. I can still access the web UI and do SSH. Setting auth is working fine on my end. I'll be closing this issue now.

PR #1771 (2023-03-07): Fix auth issue in Web UI.
#### Overview

- Fixes #1764
closed 2023-03-08 by nicomiguelino